networkupstools/nut #2044: upsmon: power value calculation for UPSes with status [OFF]

Bomorav, 2023-09-07:
I have nut-2.8.0 controlling two Cyberpower SNMP UPSes supplying redundant power

MONITOR ups1 1 upsmaster secret primary
MONITOR ups2 1 upsmaster secret primary
MINSUPPLIES 1

If power failure occures and the first UPS reaches low battery it goes to sleep. The power is supplying from the second UPS. The problem is that if the second UPS is discharged NUT doesn't initiate system shutdown.

As I found out, the discharged UPS in sleep mode still communicates and cyberpower-mib.c driver reports status [OFF]. Upsmon does not consider this status and still counts this UPS to overall power value (2) which is greater than MINSUPPLIES(1).

jimklimov, 2023-09-08:
Nice finding!.. Just in case - is this with NUT 2.8.0 release (e.g. package)? I think one of the first found and fixed regressions dealt with something about misinterpreted states. Wondering if this issue is different?..

Can you please check if the problem is reproducible with current master (or 2.7.4)?

jimklimov, 2023-09-08:
Thanks. So the driver *is* from some recent master and should include those earlier fixes (including your PRs of late), right?

I think the fix I had in mind earlier was https://github.com/networkupstools/nut/pull/1432/files and probably does not impact detection of "OFF" state after all (but could impact the "NULL" state, in case this is what might bite you - IF you run the driver built from 2.8.0 code plus your patches, but not recent-ish master plus your patches).

Bomorav, 2023-09-08:
Yes, I'm using the latest driver, which works fine and returns the correct UPS states.

https://github.com/networkupstools/ConfigExamples/releases/download/book-3.0-20230319-nut-2.8.0/ConfigExamples.pdf
Pages 22-23

> The value of MINSUPPLIES is the key element in determining if a server with multiple power supplies should shut down. When all the UPS units can be contacted, and when their ups.status values are known, then it is the count A of those that are active, that is without [LB], which is determinant.

So if my UPS in sleep mode can be contacted and its ups.status value is known ([OFF]),  it is considered as active by upsmon?

jimklimov, 2023-09-14:
Sorry about the delay, finally got a moment to look at this intently. While scrolling through `upsmon.c::parse_status()` I found that it considers several status key words, but not `"OFF"`: https://github.com/networkupstools/nut/blob/91b3ee05743a9f491276b57877050d283c978085/clients/upsmon.c#L1887-L1939

The "OFF" value is used and `status_set()` extensively in the codebase however, and is long defined in our standards as "UPS is offline and is not supplying power to the load ": https://github.com/networkupstools/nut/blob/91b3ee05743a9f491276b57877050d283c978085/docs/new-drivers.txt#L210-L261

So currently my best guess is that you could benefit from tinkering with code to add handling for such situation (UPS manageable, but its load is powered off) and if there's an iteration that works better than now - propose a PR ;)

I think I remember some discussions about it, that a very closely related case is about support of manageable outlet groups (or individual outlets, especially on ePDUs) - so the UPS overall maybe "OL" but the electric socket your server knows it is fed from is "OFF"; such situations could need additional `MONITOR` parameters or some extended syntax for `powerdevice@hostname` to monitor that outlet group. So maybe that got bogged down in talks and neither case got addressed yet.

On a related note, the `BYPASS` case does not seem to be handled here either. For practical purposes it could be similar to `LB` I suppose - there's maintenance on the UPS, load is fed from the wall, and might disappear any moment without clear notice. At least, if the UPS is not longer manageable, we might want to consider it dead, same as loss of connection during an outage, maybe? (An unplugged comms cable would be considered a power cut though) Not sure here, either...

Bomorav, 2023-09-15:
Thanks.
This OFF state is probably specific to cyberpower snmp UPSes. I modified uspmon.c for my needs, here is my modification against 2.8.0 if it would be useful for someone (sorry, I'm not a programmer :-))

```c
diff -uNrp nut-2.8.0/clients/upsmon.h nut-2.8.0.p/clients/upsmon.h
--- nut-2.8.0/clients/upsmon.h	2022-04-27 00:03:31.000000000 +0200
+++ nut-2.8.0.p/clients/upsmon.h	2023-04-27 15:08:15.044584966 +0200
@@ -31,6 +31,7 @@
 #define ST_LOGIN       (1 << 5)       /* we are logged into this UPS              */
 #define ST_CONNECTED   (1 << 6)       /* upscli_connect returned OK               */
 #define ST_CAL         (1 << 7)       /* UPS calibration in progress (CAL)        */
+#define ST_OFF         (1 << 8)       /* UPS is off or on sleep (OFF)             */
 
 /* required contents of flag file */
 #define SDMAGIC "upsmon-shutdown-file"
``` 

```c
diff -uNrp nut-2.8.0/clients/upsmon.c nut-2.8.0.p/clients/upsmon.c
--- nut-2.8.0/clients/upsmon.c	2022-04-23 13:56:06.000000000 +0200
+++ nut-2.8.0.p/clients/upsmon.c	2023-04-27 23:29:00.418401200 +0200
@@ -442,6 +442,7 @@ static void ups_on_batt(utype_t *ups)
 	do_notify(ups, NOTIFY_ONBATT);
 	setflag(&ups->status, ST_ONBATT);
 	clearflag(&ups->status, ST_ONLINE);
+	clearflag(&ups->status, ST_OFF);
 }
 
 static void ups_on_line(utype_t *ups)
@@ -463,6 +464,7 @@ static void ups_on_line(utype_t *ups)
 
 	setflag(&ups->status, ST_ONLINE);
 	clearflag(&ups->status, ST_ONBATT);
+	clearflag(&ups->status, ST_OFF);
 }
 
 /* create the flag file if necessary */
@@ -809,7 +811,7 @@ static void recalc(void)
 		/* crit = (FSD) || (OB & LB) > HOSTSYNC seconds */
 		if (is_ups_critical(ups))
 			upsdebugx(1, "Critical UPS: %s", ups->sys);
-		else
+		else if (!flag_isset(ups->status, ST_OFF))
 			val_ol += ups->pv;
 
 		ups = ups->next;
@@ -1697,7 +1699,11 @@ static void parse_status(utype_t *ups, c
 			upsreplbatt(ups);
 		if (!strcasecmp(statword, "CAL"))
 			ups_cal(ups);
-
+		if (!strcasecmp(statword, "OFF")) {
+			setflag(&ups->status, ST_OFF);
+			clearflag(&ups->status, ST_ONLINE);
+			clearflag(&ups->status, ST_ONBATT);
+		}
 		/* do it last to override any possible OL */
 		if (!strcasecmp(statword, "FSD"))
 			ups_fsd(ups);
```

jimklimov, 2023-09-15:
Seems reasonable, after that research :)

Care to post a PR, to log the changeset in your name? :)

As for "CPS SNMP only" - no, there are many (sub)drivers with different techs that have a `status_set("OFF")` or equivalent in their sources - e.g. many USB UPSes stay connected even if their load is administratively off, it seems. Even if it were just one device type... you've stepped into this problem - chances are, someone else will.

Bomorav, 2023-09-15:
I'll leave it up to you to post a PR, no need to mention my name.

I have been using this code with dual Cyberpower UPS for several months now and have not experienced a problem.

jimklimov, 2023-09-15:
Several months? So, you've actually come to similar conclusions, and proofed them IRL? Nice to have convergences like this, builds confidence up a bit :)

jimklimov, 2023-09-16:
Hello, would you have a chance to check how the proposed change in https://github.com/jimklimov/nut/tree/issue-2044 behaves for you?

jimklimov, 2023-09-17:
Behaved well in local tests simulated with NIT, PR merged.

PR #2055 (2023-09-17): `upsmon` should handle `OFF` and `BYPASS` UPS states for power value calculation
Closes: #2044
closed 2023-09-17 by jimklimov