networkupstools/nut #2050: Small question about building driver from source for UPS connected to OpenWRT router

Voidstranger, 2023-09-14:
Hi, I have PowerWalker 2200 SHL UPS connected to OpenWRT router. I'm using it with usbhid-ups driver.
I want to have an option to switch UPS's beeper off, but now I can't do it.
I want to build usbhid-ups driver from source (maybe newest build has some fixes for this UPS) and install it on the router.
How can I do that?

jimklimov, 2023-09-14:
Cheers, your question got me to revise the few hits of "openwrt" in this issue tracker, and labeling them for easier search later.

It seems your best shot would be to start from the official packaging recipes, substitute newer sources (eventually some results of your own experiments) and try to get that built. Also can ask that community for more points of view :)
* https://github.com/openwrt/packages/tree/master/net/nut

That said, maybe you can just get away with a cross-build (or a build with qemu-backed container or VM "natively" running the CPU of your router - probably some MIPS, RISC or ARM chip), and copying the binaries and libraries involved. I did something like this to build newer `e2fsck` utilities for an Android phone, on Windows (well, WSL2 Ubuntu) of all places, and it worked.

I had a write-up about LXC container setup for NUT CI at https://github.com/networkupstools/nut/blob/master/docs/ci-farm-lxc-setup.txt and there are some hints about using qemu-static for non-native CPU builds, but there's really not much to it after registering the binary handler in the host. Maybe Debian buildroots (or Dockerfiles people float around) would do most of the same for you, without the need for a fully-fledged container, but I did not follow those paths myself. Then there's also Termux project, to use your Android phone as a native non-x86 Linux system ;)

It *may* be easier however, for the first shot, to build and run NUT on your workstation/laptop/... e.g. following https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests and connect the USB cable from your UPS there, and check if it offers the beeper control. This ability may depend on the "subdriver" involved (currently not choosable manually, depends on USB IDs built into NUT subdriver sources) and on the USB HID endpoints that the particular device exposes. If it is not currently supported for yours, you can trawl the driver debug logs (bump verbosity) and/or use `usbhid-ups -x explore` to try discovering if there are reports that your device shouts and the current mapping table in the subdriver does not consider yet (alas, for many data points and commands, there are several ways to pass them) - see https://github.com/networkupstools/nut/blob/master/docs/hid-subdrivers.txt for general suggestions.

Voidstranger, 2023-09-14:
Unfortunaly, I'm far from that UPS and router now. So I can't just connect UPS to laptop.
I will try to ask this question on OpenWRT forums. It seems, OpenWRT repository has pretty outdated usbhid-ups driver. 
Here, for example, I can see, that NUT use MGE HID driver (which has beeper disabling function). My NUT is using Phoenixtec/Liebert HID driver, which lacks most of the function.

jimklimov, 2023-09-14:
Those two are a known "problem", since some devices by either have same USB VID:PID and are differentiated by string (manufacturer) or similar clues to pick one or another subdriver. This was supposed to be based on vendor info, but for most cases was a loose guess and the default catch-all varied between NUT releases (and a motivator to get a `subdriver` config keyword there). On the master branch there was recently a PR adding many data-points to one of those...