networkupstools/nut #2052: passing null values to upsc results in kernel segmentation fault

desertwitch, 2023-09-14:
when bash scripting with `upsc` it could happen that null variables are passed to `upsc` by mistake, reproducible with the terminal command `upsc $UPS@$HOST` for example - which would result in: `upsc @` when those variables are in fact not set and/or null.

`upsc @` currently results in a kernel segmentation fault, see:

```
root@Tower:~# upsc $UPS@$HOST
Segmentation fault
```

```
Sep 14 20:06:27 Tower kernel: upsc[28685]: segfault at 8 ip 000014a0bbce4236 sp 00007ffd21f42468 error 4 in libc-2.37.so[14a0bbc58000+169000] likely on CPU 1 (core 1, socket 0)
Sep 14 20:06:27 Tower kernel: Code: 00 00 66 90 66 0f ef c0 66 0f ef c9 66 0f ef d2 66 0f ef db 48 89 f8 48 89 f9 48 81 e1 ff 0f 00 00 48 81 f9 cf 0f 00 00 77 6a <f3> 0f 6f 20 66 0f 74 e0 66 0f d7 d4 85 d2 74 04 0f bc c2 c3 48 83

```

is there a way to recover from such an (easy to happen) mistake better than with a kernel fault - perhaps such a malformation in the arguments could be caught before it would throw the kernel in distress?

jimklimov, 2023-09-14:
Nice find, never hit that :)

That's with master?

Also, it is known as "segfault", not "kernel fault". Although your syslog
emitter could be kernel's default catcher for null pointer dereference
(what probably happens here).

desertwitch, 2023-09-14:
yes, compiled from today's state of master branch
indeed - seg fault - shouldn't raise github issues from the train ;-)

jimklimov, 2023-09-14:
Thanks! PR #2053 should solve this:

````
$ ./clients/upsc @
upscli_splitname: xstrdup failed
Error: invalid UPS definition.
Required format: upsname[@hostname[:port]]
````

closed 2023-09-16 by jimklimov

desertwitch, 2023-09-17:
confirmed working on my system now - thanks a lot

PR #2053 (2023-09-16): Fix segfault in NUT clients with empty host/device name components
Closes: #2052 

The culprit was that in some libc implementations, calling `strdup(NULL)` causes a segfault, where such `NULL` can be returned by `strtok()` if there were no hits. Using `xstrdup()` introduced across NUT codebase specifically for safety checks like these turns it into a graceful failure that calling code can handle meaningfully.

See also #677 for related concerns.