networkupstools/nut #2073: low battery events without battery.*.low variable

xorbug, 2023-09-22:
[Configuring automatic shutdowns for low battery events](https://networkupstools.org/docs/user-manual.chunked/Configuration_notes.html#UPS_shutdown) section in the docs says, in step 2:

> The UPS reaches low battery (a "critical" UPS), that is to say, upsc displays:
> ups.status: OB LB
> The exact behavior depends on the specific device, and is related to such settings and readings as:
> - battery.charge and battery.charge.low
> - battery.runtime and battery.runtime.low 

but what if my ups doesn't report any such variable? `upsc` reports:
```
battery.charge: 100
battery.charge.warning: 30
battery.runtime: 2811
battery.voltage: 13.5
device.mfr: Legrand  
device.model: Legrand UPS  
device.type: ups
driver.name: usbhid-ups
driver.parameter.bus: 001
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.product: Legrand UPS
driver.parameter.productid: 0032
driver.parameter.synchronous: auto
driver.parameter.vendor: Legrand
driver.parameter.vendorid: 1CB0
driver.version: 2.8.0
driver.version.data: Legrand HID 0.2
driver.version.internal: 0.47
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.voltage: 175.0
output.voltage: 2160.0
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 0
ups.mfr: Legrand  
ups.model: Legrand UPS  
ups.productid: 0032
ups.realpower.nominal: 800
ups.status: OL CHRG
ups.vendorid: 1cb0
```
My questions are:
1. would `battery.charge.warning` do it?

jimklimov, 2023-09-22:
Can't really say right now, perhaps the best approach would be to run an experiment with some load you are okay to have abruptly powered off (a lamp?)

As the NUT docs highlight, such hardware-backed warnings depend on hardware abilities (and the presence of protocol support to set values, if configurable, and that NUT contributors found a way to send such commands). Some dumber devices may have the low-battery status just raised by hardware at a factory-defined situation.

Gotta really look into sources to confirm the next statement, but I am under impression that an `override...low` setting can be interpreted by drivers and/or upsmon to trigger shutdowns at specified points. There were mailing list discussions to that effect, with a set-up using several `dummy-ups` drivers to relay info from a "physical device" driver with different overrides to initiate shutdowns at different times for loads of different criticality - their `upsmon`s MONITOR different dummy devices.

xorbug, 2023-09-22:
Thanks @jimklimov, I think I can have some test when I can disconnect my devices a bit then, and report back (even just for the records, if my individual case is not representative enough of anything).