networkupstools/nut #2074: [HCL] Tecnoware UPS EXA PLUS 1500 supported by BLAZER_USB

ginlik1996, 2023-09-22:
Tecnoware UPS EXA PLUS 1500

https://www.tecnoware.com/it-IT/Dettaglio/Famiglia/186
https://www.tecnoware.com/Downloads/EXA_PLUS/Manual_EXA_PLUS_1000-1500_ITA.pdf

UPSC:
```
battery.charge: 100
battery.runtime: 2143
battery.voltage: 28.00
battery.voltage.high: 26.00
battery.voltage.low: 20.80
battery.voltage.nominal: 24.0
device.mfr:
device.model: 1200VA
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: blazer_usb
driver.parameter.default.battery.voltage.high: 26.00
driver.parameter.default.battery.voltage.low: 20.80
driver.parameter.default.battery.voltage.nominal: 24.00
driver.parameter.pollinterval: 2
driver.parameter.port: 005
driver.parameter.runtimecal: 300,100,600,50
driver.parameter.synchronous: auto
driver.state: quiet
driver.version: 2.8.0.1
driver.version.internal: 0.17
driver.version.usb: libusb-1.0.26 (API: 0x1000108)
input.current.nominal: 5.0
input.frequency: 50.0
input.frequency.nominal: 50
input.voltage: 233.0
input.voltage.fault: 0.5
input.voltage.nominal: 230
output.voltage: 230.5
ups.beeper.status: enabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.firmware:
ups.load: 14
ups.mfr:
ups.model: 1200VA
ups.productid: 5161
ups.status: OL
ups.temperature: 28.0
ups.type: offline / line interactive
ups.vendorid: 0665
``` 

UPSCMD:
```
Instant commands supported on UPS [Tecnoware]:

beeper.toggle - Toggle the UPS beeper
driver.killpower - Description unavailable
driver.reload - Description unavailable
driver.reload-or-error - Description unavailable
driver.reload-or-exit - Description unavailable
load.off - Turn off the load immediately
load.on - Turn on the load immediately
shutdown.return - Turn off the load and return when power is back
shutdown.stayoff - Turn off the load and remain off
shutdown.stop - Stop a shutdown in progress
test.battery.start - Start a battery test
test.battery.start.deep - Start a deep battery test
test.battery.start.quick - Start a quick battery test
test.battery.stop - Stop the battery test
```

ginlik1996, 2023-09-27:
Just to update everyone, the UPS was disconnecting after some hour, and i had to reboot the server to fix it.
Since then i added the following (same settings needed for CyberPower UPS) and now is working fine.

ups.conf
```
pollinterval=15
``` 

upsmon.conf
```
DEADTIME 25
MAXAGE 25
``` 

I also added the following to get the "Runtime Left" working.

ups.conf
```
default.battery.voltage.high = 26.00
default.battery.voltage.low = 20.80
default.battery.voltage.nominal = 24.00
runtimecal = 300,100,600,50
```

griland-code, 2023-10-07:
Hi,
I'm exeperiencing issues to configure this kind of UPS.
take in consideration that on the same host I have successfully configured another APC UPS on another USB port (051d:0002) with the driver **usbhid-ups**. Below all the details.
I don't know if I'm doing something wrong.
May I kindly ask you a little support on that?

Looking forward to your feedback, best regards 

**OS VERSION** - _Virtualized - USB passthrough by bus 0665:5161_
```
NAME="Ubuntu"
VERSION="20.04.3 LTS (Focal Fossa)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 20.04.3 LTS"
```

/var/log/syslog
```
Oct  7 19:31:38  upsmon[2464]: Poll UPS [B1-TW-EP1500@192.168.X.X] failed - Driver not connected
Oct  7 19:31:38  upsmon[2464]: UPS B1-TW-EP1500@192.168.X.X is unavailable
```

**ups.conf**
```
[B1-TW-EP1500]
    driver = "blazer_usb"
    port   = auto
    pollinterval = 15
    desc = "TECNOWARE UPS EXAPLUS 1500"
    langid_fix = "0x409"
```

**upsmon.conf**
```
MONITOR B1-TW-EP1500@192.168.X.X 1 upsmon XXXXXXXXXXXX master
```

**lsusb** - _output_
```
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 003: ID 0665:5161 Cypress Semiconductor USB to Serial
Bus 002 Device 002: ID 051d:0002 American Power Conversion Uninterruptible Power Supply
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 0627:0001 Adomax Technology Co., Ltd QEMU USB Tablet
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
```

**lsusb -v -s 002:003** - _output_
```
Bus 002 Device 003: ID 0665:5161 Cypress Semiconductor USB to Serial
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0         8
  idVendor           0x0665 Cypress Semiconductor
  idProduct          0x5161 USB to Serial
  bcdDevice            0.02
  iManufacturer           3 HID UPS
  iProduct                1 HID UPS
  iSerial                 5
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x003b
[... 101 lines cut ...]
   1.297278	blazer_status: short reply
   1.297346	Status read 3 failed
   1.297357	Trying mustek protocol...
   1.689275	blazer_status: short reply
   1.689350	Status read 1 failed
   2.081345	blazer_status: short reply
   2.081424	Status read 2 failed
   2.469249	blazer_status: short reply
   2.469316	Status read 3 failed
   2.469327	Trying megatec/old protocol...
   2.861314	blazer_status: short reply
   2.861380	Status read 1 failed
   3.249314	blazer_status: short reply
   3.249390	Status read 2 failed
   3.641241	blazer_status: short reply
   3.641315	Status read 3 failed
   3.641327	Trying zinto protocol...
   4.033284	blazer_status: short reply
   4.033355	Status read 1 failed
   4.421353	blazer_status: short reply
   4.421444	Status read 2 failed
   4.809320	blazer_status: short reply
   4.809382	Status read 3 failed
   4.809392	No supported UPS detected
```

griland-code, 2023-10-13:
hi @jimklimov,
thank you for follow up!
Ahead of your commit, what should I do? download the new version of NUT?
take in consideration that this the version of **NUT** sw installed:
- nut-client/focal,now 2.7.4-11ubuntu4 amd64 [installed]
- nut-server/focal,now 2.7.4-11ubuntu4 amd64 [installed]

any advice will be appreciated!
looking forward to your feedback, best regards

jimklimov, 2023-10-14:
Hi, one idea to try would be `nutdrv_qx` driver as it tries to incorporate several older ones (including blazer) that talk similar protocols.

The short read might be related to an issue that could fixed recently (during work on PR for armac subdriver IIRC), different vendors took different liberties with the protocol implementations. For checking if it is fixed and if it indeed is the problem you face, you'd currently need to build the current NUT sources (see wiki on "in-place builds") and run the driver from build workspace to probe if it sees the device.

DjMayone, 2024-04-24:
> Just to update everyone, the UPS was disconnecting after some hour, and i had to reboot the server to fix it.
> 
> Since then i added the following (same settings needed for CyberPower UPS) and now is working fine.
> 
> 
> 
> ups.conf
> 
> ```
> 
> pollonly = "enabled"
> 
> ``` 
> 
> 
> 
> upsd.conf
> 
> ```
> 
> MAXAGE 30
> 
> ``` 
> 
> 
> 
> upsmon.conf
> 
> ```
> 
> DEADTIME 30
> 
> ``` 
> 
> 
> 
> I also added the following to get the "Runtime Left" working.
> 
> 
> 
> ups.conf
> 
> ```
> 
> default.battery.voltage.high = 26.00
> 
> default.battery.voltage.low = 20.80
> 
> default.battery.voltage.nominal = 24.00
> 
> runtimecal = 300,100,600,50
> 
> ``` 

Do you also have this parameters for Era Plus 2000? I would like to set up runtimecal but I don't know how... thank you!