networkupstools/nut #2075: tripplite_usb driver bugs

maximilanna, 2023-09-22:
The man page for the tripplite usb driver states the driver supports extra arguments which it doesn't. Specifically:

1. device, the accepts a bus argument but not a device, which becomes a problem when you have two of the same devices
2. serial, this data isn't passed to the driver instead what is passed is ups.id which is a settable identifier with a range of 1 - 65535 using 'upsrw' utility, by default it is set to 65535. i propose this should be utilized instead of serial so that one can monitor multiple units of the same type and model.
3. allow_duplicates, the driver rejects this argument stating incompatibility. 
--
@jimklimov @seanm @sshimko @mbastiaan

jimklimov, 2023-09-22:
What version of code are you running? 1 and 3 are master branch (will be 2.8.1 soonish). Seems you run older packaged code - it should have the older man pages accompanying (or on web site - see historic release sub-site).

jimklimov, 2023-09-22:
not sure about 2 right now

jimklimov, 2023-09-22:
not at a computer now to double-check the master branch - it might be that the driver was missed when updating usb args handling across the board...

maximilanna, 2023-09-23:
I feel stupid now, I’m running Debian bullseye and the latest package available is 2.7.4-13. 

I still think #2 is a far more elegant solution. Now I gotta figure out how to compile the latest version and install it over

jimklimov, 2023-09-23:
Well, I don't think an `ups.id` is something every device supports or can
set... does it also persist across UPS power cycles for you (saved to
ppwer-independent flash)?

For devices that have it, matching can be an option. Maybe even in standard
shared code. PRs welcome, see the recent one for busport.

jimklimov, 2023-09-23:
I think that the bigger problem would be that asking for an `id` would require actual (usb hid? qx? other?) protocol interaction, while the data for initial match comes from libusb basics. So it may happen thatsolutiobs need to be in drivets' claim methods, not shared once and for all. Give it a shot, though, to see how low the friut hangs...

maximilanna, 2023-09-23:
The ups.id persists across reboots there is also no easy way to reset it, the only thing you can do is change it again. 

It seems the driver is still rejecting the device argument. Yesterday I installed 2.8.0 and now I get the error: 
`/lib/nut/tripplite_usb -s ups -x bus="001" -x port=auto -x device="004"                Network UPS Tools - Tripp Lite OMNIVS / SMARTPRO driver 0.33 (2.8.0)
Warning: This is an experimental driver.
Some features may not function correctly.

libusb1: Could not open any HID devices: insufficient permissions on everything
No matching USB/HID UPS found `

if I remove the “device” argument it works well.

maximilanna, 2023-09-23:
Also interestingly enough when I omit the device argument and it connects, it actually spits out the ups.id which only case is 101. 

` /lib/nut/tripplite_usb -s ups -x bus="001" -x port=auto
Network UPS Tools - Tripp Lite OMNIVS / SMARTPRO driver 0.33 (2.8.0)
Warning: This is an experimental driver.
Some features may not function correctly.

Detected a UPS: TRIPP LITE/TRIPP LITE SMART1000RM2U    
Using SMARTPRO protocol (3003)
Unit ID: 101
Attached to Tripp Lite SMART1000RM2U `

maximilanna, 2023-09-23:
Also allow duplicates isn’t working on 2.8.0 

`Fatal error: 'allow_duplicates' is not a valid flag for this driver. `

jimklimov, 2023-09-23:
As noted above, "#1" and "#3" (proper handling of "device" as well as relaxation to "allow_duplicates") appeared on master branch recently -- so are not part of 2.8.0 release which appeared before.

As for `101` showing up in `Unit ID` - that is a helpful bit -- at least the driver-specific recognizer might harness it after all.

maximilanna, 2023-09-23:
i'll try ;) 
fair to assume all the code that needs to be modified is simply in the single driver file?

jimklimov, 2023-09-24:
Not necessarily. The recent PR #2054 to add a `busport` matcher for USB connections in general touches on all files involved in that, I think - several USB drivers (some are "too legacy" to touch with this sort of changes easily, as I found in earlier PRs), shared code in `libusb{0,1}.c` or `usb-common.c`, docs, `nut-scanner`, etc.

But the driver itself is a fair starting point to begin unraveling from (look at code, add debug messages, maybe run it with an IDE like NetBeans to step through the code as it learns about the device, etc.).

jimklimov, 2023-09-25:
Skimmed the code going from that logged message, and the `ups.id` seems to be device-dependent:
* https://github.com/networkupstools/nut/blob/d00e5e012cd95e4e960f551587ed09e7f2d82c4f/drivers/tripplite_usb.c#L993-L996 defines certain messages that can be posted to the device
* https://github.com/networkupstools/nut/blob/d00e5e012cd95e4e960f551587ed09e7f2d82c4f/drivers/tripplite_usb.c#L1133-L1153 tries to post the query and interpret the result as an Unit ID
* with a `send_cmd()` implementation relying on having ownership of the device, to send commands and await the reports: https://github.com/networkupstools/nut/blob/d00e5e012cd95e4e960f551587ed09e7f2d82c4f/drivers/tripplite_usb.c#L582-L664

The `nut_libusb_open()` (namely in https://github.com/networkupstools/nut/blob/master/drivers/libusb1.c#L690-L694 and https://github.com/networkupstools/nut/blob/master/drivers/libusb0.c#L610-L614 lines) actually establishes an ability to use a `*callback()` method to filter if "caller" likes the device after it has been grabbed by the driver (per shared code in this method) and collected initial HID report descriptors - find more per `open_dev` function pointer in https://github.com/networkupstools/nut/blob/master/drivers/nut_libusb.h#L55-L59. This happens after https://github.com/networkupstools/nut/blob/master/drivers/libusb0.c#L463-L465 or https://github.com/networkupstools/nut/blob/master/drivers/libusb1.c#L523-L527 consider if `callback` is `NULL` (in which case they skip evaluation of report descriptors). Either way, the initial set of matchers (e.g. regexes by basic information that libusb has) happen before that.

At https://github.com/networkupstools/nut/blob/d00e5e012cd95e4e960f551587ed09e7f2d82c4f/drivers/tripplite_usb.c#L1601 and https://github.com/networkupstools/nut/blob/d00e5e012cd95e4e960f551587ed09e7f2d82c4f/drivers/tripplite_usb.c#L334 we pass `NULL` for the `callback` argument - so this seems like the place you could extend to add matching by `ups.id`.

Hope this helps ;)
Jim Klimov

maximilanna, 2023-10-07:
Thank you @jimklimov, that was very helpful, i'm took a stab at it. https://github.com/networkupstools/nut/pull/2093
Would love your thoughts

PR #2102 (2023-10-13): tripplite_usb.c Enhanced Device Identification in the tripplite_usb Driver for Tripp Lite USB UPSes
@jimklimov My apologies, I screwed up the pull request with a rebase. I just started a new pull request to have a clean slate scenario. Added the NUT_UNUSED_VARIABLE  per your recommendation.

background:
tripplite_usb.c Added UPS ID (upsid) Support: The code now includes support for matching and uniquely identifying UPS devices using the UPS ID (upsid). The upsid configuration option has been added, which accepts a regular expression to match the UPS ID string. This was done by adding a new fun [... cut]
closed 2023-10-13 by jimklimov