networkupstools/nut #2077: nut-monitor says, "UPS OR1500LCDRT2U@localhost is unavailable" even though upsc can communicate with nut-server just fine

UntouchedWagons, 2023-09-26:
My pfsense box acts as a main hub for my UPS, it communicates with the management card in my CyberPower OR1500LCDRT2U and everything else communicates with pfsense. PFSense is able to communicate with the UPS just fine but for some reason nut-monitor on my proxmox box is unable to communicate with nut-server on pfsense even though `upsc` can pull the information just fine:

    root@pve-node-01:~# systemctl status nut-monitor
    ● nut-monitor.service - Network UPS Tools - power device monitor and shutdown controller
    Loaded: loaded (/lib/systemd/system/nut-monitor.service; enabled; preset: enabled)
    Active: active (running) since Tue 2023-09-26 17:53:10 EDT; 2min 46s ago
    Main PID: 189475 (upsmon)
    Tasks: 2 (limit: 154479)
    Memory: 932.0K
    CPU: 39ms
    CGroup: /system.slice/nut-monitor.service
            ├─189475 /lib/nut/upsmon -F
            └─189476 /lib/nut/upsmon -F

    Sep 26 17:53:10 pve-node-01 systemd[1]: Started nut-monitor.service - Network UPS Tools - power device monitor and shutdown controller.
    Sep 26 17:53:10 pve-node-01 nut-monitor[189475]: fopen /run/nut/upsmon.pid: No such file or directory
    Sep 26 17:53:10 pve-node-01 nut-monitor[189475]: Could not find PID file to see if previous upsmon instance is already running!
    Sep 26 17:53:10 pve-node-01 nut-monitor[189475]: UPS: OR1500LCDRT2U@192.168.0.1 (secondary) (power value 1)
    Sep 26 17:53:10 pve-node-01 nut-monitor[189475]: Using power down flag file /etc/killpower
    Sep 26 17:53:10 pve-node-01 nut-monitor[189476]: Init SSL without certificate database
    Sep 26 17:53:10 pve-node-01 nut-monitor[189476]: Communications with UPS OR1500LCDRT2U@192.168.0.1 lost
    Sep 26 17:53:10 pve-node-01 nut-monitor[189480]: Network UPS Tools upsmon 2.8.0
    Sep 26 17:53:15 pve-node-01 nut-monitor[189476]: UPS OR1500LCDRT2U@192.168.0.1 is unavailable
    Sep 26 17:53:15 pve-node-01 nut-monitor[190223]: Network UPS Tools upsmon 2.8.0

    root@pve-node-01:~# upsc OR1500LCDRT2U@192.168.0.1
    Init SSL without certificate database
    battery.charge: 100
    battery.current: 0
    battery.runtime: 1740
    battery.runtime.elapsed: 0
    battery.voltage.nominal: 24
    device.contact: UntouchedWagons
    device.description: UPS SNMP Card
    device.location: Basement server rack
    device.mfr: CYBERPOWER
    device.model: OR1500LCDRT2U
    device.serial:
    device.type: ups
    driver.name: snmp-ups
    driver.parameter.mibs: cyberpower
    driver.parameter.pollinterval: 2
    driver.parameter.port: rmcard205.untouchedwagons.site
    driver.parameter.synchronous: auto
    driver.version: 2.8.0
    driver.version.data: cyberpower MIB 0.51
    driver.version.internal: 1.21
    input.frequency: 60
    input.voltage: 119
    output.current: 0
    output.frequency: 60
    output.voltage: 119
    ups.delay.reboot: 0
    ups.firmware: BFAD101*781
    ups.load: 45
    ups.mfr: CYBERPOWER
    ups.model: OR1500LCDRT2U
    ups.serial:
    ups.status:

Here is my upsmon.conf file:

    root@pve-node-01:~# cat /etc/nut/upsmon.conf

    RUN_AS_USER root

    MONITOR OR1500LCDRT2U@192.168.0.1 1 monuser NowGuideStandOwnJump slave

    MINSUPPLIES 1
    SHUTDOWNCMD "/sbin/shutdown -h"
    NOTIFYCMD /usr/sbin/upssched
    POLLFREQ 2
    POLLFREQALERT 1
    HOSTSYNC 15
    DEADTIME 15
    POWERDOWNFLAG /etc/killpower

    NOTIFYMSG ONLINE    "UPS %s on line power"
    NOTIFYMSG ONBATT    "UPS %s on battery"
    NOTIFYMSG LOWBATT   "UPS %s battery is low"
    NOTIFYMSG FSD       "UPS %s: forced shutdown in progress"
    NOTIFYMSG COMMOK    "Communications with UPS %s established"
    NOTIFYMSG COMMBAD   "Communications with UPS %s lost"
    NOTIFYMSG SHUTDOWN  "Auto logout and shutdown proceeding"
    NOTIFYMSG REPLBATT  "UPS %s battery needs to be replaced"
    NOTIFYMSG NOCOMM    "UPS %s is unavailable"
    NOTIFYMSG NOPARENT  "upsmon parent process died - shutdown impossible"

    NOTIFYFLAG ONLINE   SYSLOG+WALL+EXEC
    NOTIFYFLAG ONBATT   SYSLOG+WALL+EXEC
    NOTIFYFLAG LOWBATT  SYSLOG+WALL
    NOTIFYFLAG FSD      SYSLOG+WALL+EXEC
    NOTIFYFLAG COMMOK   SYSLOG+WALL+EXEC
    NOTIFYFLAG COMMBAD  SYSLOG+WALL+EXEC
    NOTIFYFLAG SHUTDOWN SYSLOG+WALL+EXEC
    NOTIFYFLAG REPLBATT SYSLOG+WALL
    NOTIFYFLAG NOCOMM   SYSLOG+WALL+EXEC
    NOTIFYFLAG NOPARENT SYSLOG+WALL

    RBWARNTIME 43200

    NOCOMMWARNTIME 600

    FINALDELAY 5

My two TrueNAS SCALE boxes seem to be fine

jimklimov, 2023-09-26:
Strangely I see no info in `ups.status` here. I wonder if it is related, and if it reflects a problem fixed in cyberpower-mib (on master branch) soon after that release... at least, there was one

UntouchedWagons, 2023-09-26:
I could quickly throw together a raspberry pi running debian and connect the UPS directly to it using a USB cable and see if that helps.

UntouchedWagons, 2023-09-26:
Yup that did the trick:

    root@pve-node-01:/etc/nut# upsc OR1500LCDRT2U@ups
    Init SSL without certificate database
    battery.charge: 100
    battery.charge.low: 10
    battery.charge.warning: 20
    battery.mfr.date: CPS
    battery.runtime: 1733
    battery.runtime.low: 300
    battery.type: PbAcid
    battery.voltage: 24.0
    battery.voltage.nominal: 24
    device.mfr: CPS
    device.model: OR1500LCDRT2U
    device.type: ups
    driver.name: usbhid-ups
    driver.parameter.bus: 001
    driver.parameter.pollfreq: 30
    driver.parameter.pollinterval: 2
    driver.parameter.port: auto
    driver.parameter.product: OR1500LCDRT2U
    driver.parameter.productid: 0601
    driver.parameter.synchronous: auto
    driver.parameter.vendor: CPS
    driver.parameter.vendorid: 0764
    driver.version: 2.8.0
    driver.version.data: CyberPower HID 0.6
    driver.version.internal: 0.47
    driver.version.usb: libusb-1.0.26 (API: 0x1000109)
    input.transfer.high: 140
    input.transfer.low: 90
    input.voltage: 118.0
    input.voltage.nominal: 120
    output.voltage: 118.0
    ups.beeper.status: disabled
    ups.delay.shutdown: 20
    ups.delay.start: 30
    ups.load: 46
    ups.mfr: CPS
    ups.model: OR1500LCDRT2U
    ups.productid: 0601
    ups.realpower.nominal: 900
    ups.status: OL
    ups.test.result: No test initiated
    ups.timer.shutdown: -60
    ups.timer.start: -60
    ups.vendorid: 0764

I still have no idea if I have auto-shutdown configured correctly but that's a project for another day.

closed 2023-09-26 by UntouchedWagons