networkupstools/nut #2078: APC Back-UPS ES 700G

ghmneumaier, 2023-09-27:
Is not working with usbhid-ups driver

jimklimov, 2023-09-27:
In particular, hear-say goes that after 2010 APC focused on providing custom Modbus-based protocols instead of common USB HID - so it could be feasible that even the same device with a FW upgrade (at home or from factory) would lose the ability to speak HID. Wondering if this could be something that impacts your situation, too.

Lack of APC Modbus support in NUT was a blind spot for too long (#139) but recently there is activity in this area, with a driver (and extension to libmodbus) growing in PR #2063.

Until that effort is completed, one approach to try would be getting a latest `apcupsd` set up to "actually" monitor the device (they added Modbus a few releases before the project went quiet), and if you need it seen in NUT-centric ecosystem - spin up NUT `apcupsd-ups` driver which would relay the info from that driver to NUT.

RainCaster, 2023-09-28:
I have been trying to get my APX Smart-UPS X 2000 to display correctly in nut. Alas, it too is "blessed" by APC's move to a proprietary scheme called Modbus. I would help with the data layout issues & development, however I can't even get the first step to work. If I install apcupsd, then it uninstalled nut. When I reinstall nut, it removes apcupsd. 
Is there a way that the two will coexist?

ghmneumaier, 2023-09-28:
I've identified the reason: a broken USB-cable. Sorry to bother you.

jimklimov, 2023-09-28:
> If I install apcupsd, then it uninstalled nut. When I reinstall nut, it removes apcupsd.

I think this was mentioned in some issue discussions. Ruled to be a packaging problem (either explicitly marked as competing packages and/or implementations of same feature, or delivering some same config/service file names and conflicting on that basis) - not a software problem. It is quite possible to build the two (or one of them, and have another packaged) and run together. Or use an OS whose packaging does not conflate NUT and apcupsd :)