networkupstools/nut #2079: nutdrv_qx reopen never triggers

EchterAgo, 2023-09-29:
When checking for #2063 how other drivers handle device reopen, I noticed that `nutdrv_qx` never really triggers its reopen code path.

The `reopen_matcher` gets used in `qc_command` only if `udev == NULL` and IIUC `udev` never will be set to NULL.

jimklimov, 2023-09-29:
I think the `fallthrough_case_reconnect` a few lines below that check should handle this - closes the device and sets `udev=NULL`.

Something to check (if this non-reconnect situation does happen in practice) is whether the conditions leading to such closure are captured by the switch/case correctly. If truly not, then there are several other USB-capable drivers to revisit similarly :)

EchterAgo, 2023-09-29:
Heh, how did I miss that? Closing this.

closed 2023-09-29 by EchterAgo