networkupstools/nut #2080: Mac OS NUT installation via Homebrew - Missing snmp-ups driver

mrmaximas, 2023-09-29:
I installed nut via homebrew then configured and tried to run the command:
@NVR ~ % upsdrvctl start
Network UPS Tools - UPS driver controller 2.8.0
Can't start /usr/local/Cellar/nut/2.8.0_1/bin/snmp-ups: No such file or directory
@NVR ~ % ls /usr/local/Cellar/nut/2.8.0_1/bin/
al175			blazer_ser		liebert-esp2		optiups			tripplitesu
apcsmart		blazer_usb		macosx-ups		powercom		upsc
apcsmart-old		clone			masterguard		powerpanel		upscmd
apcupsd-ups		clone-outlet		metasys			rhino			upscode2
bcmxcp			dummy-ups		mge-shut		richcomm_usb		upslog
bcmxcp_usb		etapro			mge-utalk		riello_ser		upsrw
belkin			everups			microdowell		riello_usb		upssched-cmd
belkinunv		gamatronic		microsol-apc		safenet			usbhid-ups
bestfcom		genericups		nutdrv_atcl_usb		skel			victronups
bestfortress		isbmex			nutdrv_qx		solis
bestuferrups		ivtscd			nutdrv_siemens-sitop	tripplite
bestups			liebert			oneac			tripplite_usb
Where is missing snmp-ups deiver and how to install it?
Thanks!

jimklimov, 2023-09-29:
CC @clepple - any pointers whom to ask about MacOS packaging?

clepple, 2023-09-29:
That appears to be how they packaged it in Homebrew (without SNMP support): https://github.com/Homebrew/homebrew-core/blob/ce0d8d33b39b37eb3ab8536ff45a85ad005924bf/Formula/n/nut.rb#L79

In Fink, we had split-offs for drivers that depended on large features like SNMP. I don't have any connections with the Homebrew maintainers, so not sure what to suggest besides contacting their support channels.

clepple, 2023-09-29:
(formula link from https://formulae.brew.sh/formula/nut )

gdt, 2023-10-02:
This is a homebrew bug, not a nut bug.   IMHO it should be closed and the submitter encouraged to bring it up with homebrew.  The nut tracker can't handle bugs for 20 packaging systems, which can't be handled in the nut source code anyway.

closed 2023-10-23 by mrmaximas