networkupstools/nut #2081: make distcheck errors with: No rule to make target 'al175.txt', needed by 'check-man-txt'

EchterAgo, 2023-09-30:
When trying to get `make distcheck` passed for #2063 I encountered an issue when checking the man pages:

```
Making check in docs
make[2]: Entering directory '/home/ago/src/nut/nut-2.8.0.1/_build/sub/docs'
Making check in man
make[3]: Entering directory '/home/ago/src/nut/nut-2.8.0.1/_build/sub/docs/man'
/usr/bin/make  check-local
make[4]: Entering directory '/home/ago/src/nut/nut-2.8.0.1/_build/sub/docs/man'
make[4]: *** No rule to make target 'al175.txt', needed by 'check-man-txt'.  Stop.
make[4]: *** Waiting for unfinished jobs....
make[4]: Leaving directory '/home/ago/src/nut/nut-2.8.0.1/_build/sub/docs/man'
make[3]: *** [Makefile:1436: check-am] Error 2
make[3]: Leaving directory '/home/ago/src/nut/nut-2.8.0.1/_build/sub/docs/man'
make[2]: *** [Makefile:789: check-recursive] Error 1
make[2]: Leaving directory '/home/ago/src/nut/nut-2.8.0.1/_build/sub/docs'
make[1]: *** [Makefile:777: check-recursive] Error 1
make[1]: Leaving directory '/home/ago/src/nut/nut-2.8.0.1/_build/sub'
make: *** [Makefile:994: distcheck] Error 1
```

This happens on master too and I haven't figured out yet what is causing this.

jimklimov, 2023-09-30:
Hi, can you check what `configure` says about its ability to build docs (does it see `asciidoc`/`a2x` and/or `aspell`)? At least that would impact man-pages, and wondering if there is anything spilling over to other docs recipe nuances, there's a cobweb of conditions there.

Did you run a `make check` before a `distcheck`? CI recipes typically do, so maybe something is created that helps pass `distcheck` not-from-scratch :\ On a related note, `make distcheck` tries to force building everything, so all weird dependencies should be present. NUT has a `make distcheck-light` to "auto"-configure the buildables instead, so it builds as much as possible on the current system (since not all platforms have "everything" to depend on). This does not seem to be the problem here, but worth keeping in mind.

Also, if it aborts the `distcheck` attempt for you, it might not delete the prototyping area involved. Do the text sources exist there, better check with `find` if there's one or more copy - maybe it gets lost among those?

Typically I'd expect the man pages to be read-only sources, so one copy e.g.:
````
:; find nut-2.8.0.1/ -name al175.txt
nut-2.8.0.1/docs/man/al175.txt
````

Given that the check runs in the out-of-tree build area `/home/ago/src/nut/nut-2.8.0.1/_build/sub/docs/man` which should only contain build products (or whatever else `make` made, in some cases symlinks or copies of original files for e.g. `dblatex` which expects all sources in one directory), maybe by some coincidence of circumstances `make` does not look into `VPATH` correctly to auto-use the originals from `srcdir`.

BTW, which platform, which `make` implementation and version?

EchterAgo, 2023-10-01:
> Hi, can you check what `configure` says about its ability to build docs (does it see `asciidoc`/`a2x` and/or `aspell`)? At least that would impact man-pages, and wondering if there is anything spilling over to other docs recipe nuances, there's a cobweb of conditions there.

`asciidoc` was not installed, it is now generating some pdfs, but still errors at the same location. I see this in the `config.log` now:

```
configure:32707: checking if asciidoc version can build manpages (minimum required 8.6.3)
configure:32742: result: yes
configure:32754: checking if a2x version can build manpages (minimum required 8.6.3)
configure:32789: result: yes
configure:32800: checking if xsltproc is present (mandatory for man page regeneration)
configure:32803: result: yes
configure:32811: checking if xmllint is present (mandatory for man page regeneration)
configure:32814: result: yes
configure:32831: checking if we have all the tools mandatory for man page regeneration
configure:32833: result: yes
configure:32836: checking if source-highlight is present (preferable for documentation generation)
configure:32842: result: no
configure:32850: checking whether to build and install documentation
configure:32852: result: auto 
configure:32951: checking desire and ability to build man documentation
configure:32953: result: auto
configure:33140: checking if we can build man
configure:33152: result: no
configure:33166: WARNING: Unable to build man documentation, and unable to install pre-built distributed copies because they are absent
configure:32951: checking desire and ability to build html-single documentation
configure:32953: result: auto
configure:32963: checking if asciidoc version can build html-single (minimum required 8.6.3)
configure:33005: result: yes
configure:32951: checking desire and ability to build html-chunked documentation
configure:32953: result: auto
configure:33022: checking if a2x version can build html-chunked (minimum required 8.6.3)
configure:33064: result: yes
configure:32951: checking desire and ability to build pdf documentation
configure:32953: result: auto
configure:33081: checking if dblatex version can build pdf (minimum required 0.2.5)
configure:33123: result: yes
configure:33210: checking whether to build specific documentation format(s)
configure:33212: result: yes  html-single html-chunked pdf
```

So yea, it is not building the man pages. When I try running `a2x --format manpage --destination-dir=. --xsltproc-opts="--nonet" docs/man/snmp-ups.txt` manually though I get the `snmp-ups.8` output.

> Did you run a `make check` before a `distcheck`? CI recipes typically do, so maybe something is created that helps pass `distcheck` not-from-scratch :\ On a related note, `make distcheck` tries to [...]

`make check` passes but running `make distcheck` after still fails in the same way.

> Also, if it aborts the `distcheck` attempt for you, it might not delete the prototyping area involved. Do the text sources exist there, better check with `find` if there's one or more copy - maybe [...]

The `al175.txt` file is not present there.

> BTW, which platform, which `make` implementation and version?

```
$ cat /etc/lsb-release 
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=23.04
DISTRIB_CODENAME=lunar
DISTRIB_DESCRIPTION="Ubuntu 23.04"
$ make -v
GNU Make 4.3
Built for x86_64-pc-linux-gnu
Copyright (C) 1988-2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

jimklimov, 2023-10-01:
Just in case, wondering if this is a fallout of #2065 somehow, at least it did impact those recipes; however neither that PR nor later builds with it merged, did not fail on CI farm. Maybe it prepares something too well that it skews the results :\ Thanks, will try to repro...

jimklimov, 2023-10-01:
Made a (proxmox) container with their Ubuntu 23.04 template image: http://download.proxmox.com/images/system/ubuntu-23.04-standard_23.04-1_amd64.tar.zst

Had issues starting systemd-tmpfiles (also same? when updating from 22.04 with `do-release-upgrade`) due to credentials loading, fixed bluntly with  https://github.com/microsoft/WSL/issues/9158#issuecomment-1323533006 and https://github.com/microsoft/WSL/issues/9158#issuecomment-1469502758 tricks.

In the container installed the bare minimum to get the build barely running, adding pkgs one by one:
````
:; apt-get install build-essential git autoconf automake libtool
````

* There `autogen.sh` exposed a warning which seems relevant, actually that one I repro'ed on other systems too so should be addressable:
````
ocs/Makefile.am:503: warning: check-local was already defined in condition TRUE, which includes condition HAVE_ASPELL and WITH_SPELLCHECK ...
docs/Makefile.am:90: ... 'check-local' previously defined here
````
Probably related to that aspell recipe refactoring linked earlier, will check.

As for the `make distcheck` failure, can't confirm yet:
````
git clone ...
./autogen.sh
./configure
make distcheck-light
### uses "auto" for targets so does not fail upon lack of libusb-dev, etc. like strict distcheck does
````

The "dist" location contains `nut-2.8.0.1/docs/man/al175.txt`...

FWIW it is also in the generated `make dist` tarball - so please check if *that* recipe somehow misbehaves on your side:
````
:; tar tzvf nut-2.8.0.1.tar.gz | grep al175
-rw-r--r-- 0/0           33316 2023-10-01 20:45 nut-2.8.0.1/drivers/al175.c
-rw-r--r-- 0/0            1632 2023-10-01 20:54 nut-2.8.0.1/docs/man/al175.txt
````

jimklimov, 2023-10-01:
FWIW, *all* the packages currently present in the container are:
````
:; dpkg -l | cat
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                            Version                                 Architecture Description
+++-===============================-=======================================-============-================================================================================
ii  adduser                         3.129ubuntu1                            all          add and remove users and groups
ii  apparmor                        3.0.8-1ubuntu2.1                        amd64        user-space parser utility for AppArmor
ii  apt                             2.6.0ubuntu0.1                          amd64        commandline package manager
ii  apt-utils                       2.6.0ubuntu0.1                          amd64        package management related utility programs
ii  autoconf                        2.71-3                                  all          automatic configure script builder
ii  automake                        1:1.16.5-1.3                            all          Tool for generating GNU Standards-compliant Makefiles
ii  autotools-dev                   20220109.1                              all          Update infrastructure for config.{guess,sub} files
ii  base-files                      12.3ubuntu2                             amd64        Debian base system miscellaneous files
ii  base-passwd                     3.6.1                                   amd64        Debian base system master password and group files
ii  bash                            5.2.15-2ubuntu1                         amd64        GNU Bourne Again SHell
ii  bash-completion                 1:2.11-6ubuntu1                         all          programmable completion for the bash shell
ii  bind9-dnsutils                  1:9.18.12-1ubuntu1.2                    amd64        Clients provided with BIND 9
ii  bind9-host                      1:9.18.12-1ubuntu1.2                    amd64        DNS Lookup Utility
ii  bind9-libs:amd64                1:9.18.12-1ubuntu1.2                    amd64        Shared Libraries used by BIND 9
ii  binutils                        2.40-2ubuntu4.1                         amd64        GNU assembler, linker and binary utilities
ii  binutils-common:amd64           2.40-2ubuntu4.1                         amd64        Common files for the GNU assembler, linker and binary utilities
ii  binutils-x86-64-linux-gnu       2.40-2ubuntu4.1                         amd64        GNU binary utilities, for x86-64-linux-gnu target
ii  bsdextrautils                   2.38.1-4ubuntu1                         amd64        extra utilities from 4.4BSD-Lite
ii  bsdutils                        1:2.38.1-4ubuntu1                       amd64        basic utilities from 4.4BSD-Lite
ii  build-essential                 12.9ubuntu3                             amd64        Informational list of build-essential packages
ii  busybox-static                  1:1.35.0-4ubuntu1                       amd64        Standalone rescue shell with tons of builtin utilities
ii  bzip2                           1.0.8-5build1                           amd64        high-quality block-sorting file compressor - utilities
ii  ca-certificates                 20230311ubuntu0.23.04.1                 all          Common CA certificates
ii  command-not-found               23.04.0                                 all          Suggest installation of packages in interactive bash sessions
ii  console-setup                   1.217ubuntu3                            all          console font and keymap setup program
ii  console-setup-linux             1.217ubuntu3                            all          Linux specific part of console-setup
ii  coreutils                       9.1-1ubuntu2                            amd64        GNU core utilities
ii  cpio                            2.13+dfsg-7.1                           amd64        GNU cpio -- a program to manage archives of files
ii  cpp                             4:12.2.0-3ubuntu1                       amd64        GNU C preprocessor (cpp)
ii  cpp-12                          12.3.0-1ubuntu1~23.04                   amd64        GNU C preprocessor
ii  cron                            3.0pl1-151ubuntu1                       amd64        process scheduling daemon
ii  cron-daemon-common              3.0pl1-151ubuntu1                       all          process scheduling daemon's configuration files
ii  dash                            0.5.12-2ubuntu1                         amd64        POSIX-compliant shell
ii  dbus                            1.14.4-1ubuntu1                         amd64        simple interprocess messaging system (system message bus)
ii  dbus-bin                        1.14.4-1ubuntu1                         amd64        simple interprocess messaging system (command line utilities)
ii  dbus-daemon                     1.14.4-1ubuntu1                         amd64        simple interprocess messaging system (reference message bus)
ii  dbus-session-bus-common         1.14.4-1ubuntu1                         all          simple interprocess messaging system (session bus configuration)
ii  dbus-system-bus-common          1.14.4-1ubuntu1                         all          simple interprocess messaging system (system bus configuration)
[... cut ...]
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=lunar
LOGO=ubuntu-logo
````

````
make --version
GNU Make 4.3
Built for x86_64-pc-linux-gnu
Copyright (C) 1988-2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
````

jimklimov, 2023-10-02:
Posted a PR with a small fix for that "duplicate check-local definition" warning. Might not be much, but can you please check in your environment if it is in fact the problem? :)

e.g. cherry-pick or apply otherwise the change in https://github.com/networkupstools/nut/pull/2086/commits/e214a008768674bf8eb341705106cace944444ea

EchterAgo, 2023-10-02:
Ok, so turns out this only happens after running a build in the configured source tree before running `make distcheck`. After running `make clean` it passes.

jimklimov, 2023-10-02:
That's still not looking right... a typical CI routine is roughly to `configure && make check && make distcheck` so a "unclean" build workspace is no excuse for the failure :\

EchterAgo, 2023-10-02:
Ah, of course! It only includes the selected drivers in the `make dist` tarball and I'm running configure with `--with-drivers=usbhid-ups,apc_modbus` in the non-working source tree.

EchterAgo, 2023-10-02:
Yes, initially I suspected something wrong with my inclusion of the `apc_modbus` man page, and I did find that I missed adding it to `HTML_MODBUS_MANS`, but when I noticed it also includes the `usbhid-ups` man page but no other driver man page I knew what was going on.

It is also interesting that the configure check for building man pages fails because `snmp-ups.txt`, maybe some dummy file or `<< EOF` input should be used for the check.

jimklimov, 2023-10-02:
Well, the `configure` check relies on having whole NUT sources, and `make dist` is the tool to create a source release archive with everything required to build NUT. So it failing on `distcheck` in this case is a feature (of autotools generally) informing us that the archive is not suitable. Maybe not-including "all" MAN pages (sources thereof) into a `make dist` tarball is a bug to address, though possibly just by docs - "this is a maintainer tool intended to run on fully prepared systems" or some such. IIRC the dist tarball includes a copy of processed man pages (not HTML/PDF though) to avoid requiring the build systems to have asciidoc. When `ccache` is involved, docs generation takes the majority of time spent in CI ;)

EchterAgo, 2023-10-02:
Yea, to me it also seems like `make dist` should always include all the files. It isn't failing because the generated files are missing, but because the source files are not in the dist tarball.

EchterAgo, 2023-10-02:
Should I close this or do you want to keep it open to track changes you want to do?

jimklimov, 2023-10-02:
I guess I'll close it when I address the issue somehow...

PR #2091 (2023-10-07): Fix docs makefiles
Closes: #2081 (to always distribute all man page sources - even if the build was configured only for a few drivers)

Note: thinking about it, this might still falter in contrived use-cases... The dist tarball is supposed to include pre-built man pages (from the distributor's system), so the building systems have an option to not re-generate the pages or even to not have tools for that, and yet to produce packages convenient for end-users. A `configure --with-drivers=usbhid-ups` sort of build config would only pre-generate and dist-tarball that driver's man page. TLDR: "real" [... cut]
closed 2023-10-07 by jimklimov