networkupstools/nut #2083: Having libusb-1.0(-dev) installed did not suffice without `pkg-config`

jimklimov, 2023-10-01:
> Strangely, I had libusb-1.0 installed 
> ```
> sudo aptitude search libusb-1.0                       
> i A libusb-1.0-0                              - userspace USB programming library
> i   libusb-1.0-0-dev                          - userspace USB programming library development fil
> ```
> but nut wouldn't find it, even when I removed 0.1
> ```
> configure: can not check libusb settings via pkg-config
> checking for libusb-config... no
> checking for libusb preferred version... none
> configure: WARNING: Defaulting libusb configuration
> checking for libusb cflags...
> checking for libusb ldflags... -lusb
> checking for usb.h... no
> checking for usb_init... no
> configure: error: USB drivers requested, but libusb not found.
> ```
> 
> Installing pkg-config got it compiling.

_Originally posted by @galapogos01 in https://github.com/networkupstools/nut/issues/139#issuecomment-1742035112_

TODO: See if there still is a `libusb-config` sort of script, perhaps by some other name. Or try more ways of detecting the library in m4 scripts (search for header and lib in standard paths, attempt to link?)

gdt, 2023-10-02:
I think we should just document that pkg-config is required, look for it, and fail.   From a packaging point of view, what we want is clear rules, and secondarily to avoid big dependencies if they aren't necessary.    But pkg-config is tiny, and everything else needs it.   If this needs qt5-something, that would be another matter!

Really this bug can be solved by documenting that pkg-config is required in the building instructions, and no more, IMHO.

jstarcher, 2024-01-23:
Just ran into this and I agree! A quick add of pkg-config and compilation worked.

jimklimov, 2024-01-23:
I am not sure it is ubiquitous, present on all platforms that NUT gets built for (including embedded like OpenWRT). This is why fallbacks to provide libs and includes (as `configure` CLI arguments) for many of the detectable dependencies exist.

Also, for the past few years there is a pretty exhaustive `docs/config-prereqs.txt` (also used as a chapter in larger docs) about needed packages on a number of platforms, based on the experience preparing NUT CI workers.