networkupstools/nut #2099: Can't connect to ARMAC Home Line Interactive 1500VA

reisept, 2023-10-10:
I have an UPS from Armac that is connected to my linux server using USB but I can't make it work with NUT and putting it here because I'm not sure if it's some issue with the code or with me.
When I run the nutdrv_qx with subdriver armac using
`~/nut/drivers$ sudo ./nutdrv_qx -DDDDD -s MyArmac -x port=auto -x vendorid=0925 -x productid=1234 -x subdriver=armac -u eduardo `
I always get the error bellow and also some lines with "voltronic" that I don't understand:
`   0.410101     [D2] voltronic_qs_hex_protocol: invalid protocol [M]
   0.410105     [D4] ups_infoval_set: failed to preprocess value [ups.firmware.aux: M]
...
   7.735151     [D4] armac command Q1
   7.738113     [D1] interrupt read error: error submitting URB: No such file or directory (-2)
`
I'm on Ubuntu 22.04 and libusb 0.1
I attached the output of lsusb and nutdrv_qx.
[lsusb_output.log](https://github.com/networkupstools/nut/files/12862251/lsusb_output.log)
[nutdrv_qx.log](https://github.com/networkupstools/nut/files/12862252/nutdrv_qx.log)

jimklimov, 2023-10-10:
This driver supports "protocol" and "subdriver" settings, you've only fixed one so it lock-picks the other.

Does `-u eduardo` own the USB devfs nodes (e.g. did you set up `udev` so and refresh its service)?

reisept, 2023-10-10:
I set a rule on udev to grant access to group nut (that eduardo is part of) if that what you mean, and restart the udev service
> cat /etc/udev/rules.d/90-nut-ups.rules
> ACTION=="add", \
> SUBSYSTEM=="usb", \
> ATTR{idVendor}=="0925", ATTR{idProduct}=="1234", \
> MODE="0660", GROUP="nut"

> ll /dev/bus/usb/008/003
> crw-rw---- 1 root nut 189, 898 out 10 23:09 /dev/bus/usb/008/003

I think that is all configured correctly.

desertwitch, 2023-10-10:
`sudo nutdrv_qx -a vultech -u root -DDDDDD -x productid=1234 -x vendorid=0925 -x subdriver=armac`

Can you test as root with the above command?

desertwitch, 2023-10-10:
`sudo ./nutdrv_qx -s MyArmac -u root -DDDDDD -x productid=1234 -x vendorid=0925 -x subdriver=armac -x port=auto`

Can you test as root with the above command?

Also see #1978 , could be relevant

reisept, 2023-10-11:
@desertwitch running as root didn't change anything (see attached log), I will look at https://github.com/networkupstools/nut/issues/1978 to see if I can use anything.
Btw, is there a way to open the usb device interactively and send commands manually?

desertwitch, 2023-10-11:
also pinging @blaa for his armac expertise

reisept, 2023-10-11:
I took a look at https://github.com/networkupstools/nut/issues/1978 but by my knowledge it not relates to my case.
Is there a way to capture  the communications between a machine and the UPS through USB? If needed I have access to a windows machine.

jimklimov, 2023-10-16:
This looks odd:
````
   6.959434	[D2] Device matches
   6.959459	[D3] nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   6.959493	[D5] send_to_all: SETINFO driver.state "reconnect.updateinfo"
   6.959505	[D4] armac command Q1
   6.962420	[D1] interrupt read error: error submitting URB: No such file or directory (-2)
   6.962464	[D5] send_to_all: SETINFO driver.state "reconnect.trying"
   6.962492	[D4] qx_process: failed to preprocess answer [input.voltage]
````

Not sure OTOH what the `interrupt read error: error submitting URB` entails, but the https://sourceforge.net/p/libusb/mailman/message/9846380/ discussion suggests that this device may have a non-zero interface and endpoint number to talk to?..

USB has long had the feature to provide several device interfaces on the same connection (e.g. printer, scanner, ethernet) but UPSes did not have a need for that - not like they double as a router or a flash card. In the past couple of years however the practical need did appear (e.g. with Arduinos used as controllers for experimental UPS devices, and default channel numbers being busy) so support for tweaking them was haphazardly added in NUT. So I currently wonder if this is what bites here, and if this driver was already blessed with the ability to play the numbers.

jimklimov, 2023-10-16:
As for USB wire captures, I haven't used any so far, but I think Wireshark was mentioned in this context often...

reisept, 2023-10-21:
Update from the company after a series of emails:
- There is a newer version of PowerManager to be able to communicate with the newer Armac UPSs
- Linux is not supported on the newer UPSs

I attached a table with the models that are not supported and if anyone needs the new Windows software to test something I can provide it, just ask.
I'm returning my UPS so for me this issue can be closed.
[image]

blaa, 2023-10-21:
I'm a bit late to the party it seems. 

So it's home/office lines that are different?

To learn anything about those you'd need to start a driver on Windows and use Wireshark to gather a USB communication dump between device and computer. Probably in following cases:
- Normal work
- Communication while the power gets disconnected to see status change.
- Communication while the power gets reconnected
- Communication while there's a power and you schedule a TEST.

Probably in separate files.

With those information it might be either really simple to get it to work, or it's a completely new driver. Probably it's similar to something in the system though and should not be very hard. Depends if it's completely binary protocol or something ascii-based.

I had some ping from other person that he has problem with Armac in new NUT version, I might have some time next week to investigate. Unsure if there's bug for that...

reisept, 2023-10-23:
@blaa only the ones from 2023 forward. I still have the UPS with me so I'll give it a try to see if I can gather that info from wireshark.
If someone can add support it's one more victory to open source!