networkupstools/nut #2100: Replaced Prolink 1500VA UPS with 1250VA UPS and NUT indicates it can't connect to the USB device

JanV123, 2023-10-11:
OS: Proxmox 8.0.4 (based on Debian 12 with kernel 6.2.16-15-pve)

I just recently replaced my Prolink PRO1501SFCU 1500VA which worked fine on my Proxmox server (nut-server installed on the host) with a Prolink PRO1250SFCU 1250VA. When I hooked up the new Prolink 1250VA, it could not communicate with the USB. I decided to remove and purge nut and start fresh `sudo apt remove --purge nut nut-server nut-client nut-monitor`. After installing nut fresh, it still could not connect to the new UPS which is confusing since both come from the same brand and model line. 

The new UPS detects as 0665:5161 which is exactly the same as the old 1500VA
```
root@pve:~# lsusb
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 002: ID 0665:5161 Cypress Semiconductor USB to Serial
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 003: ID 1d6b:0104 Linux Foundation Multifunction Composite Gadget
Bus 003 Device 002: ID 0b05:18f3 ASUSTek Computer, Inc. AURA LED Controller
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Running nut scanner 
```
root@pve:~# nut-scanner -U
Scanning USB bus.
[nutdev1]
        driver = "nutdrv_qx"
        port = "auto"
        vendorid = "0665"
        productid = "5161"
        product = "USB to Serial"
        vendor = "INNO TECH"
        bus = "005"
```

This is what I have on `/etc/nut/ups.conf`
```
root@pve:~# cat /etc/nut/ups.conf
pollinterval = 15
maxretry = 3
offdelay = 180
ondelay = 300

#Enter the nut-scanner -U values here
[prolink]
        driver = nutdrv_qx
        port = auto
        desc = "Prolink PRO1250SFCU"
        vendorid = 0665
        productid = 5161
```

When running  upsdrvctl
```
root@pve:~# sudo upsdrvctl stop
Network UPS Tools - UPS driver controller 2.8.0
root@pve:~# sudo upsdrvctl start
Network UPS Tools - UPS driver controller 2.8.0
Network UPS Tools - Generic Q* USB/Serial driver 0.32 (2.8.0)
USB communication driver (libusb 1.0) 0.43
Duplicate driver instance detected (PID file /run/nut/nutdrv_qx-prolink.pid exists)! Terminating other driver!
Duplicate driver instance detected (PID file /run/nut/nutdrv_qx-prolink.pid exists)! Terminating other driver!
Duplicate driver instance detected (PID file /run/nut/nutdrv_qx-prolink.pid exists)! Terminating other driver!
Can't claim USB device [0665:5161]@0/0: Entity not found
Driver failed to start (exit status=1)
Network UPS Tools - Generic Q* USB/Serial driver 0.32 (2.8.0)
USB communication driver (libusb 1.0) 0.43
Can't claim USB device [0665:5161]@0/0: Entity not found
Driver failed to start (exit status=1)
Network UPS Tools - Generic Q* USB/Serial driver 0.32 (2.8.0)
USB communication driver (libusb 1.0) 0.43
Device not supported!
Device not supported!
Driver failed to start (exit status=1)
```

Attempting to run upsc
```
root@pve:~# upsc prolink@localhost
Init SSL without certificate database
Error: Driver not connected
```

Syslog
```
root@pve:~# cat /var/log/syslog | tail
2023-10-11T23:24:56.015613+08:00 pve nut-driver@prolink[242520]: Device not supported!
2023-10-11T23:24:56.015650+08:00 pve nut-driver@prolink[242520]: Network UPS Tools - Generic Q* USB/Serial driver 0.32 (2.8.0)
2023-10-11T23:24:56.015685+08:00 pve nut-driver@prolink[242520]: USB communication driver (libusb 1.0) 0.43
2023-10-11T23:24:56.015854+08:00 pve nut-driver@prolink[242302]: Driver failed to start (exit status=1)
2023-10-11T23:24:56.041632+08:00 pve nut-monitor[110205]: Poll UPS [prolink@localhost] failed - Driver not connected
2023-10-11T23:25:01.041848+08:00 pve nut-monitor[110205]: Poll UPS [prolink@localhost] failed - Driver not connected
2023-10-11T23:25:06.041981+08:00 pve nut-monitor[110205]: Poll UPS [prolink@localhost] failed - Driver not connected
2023-10-11T23:25:09.019777+08:00 pve systemd[1]: nut-driver@prolink.service: start operation timed out. Terminating.
2023-10-11T23:25:11.042115+08:00 pve nut-monitor[110205]: Poll UPS [prolink@localhost] failed - Driver not connected
2023-10-11T23:25:16.042261+08:00 pve nut-monitor[110205]: Poll UPS [prolink@localhost] failed - Driver not connected
```

nut server status
```
root@pve:~# systemctl status nut-server
● nut-server.service - Network UPS Tools - power devices information server
     Loaded: loaded (/lib/systemd/system/nut-server.service; enabled; preset: enabled)
     Active: active (running) since Wed 2023-10-11 23:28:00 PST; 44s ago
    Process: 243993 ExecReload=/sbin/upsd -c reload -P $MAINPID (code=exited, status=0/SUCCESS)
   Main PID: 243761 (upsd)
      Tasks: 1 (limit: 76970)
     Memory: 624.0K
        CPU: 10ms
     CGroup: /system.slice/nut-server.service
             └─243761 /lib/nut/upsd -F

Oct 11 23:28:30 pve nut-server[243761]: mainloop: Interrupted system call
Oct 11 23:28:30 pve nut-server[243993]: Network UPS Tools upsd 2.8.0
Oct 11 23:28:30 pve upsd[243761]: mainloop: Interrupted system call
Oct 11 23:28:30 pve nut-server[243761]: SIGHUP: reloading configuration
Oct 11 23:28:30 pve nut-server[243761]: /etc/nut/upsd.conf is world readable
Oct 11 23:28:30 pve nut-server[243761]: /etc/nut/upsd.users is world readable
Oct 11 23:28:30 pve upsd[243761]: SIGHUP: reloading configuration
Oct 11 23:28:30 pve systemd[1]: Reloaded nut-server.service - Network UPS Tools - power devices information server.
Oct 11 23:28:30 pve upsd[243761]: /etc/nut/upsd.conf is world readable
Oct 11 23:28:30 pve upsd[243761]: /etc/nut/upsd.users is world readable
```

I also read the nut documentation and tried moving the USB of the UPS to another port and I also tried out the usb_resetter script
```
root@pve:~# usb_resetter -d 0665:5161 --reset-device
Sending signal reset to usb device /dev/bus/usb/005/002
root@pve:~# usb_resetter -d 0665:5161 --reset-hub
unbind hub /sys/bus/usb/devices/5-1
bind hub /sys/bus/usb/devices/5-1
root@pve:~#
root@pve:~# usb_resetter --reset-hub --device 0665:5161
unbind hub /sys/bus/usb/devices/5-1
bind hub /sys/bus/usb/devices/5-1
root@pve:~# dpkg --search usbhid-ups
nut-server: /lib/nut/usbhid-ups
nut-server: /usr/share/man/man8/usbhid-ups.8.gz
```

I also tried `udevadm trigger --subsystem-match=usb --action=change`

None have fixed the problem for me.

jimklimov, 2023-10-15:
This part: `Duplicate driver instance detected (PID file /run/nut/nutdrv_qx-prolink.pid exists)! Terminating other driver!` indicates that another driver was running (whether connected or not), so your CLI invocation of `upsdrvctl` conflicted with the one managed by systemd - which in turn resuscitated the "failed" daemon so might conflict back with the one you are running in the terminal (the newest kid on the block should eventually kill older instances - part of stall avoidance).

For clearer experiment results, be sure to temporarily stop the driver unit e.g. `systemctl stop nut-driver@prolink` and then try to start the driver on terminal with higher debug verbosity, e.g. `nutdrv_qx -a prolink -DDDDDD`. I wonder if it would show up a "short read" - lately had several reports of that, which might have been addressed by recent work (on NUT master branch, not yet packaged by most distros) for the "armac" subdriver. To check if the current code fares better with that device, see https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests about building current NUT and running a probe with the driver from the build workspace (don't have to "corrupt" the existing installation until you confirm the replacement is better).

Also check the "world readable file" warnings, I don't think they can cause the file with sensitive info to be ignored, but better safe than sorry anyways.

jimklimov, 2023-10-15:
Note: if you want to collect debug logs "in-vivo" for the running NUT services, check with https://github.com/networkupstools/nut/wiki/Changing-NUT-daemon-debug-verbosity about different approaches for different releases.

JanV123, 2023-10-16:
> This part: `Duplicate driver instance detected (PID file /run/nut/nutdrv_qx-prolink.pid exists)! Terminating other driver!` indicates that another driver was running (whether connected or not), so [...]

Hi Jim. Appreciate the response. I tried launching the driver with higher verbosity as you mentioned. Is that the right option?
```
root@pve:/etc/nut# systemctl start nutdrv_qx -a prolink -DDDDDD
systemctl: invalid option -- 'D'
```
Im getting an error saying '-D' is an invalid option.

jimklimov, 2023-10-16:
Not quite so. The `systemctl` command manages SystemD units - and it indeed does not recognize such options.

Here the debugging run should have just literally started the NUT driver program with the options. Locations may vary per OS and build configuration, on Debian/Ubuntu systems it is usually tucked away under `/lib/nut`, IIRC. So try:
````
systemctl stop nut-driver@prolink
/lib/nut/nutdrv_qx -a prolink -DDDDDD # -d1
````

(the usually added `-d1` option has the driver dump info after one collection loop and exit - the probing mode; here the concern is if it would survive that far and find/initialize the device, so it might not matter much)

JanV123, 2023-10-17:
> Not quite so. The `systemctl` command manages SystemD units - and it indeed does not recognize such options.
>
> Here the debugging run should have just literally started the NUT driver program with [...]

Thank did it. Here's what I'm getting

```
root@pve:/home/kristian# systemctl stop nut-driver@prolink
root@pve:/home/kristian# ls /lib/nut/
al175         bcmxcp_usb    bestuferrups  clone-outlet  genericups    masterguard  microsol-apc          optiups       riello_ser  tripplitesu    usbhid-ups
apcsmart      belkin        bestups       dummy-ups     isbmex        metasys      nutdrv_atcl_usb       powercom      riello_usb  tripplite_usb  victronups
apcsmart-old  belkinunv     blazer_ser    etapro        ivtscd        mge-shut     nutdrv_qx             powerpanel    safenet     upscode2
apcupsd-ups   bestfcom      blazer_usb    everups       liebert       mge-utalk    nutdrv_siemens-sitop  rhino         solis       upsd
bcmxcp        bestfortress  clone         gamatronic    liebert-esp2  microdowell  oneac                 richcomm_usb  tripplite   upsmon

Broadcast message from root@pve (somewhere) (Tue Oct 17 14:59:05 2023):

UPS prolink@localhost is unavailable

root@pve:/home/kristian# /lib/nut/nutdrv_qx -a prolink -DDDDDD # -d1
Network UPS Tools - Generic Q* USB/Serial driver 0.32 (2.8.0)
USB communication driver (libusb 1.0) 0.43
   0.000000     [D1] debug level is '6'
   0.000724     [D5] send_to_all: SETINFO device.type "ups"
   0.000731     [D1] upsdrv_initups...
   0.006448     [D2] Checking device 1 of 10 (1D6B/0003)
   0.006468     [D1] Failed to open device (1D6B/0003), skipping: Access denied (insufficient permissions)
   0.006473     [D2] Checking device 2 of 10 (0665/5161)
   0.275496     [D2] - VendorID: 0665
   0.275509     [D2] - ProductID: 5161
   0.275513     [D2] - Manufacturer: INNO TECH
   0.275516     [D2] - Product: USB to Serial
   0.275519     [D2] - Serial Number: unknown
   0.275521     [D2] - Bus: 005
   0.275526     [D2] - Device: unknown
   0.275530     [D2] - Device release number: 0002
   0.275537     [D2] Trying to match device
   0.275541     [D3] match_function_regex: matching a device...
   0.275620     [D2] Device matches
   0.275624     [D2] Reading first configuration descriptor
   0.275631     [D3] libusb_kernel_driver_active() returned 0
   0.275646     [D2] Claimed interface 0 successfully
   0.275651     [D3] nut_usb_set_altinterface: skipped libusb_set_interface_alt_setting(udev, 0, 0)
   0.275660     [D5] send_to_all: SETINFO ups.vendorid "0665"
   0.275665     [D5] send_to_all: SETINFO ups.productid "5161"
   0.278495     [D3] send: QGS
   1.278579     [D3] read: Operation timed out (-7)
   1.278594     [D4] qx_process: failed to preprocess answer [input.voltage]
   1.280507     [D3] send: QGS
   2.280593     [D3] read: Operation timed out (-7)
   2.280606     [D4] qx_process: failed to preprocess answer [input.voltage]
   2.282517     [D3] send: QGS
   3.282606     [D3] read: Operation timed out (-7)
   3.282620     [D4] qx_process: failed to preprocess answer [input.voltage]
   3.284528     [D3] send: M
   4.284617     [D3] read: Operation timed out (-7)
   4.284630     [D4] qx_process: failed to preprocess answer [ups.firmware.aux]
   4.286539     [D3] send: M
   5.286620     [D3] read: Operation timed out (-7)
   5.286632     [D4] qx_process: failed to preprocess answer [ups.firmware.aux]
   5.288551     [D3] send: M
   6.288649     [D3] read: Operation timed out (-7)
   6.288662     [D4] qx_process: failed to preprocess answer [ups.firmware.aux]
   6.290574     [D3] send: M
   7.290663     [D3] read: Operation timed out (-7)
   7.290677     [D4] qx_process: failed to preprocess answer [ups.firmware.aux]
   7.292559     [D3] send: M
   8.292643     [D3] read: Operation timed out (-7)
   8.292655     [D4] qx_process: failed to preprocess answer [ups.firmware.aux]
   8.294586     [D3] send: M
[... 67 lines cut ...]
  29.334889     [D4] qx_process: failed to preprocess answer [input.voltage]
  29.336823     [D3] send: Q1
  30.336917     [D3] read: Operation timed out (-7)
  30.336934     [D4] qx_process: failed to preprocess answer [input.voltage]
  30.338834     [D3] send: Q1
  31.338914     [D3] read: Operation timed out (-7)
  31.338927     [D4] qx_process: failed to preprocess answer [input.voltage]
  31.340846     [D3] send: Q1
  32.340943     [D3] read: Operation timed out (-7)
  32.340960     [D4] qx_process: failed to preprocess answer [input.voltage]
  32.342859     [D3] send: Q1
  33.342949     [D3] read: Operation timed out (-7)
  33.342963     [D4] qx_process: failed to preprocess answer [input.voltage]
  33.344879     [D3] send: Q1
  34.344984     [D3] read: Operation timed out (-7)
  34.345000     [D4] qx_process: failed to preprocess answer [input.voltage]
  34.346897     [D3] send: Q1
  35.346988     [D3] read: Operation timed out (-7)
  35.347002     [D4] qx_process: failed to preprocess answer [input.voltage]
  35.348892     [D3] send: Q1
  36.349001     [D3] read: Operation timed out (-7)
  36.349018     [D4] qx_process: failed to preprocess answer [input.voltage]
  36.349025     Device not supported!
  36.349029     Device not supported!
```

jimklimov, 2023-10-17:
That looks odd, maybe the vendor changed protocols on the fly - so the same-identifying model no longer talks QX?

For kicks, can you check with USB HID (also with all competing driver instances stopped)?

````
:; ./drivers/usbhid-ups -x explore -s test -x port=auto -x vendorid=0655 -x productid=5161 -DDDDDD -d1
````

If it returns lots of "Path" info like below, that may be it...
````
   4.819043     [D1] Path: UPS.PowerSummary.PresentStatus.Charging, Type: Feature, ReportID: 0x01, Offset: 2, Size: 1, Value: 1
````

@blaa : could the issue you fixed on master, with byte counts and zero or other chars terminating a QX reply, pop up as an "Operation timed out" - or would that be only a "Short read"?

JanV123, 2023-10-17:
Here's what I'm getting. I also tried a couple of the other drivers as you suggested.
```
root@pve:/lib/nut# ls
al175         bcmxcp_usb    bestuferrups  clone-outlet  genericups    masterguard  microsol-apc          optiups       riello_ser  tripplitesu    usbhid-ups
apcsmart      belkin        bestups       dummy-ups     isbmex        metasys      nutdrv_atcl_usb       powercom      riello_usb  tripplite_usb  victronups
apcsmart-old  belkinunv     blazer_ser    etapro        ivtscd        mge-shut     nutdrv_qx             powerpanel    safenet     upscode2
apcupsd-ups   bestfcom      blazer_usb    everups       liebert       mge-utalk    nutdrv_siemens-sitop  rhino         solis       upsd
bcmxcp        bestfortress  clone         gamatronic    liebert-esp2  microdowell  oneac                 richcomm_usb  tripplite   upsmon
root@pve:/lib/nut# :; /lib/nut/usbhid-ups -x explore -s test -x port=auto -x vendorid=0655 -x productid=5161 -DDDDDD -d1
Network UPS Tools - Generic HID driver 0.47 (2.8.0)
USB communication driver (libusb 1.0) 0.43
   0.000000     [D1] debug level is '6'
   0.000726     [D5] send_to_all: SETINFO device.type "ups"
   0.000735     [D2] Initializing an USB-connected UPS with library libusb-1.0.26 (API: 0x1000109) (NUT subdriver name='USB communication driver (libusb 1.0)' ver='0.43')
   0.000738     [D1] upsdrv_initups (non-SHUT)...
   0.007109     [D2] Checking device 1 of 10 (1D6B/0003)
   0.007126     [D1] Failed to open device (1D6B/0003), skipping: Access denied (insufficient permissions)
   0.007129     [D2] Checking device 2 of 10 (0665/5161)
   0.276117     [D2] - VendorID: 0665
   0.276130     [D2] - ProductID: 5161
   0.276133     [D2] - Manufacturer: INNO TECH
   0.276135     [D2] - Product: USB to Serial
   0.276138     [D2] - Serial Number: unknown
   0.276141     [D2] - Bus: 005
   0.276147     [D2] - Device: unknown
   0.276152     [D2] - Device release number: 0002
   0.276158     [D2] Trying to match device
   0.276162     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.276167     [D3] match_function_regex: matching a device...
   0.276239     [D2] match_function_regex: failed match of VendorID:  665
   0.276243     [D2] Device does not match - skipping
   0.276260     [D2] Checking device 3 of 10 (1D6B/0002)
   0.276275     [D1] Failed to open device (1D6B/0002), skipping: Access denied (insufficient permissions)
   0.276279     [D2] Checking device 4 of 10 (1D6B/0003)
   0.276288     [D1] Failed to open device (1D6B/0003), skipping: Access denied (insufficient permissions)
   0.276292     [D2] Checking device 5 of 10 (1D6B/0104)
   0.276301     [D1] Failed to open device (1D6B/0104), skipping: Access denied (insufficient permissions)
   0.276305     [D2] Checking device 6 of 10 (0B05/18F3)
   0.276314     [D1] Failed to open device (0B05/18F3), skipping: Access denied (insufficient permissions)
   0.276317     [D2] Checking device 7 of 10 (1D6B/0002)
   0.276326     [D1] Failed to open device (1D6B/0002), skipping: Access denied (insufficient permissions)
   0.276330     [D2] Checking device 8 of 10 (1D6B/0003)
   0.276338     [D1] Failed to open device (1D6B/0003), skipping: Access denied (insufficient permissions)
   0.276342     [D2] Checking device 9 of 10 (05E3/0608)
   0.276349     [D1] Failed to open device (05E3/0608), skipping: Access denied (insufficient permissions)
   0.276353     [D2] Checking device 10 of 10 (1D6B/0002)
   0.276362     [D1] Failed to open device (1D6B/0002), skipping: Access denied (insufficient permissions)
   0.276365     [D2] libusb1: No appropriate HID device found
   0.276370     libusb1: Could not open any HID devices: insufficient permissions on everything
   0.276375     No matching HID UPS found
root@pve:/lib/nut# :; /lib/nut/blazer_usb -x explore -s test -x port=auto -x vendorid=0655 -x productid=5161 -DDDDDD -d1
Network UPS Tools - Megatec/Q1 protocol USB driver 0.14 (2.8.0)

Fatal error: 'explore' is not a valid flag for this driver.

Look in the man page or call this driver with -h for a list of
valid variable names and flags.
root@pve:/lib/nut# :; /lib/nut/blazer_ser -x explore -s test -x port=auto -x vendorid=0655 -x productid=5161 -DDDDDD -d1
Network UPS Tools - Megatec/Q1 protocol serial driver 1.58 (2.8.0)

Fatal error: 'explore' is not a valid flag for this driver.

Look in the man page or call this driver with -h for a list of
valid variable names and flags.
root@pve:/lib/nut# :; /lib/nut/genericups -x explore -s test -x port=auto -x vendorid=0655 -x productid=5161 -DDDDDD -d1
Network UPS Tools - Generic contact-closure UPS driver 1.38 (2.8.0)

Fatal error: 'explore' is not a valid flag for this driver.

Look in the man page or call this driver with -h for a list of
[... 9 lines cut ...]
Look in the man page or call this driver with -h for a list of
valid variable names and flags.
root@pve:/lib/nut# :; /lib/nut/apcupsd-ups -x explore -s test -x port=auto -x vendorid=0655 -x productid=5161 -DDDDDD -d1
Network UPS Tools - apcupsd network client UPS driver 0.6 (2.8.0)

Fatal error: 'explore' is not a valid flag for this driver.

Look in the man page or call this driver with -h for a list of
valid variable names and flags.
root@pve:/lib/nut# :; /lib/nut/nutdrv_qx -x explore -s test -x port=auto -x vendorid=0655 -x productid=5161 -DDDDDD -d1
Network UPS Tools - Generic Q* USB/Serial driver 0.32 (2.8.0)
USB communication driver (libusb 1.0) 0.43

Fatal error: 'explore' is not a valid flag for this driver.

Look in the man page or call this driver with -h for a list of
valid variable names and flags.
root@pve:/lib/nut# :; /lib/nut/gamatronic -x explore -s test -x port=auto -x vendorid=0655 -x productid=5161 -DDDDDD -d1
Network UPS Tools - Gamatronic UPS driver 0.03 (2.8.0)

Fatal error: 'explore' is not a valid flag for this driver.

Look in the man page or call this driver with -h for a list of
valid variable names and flags.
```

jimklimov, 2023-10-17:
Yeah, "explore" is a feature of `usbhid-ups` only so far.

Seems I had a typo with `vendorid` there - `0665` vs `0655`, sorry about that - can you try again with it fixed?

JanV123, 2023-10-17:
All good. I also failed to spot that typo. Here are the results
```
root@pve:/lib/nut# systemctl stop nut-driver@prolink
root@pve:/lib/nut# :; /lib/nut/usbhid-ups -x explore -s test -x port=auto -x vendorid=0665 -x productid=5161 -DDDDDD -d1
Network UPS Tools - Generic HID driver 0.47 (2.8.0)
USB communication driver (libusb 1.0) 0.43
   0.000000     [D1] debug level is '6'
   0.000872     [D5] send_to_all: SETINFO device.type "ups"
   0.000880     [D2] Initializing an USB-connected UPS with library libusb-1.0.26 (API: 0x1000109) (NUT subdriver name='USB communication driver (libusb 1.0)' ver='0.43')
   0.000883     [D1] upsdrv_initups (non-SHUT)...
   0.008812     [D2] Checking device 1 of 10 (1D6B/0003)
   0.008831     [D1] Failed to open device (1D6B/0003), skipping: Access denied (insufficient permissions)
   0.008834     [D2] Checking device 2 of 10 (0665/5161)
   0.276105     [D2] - VendorID: 0665
   0.276113     [D2] - ProductID: 5161
   0.276116     [D2] - Manufacturer: INNO TECH
   0.276119     [D2] - Product: USB to Serial
   0.276122     [D2] - Serial Number: unknown
   0.276125     [D2] - Bus: 005
   0.276130     [D2] - Device: unknown
   0.276134     [D2] - Device release number: 0002
   0.276138     [D2] Trying to match device
   0.276142     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.276149     [D3] match_function_regex: matching a device...
   0.276191     [D2] Device matches
   0.276195     [D2] Reading first configuration descriptor
   0.276200     [D3] libusb_kernel_driver_active() returned 0
   0.276214     [D2] Claimed interface 0 successfully
   0.276217     [D3] nut_usb_set_altinterface: skipped libusb_set_interface_alt_setting(udev, 0, 0)
   0.279108     [D3] HID descriptor, method 1: (9 bytes) => 09 21 00 01 00 01 22 1b 00
   0.279115     [D3] HID descriptor length (method 1) 27
   0.279118     [D4] i=0, extra[i]=09, extra[i+1]=21
   0.279122     [D3] HID descriptor, method 2: (9 bytes) => 09 21 00 01 00 01 22 1b 00
   0.279125     [D3] HID descriptor length (method 2) 27
   0.279130     [D2] HID descriptor length 27
   0.282105     [D2] Report Descriptor size = 27
   0.282114     [D3] Report Descriptor: (27 bytes) => 06 00 ff 09 01 a1 01 09 02 15 00 26 ff 00
   0.282118     [D3]  75 08 95 08 81 82 09 02 95 08 91 82 c0
   0.282144     Using subdriver: EXPLORE HID 0.2
   0.282149     [D1] 2 HID objects found
   0.282154     [D4] Entering libusb_get_report
   0.285103     nut_libusb_get_report: Success
   0.285112     [D1] Can't retrieve Report 00: Resource temporarily unavailable
   0.285116     [D5] hid_lookup_path: ff000001 -> not found in lookup table
   0.285120     [D5] hid_lookup_path: ff000002 -> not found in lookup table
   0.285123     [D1] Path: ff000001.ff000002, Type: Input, ReportID: 0x00, Offset: 0, Size: 8
   0.285129     [D4] Entering libusb_get_report
   0.288103     nut_libusb_get_report: Success
   0.288112     [D1] Can't retrieve Report 00: Resource temporarily unavailable
   0.288115     [D5] hid_lookup_path: ff000001 -> not found in lookup table
   0.288119     [D5] hid_lookup_path: ff000002 -> not found in lookup table
   0.288122     [D1] Path: ff000001.ff000002, Type: Output, ReportID: 0x00, Offset: 0, Size: 8
   0.288130     [D5] send_to_all: SETINFO ups.mfr "INNO TECH"
   0.288134     [D5] send_to_all: SETINFO ups.model "USB to Serial"
   0.288139     [D5] send_to_all: SETINFO ups.vendorid "0665"
   0.288143     [D5] send_to_all: SETINFO ups.productid "5161"
   0.288147     [D2] Report descriptor retrieved (Reportlen = 27)
   0.288152     [D2] Found HID device
   0.288157     [D1] Detected a UPS: INNO TECH/USB to Serial
   0.288162     [D2] find_nut_info: unknown info type: load.off.delay
   0.288166     [D2] find_nut_info: unknown info type: load.on.delay
   0.288169     [D2] find_nut_info: unknown info type: load.off.delay
   0.288174     [D5] send_to_all: SETINFO driver.version "2.8.0"
   0.288178     [D5] send_to_all: SETINFO driver.version.internal "0.47"
   0.288185     [D5] send_to_all: SETINFO driver.name "usbhid-ups"
   0.288188     [D1] upsdrv_initinfo...
   0.288193     [D5] send_to_all: SETINFO driver.version.data "EXPLORE HID 0.2"
   0.288198     [D5] send_to_all: SETINFO driver.parameter.pollfreq "30"
   0.288202     [D1] upsdrv_updateinfo...
   1.038282     [D2] nut_libusb_get_interrupt: Connection timed out
   1.038297     [D1] Got 0 HID objects...
   1.038302     [D1] Quick update...
   1.038310     [D5] send_to_all: SETINFO ups.status "OB"
   1.038316     [D5] send_to_all: DATAOK
   1.038321     [D5] send_to_all: SETINFO driver.parameter.pollinterval "2"
   1.038326     [D5] send_to_all: SETINFO driver.parameter.synchronous "auto"
   1.038332     [D5] send_to_all: SETINFO device.mfr "INNO TECH"
   1.038336     [D5] send_to_all: SETINFO device.model "USB to Serial"
   1.038339     [D1] upsdrv_updateinfo...
   1.788424     [D2] nut_libusb_get_interrupt: Connection timed out
   1.788437     [D1] Got 0 HID objects...
   1.788442     [D1] Quick update...
   1.788447     [D1] upsdrv_updateinfo...
   2.538527     [D2] nut_libusb_get_interrupt: Connection timed out
   2.538541     [D1] Got 0 HID objects...
   2.538546     [D1] Quick update...
   2.538552     [D3] Entering dstate_dump
device.mfr: INNO TECH
device.model: USB to Serial
device.type: ups
driver.flag.explore: enabled
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: 5161
driver.parameter.synchronous: auto
driver.parameter.vendorid: 0665
driver.version: 2.8.0
driver.version.data: EXPLORE HID 0.2
driver.version.internal: 0.47
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
ups.mfr: INNO TECH
ups.model: USB to Serial
ups.productid: 5161
ups.status: OB
ups.vendorid: 0665
   2.538600     [D1] upsdrv_cleanup...
```

jimklimov, 2023-10-17:
Well... that did not amount to much. Apparently some minimal USB HID responses exist, but lead directly to just a couple of vendor-extension values.

I guess we'd have to wait for ideas from people better versed in Qx protocol and/or these devices.

Do you know if the device works better with some other software, and which (and if its logs etc. would show - using which protocol)? 

For that matter, is the cabling/port okay? The "nut_libusb_get_interrupt: Connection timed out" and "[D1] Can't retrieve Report 00: Resource temporarily unavailable" sort of messages are not exactly encouraging.

Is any virtualization involved (either this being a host system that might delegate USB to a guest/container, or this being a guest and the host might have barged into the delegated device on its own)?

JanV123, 2023-10-17:
The official software for this line of Prolink UPS models is [ViewPower](http://power-software-download.com/) but I could never get it to work on Linux. The ViewPower software seems ancient and runs on Java Runtime Environment (JRE). I have tried moving around the USB cable to 2 different ports but I'll try again and report back. The cable should be okay though as it's the same one I used on the 1500VA that I replaced but I'll try another cable just to test it out just in case. NUT was indeed installed on the host Proxmox system and not on any VM. I also have not passed through specific USB ports/controllers for any VM. The only device I have passed through to a VM is a HBA holding 8 drives form my TrueNAS Core VM.
If I were to completely install NUT fresh, is there any special procedure on how to do it so that all of the previous configs as well as drivers are completely removed? All I do is `sudo apt remove --purge nut nut-client nut-server nut-monitor` but I'm not sure it completely wipes everything which might be causing the driver conflict once I install NUT again.

jimklimov, 2023-10-17:
Can't comment much on the packaging side. As far as NUT configs are concerned, they should be concentrated under `/etc/nut` or `/etc/ups`. In case of systemd customizations, there's also its `/etc/systemd/...` area to look at. And system accounts (group, user). The rest should come and go via packaged files.

You can also try a custom build e.g. per https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests

jimklimov, 2023-10-17:
And thinking out loud, while grasping at straws: what if you in fact made a Windows VM (IIRC they have trial versions nowadays, to test quickly and license or remove later) and pass-through the UPS into it as an USB device, and try ViewPower for Windows?

At least, similar tricks helped me check NUT for Windows hosted on Proxmox, to probe an USB connected UPS.

JanV123, 2023-10-18:
So I tested this new 1250VA UPS using their official ViewPower software on my Windows 10 laptop and it also would not work. I used the same cable and laptop and hooked it up to the older Prolink 1500VA UPS and it works fine. Based on this troubleshooting, it seems the problem is the USB controller of this new 1250VA unit. It detects the USB connection itself but does not communicate status information on the UPS itself. I am now working on getting this RMAd and will get back to this thread if the replacement unit will resolve this issue.

jimklimov, 2023-10-18:
My condolences on the UPS, but congratulations on probably figuring out the problem (and that NUT was not it ;) )

JanV123, 2023-10-18:
> My condolences on the UPS, but congratulations on probably figuring out the problem (and that NUT was not it ;) )

I'm thinking this is highly likely the issue. If so I would like to apologize to you for taking up your time. I really appreciate you helping me out. I'll report back once the new unit comes in and I have it hooked up.

jimklimov, 2023-10-18:
My pleasure. At least, it would have been fun to catch a vendor releasing a completely different device (protocol-wise) without changing identifications :)

JanV123, 2023-10-28:
UPDATE: So I finally got the replacement unit from Prolink. Everything works fine with the replacement. I was just unlucky the previous unit had a defective USB device.

jimklimov, 2023-10-28:
Congratulations! :)

closed 2023-10-28 by jimklimov