networkupstools/nut #2101: "Fatal EEPROM fault!" on Eaton 5SC1500

Cold-Lemonade, 2023-10-11:
I have a brand new Eaton 5SC1500 and shortly after NUT server connects to it, the ups.alarm variable reads "Fatal EEPROM fault!" and the ups.status variable reads "ALARM OL CHRG". This seems to be the same issue that has been reported previously in #1444. The user there reports that the issue seemed to clear up on its own.

From context, I am running NUT v0.12.1 as an add-on to Home Assistant along with the NUT integration. My Home Assistant is: v2023.10.1, Supervisor v2023.10.0, Operating System v10.5, and Frontend v20231005.0 (all of which is current as of today 10/11/2023).

Any insight into how to resolve this alarm message? The UPS device itself is not showing any alarms on its LCD display.

arnaudquette-eaton, 2023-10-12:
FYI, investigations in progress

Cold-Lemonade, 2023-10-12:
Thank you. In case it is useful, my 5SC1500's firmware is version 04.02.0001.

arnaudquette-eaton, 2023-10-12:
Answer from my fellows working on the UPS HID Core:

Here is the description of this HID object
EEPROM Configuration failure made that the configuration has been reset to default

This alarm is generally the consequence of a UPS FW upgrade, 
If the newer version of the UPS FW has a different EEPROM structure than the previous, this can trigger an EEPROM sanitization of the configuration.

FYI, IPP and other proprietary Eaton SW are not handling it.
In all cases, this is just a warning and not an error.

I've no final decision still. Either disable or make it clear that it's a warning

dacarson, 2024-01-15:
I have exactly the same error. 
I attached the brand new unit to a RPi and started monitoring it with NUT. It immediately showed the ALARM in ups.status and in ups.alarm it has "Fatal EEPROM fault!"
I reported it to Eaton, and they sent me another unit. The new unit reports the same thing. 

I ran `./usbhid-ups -a nutdev1 -DDDDD &> eaton_eprom_fault.txt` and attached the results as a file [eaton_eprom_fault.txt](https://github.com/networkupstools/nut/files/13942723/eaton_eprom_fault.txt).
Here is relevant parts from the file:
```
   5.922773     [D5] hid_lookup_usage: UPS -> 00840004
   5.922783     [D5] hid_lookup_usage: PowerSummary -> 00840024
   5.922802     [D5] hid_lookup_usage: PresentStatus -> 00840002
   5.922811     [D5] hid_lookup_usage: ConfigurationFailure -> ffff002d
   5.922818     [D4] string_to_path: depth = 4
   5.922829     [D3] Report[buf]: (5 bytes) => 03 01 00 00 00
   5.922837     [D5] PhyMax = 0, PhyMin = 0, LogMax = 255, LogMin = 0
   5.922845     [D5] Unit = 00000000, UnitExp = 0 
   5.922853     [D5] Exponent = 0
   5.922862     [D2] Path: UPS.PowerSummary.PresentStatus.ConfigurationFailure, Type: Feature, ReportID: 0x03, Offset: 0, Size: 8, Value: 1
   5.922870     [D5] hu_find_infoval: found Fatal EEPROM fault! (value: 1) 
...
   5.922773     [D5] hid_lookup_usage: UPS -> 00840004
   5.922783     [D5] hid_lookup_usage: PowerSummary -> 00840024
   5.922802     [D5] hid_lookup_usage: PresentStatus -> 00840002
   5.922811     [D5] hid_lookup_usage: ConfigurationFailure -> ffff002d
   5.922818     [D4] string_to_path: depth = 4
   5.922829     [D3] Report[buf]: (5 bytes) => 03 01 00 00 00
   5.922837     [D5] PhyMax = 0, PhyMin = 0, LogMax = 255, LogMin = 0
   5.922845     [D5] Unit = 00000000, UnitExp = 0 
   5.922853     [D5] Exponent = 0
   5.922862     [D2] Path: UPS.PowerSummary.PresentStatus.ConfigurationFailure, Type: Feature, ReportID: 0x03, Offset: 0, Size: 8, Value: 1
   5.922870     [D5] hu_find_infoval: found Fatal EEPROM fault! (value: 1) 
```

dacarson, 2024-01-16:
Is there a way to clear this ALARM as right now it is causing issue with my UPS monitoring software.

jimklimov, 2024-11-01:
Seems like a duplicate of #1444?

dacarson, 2024-11-01:
This problem still exists for me. I reported it to Eaton (CASE 00181249), all they did was send me another unit, so now I have two. Both of them had the same issue. However one of the UPS units are bricked. See my email to Eaton below, sent on 8/21/24 but I never got a reply:
--
In a continued effort to fix CASE 00181249, where the units tell the NUT driver that there is an Alarm. I spent a week trying to get a Windows setup (I'm a mac user, and only have macs) that would allow me to use your Windows only update tools. I would like to fix the Alarm issue so that I can rely on the Alarm when there is actually a real issue with your UPS devices. 
     
FWIW, your Windows drivers require an Intel based VM and don't work on a ARM based VM. When running Windows on Apple Silicon Mac, it runs the ARM version of Windows, and your drivers look like they install, however, they don't.
I am now running VirtualBox+Windows 10 on a Intel Mac Mini. 

Your SetUPS app did see that UPS (Hooray!!!) I was then able to kick off an update to the firmware 'eaton_5sc_lvhv11_e0_v04_02_0001_tl00.sta' however something happened during the update and it just stopped. I had the progress bar, it might have been about 40-ish percent through then it showed the Update Firmware button again. 
The 5SC1500 seems to be stuck in limbo now. The screen is lit up, but blank. The VirtualBox has identified the USB device as 'EATON BOOTLOADER'. 
I have attempted multiple times a hard reset (disconnect the internal battery, unplug the UPS from the wall, wait 30s or longer, then reconnect the battery and then plug into the wall), and it never comes back to life. It is still showing 'EATON BOOTLOADER' on the USB port.
Is there another way to recover the unit or is the unit unrecoverable?
Thanks
Davidc
--

Cold-Lemonade, 2025-07-19:
Has Eaton done anything been done to address erroneous "Fatal EEPROM fault!" status?

muccitod, 2026-11-12:
Experiencing the same with a 5SC1500.  Any fixes either with NUT or firmware to address this?  Thanks.

Cold-Lemonade, 2026-11-13:
I am running NUT as an add-on in Home Assistant (what I think is now called an "app" by Home Assistant), specifically version 0.18.2 (https://github.com/hassio-addons/app-nut) and the "Fatal EEPROM fault!" message has disappeared. I'm also running firmware 04.02.0001 on my Eaton 5SC 1500.

muccitod, 2026-11-13:
> I am running NUT as an add-on in Home Assistant (what I think is now called an "app" by Home Assistant), specifically version 0.18.2 (https://github.com/hassio-addons/app-nut) and the "Fatal EEPROM fault!" message has disappeared. I'm also running firmware 04.02.0001 on my Eaton 5SC 1500.

Interesting... I am running same versions of both and had the alarm (first set it up NUT a couple of days ago)...up until today.  UPS had been operating without a server for a year or so.  One thing I did today for the first time in...forever...was a full shutdown of the UPS.  Not just pulling the plug, letting it run on battery for a while, and plugging it back in.  But an actual shutdown and reboot of the UPS.  It might have cleared the alarm, because I'm not seeing it now either.  Thanks!