networkupstools/nut #2104: APC Back-UPS RS bi-monthly battery self-tests result in shutdown

desertwitch, 2023-10-12:
As the maintainer of the NUT plugin for UNRAID (which basically acts as a frontend for NUT) I have two people with the APC series **Back-UPS RS** (models 900MI and 1350MS) reporting that the bi-monthly battery self-tests result in NUT setting FSD on their systems. **This happens on the build from master, but does not happen on 2.8.0 or 2.7.4. stables respectively**, so it's likely an issue with changes introduced after the 2.8.0 release - the driver affected seems to be **usbhid-ups.**

The problem seems to be that right before NUT sets FSD, the driver thinks to receive an OFF event from the UPS, resulting in the MinSupplies falling below 1 (on a home setup with one UPS configured) and NUT proceeding to set FSD. This can be re-produced by triggering a manual self-test on the affected APC series. Setting the MinSupplies to 0 does not result in FSD being set by NUT.

I only have limited logs available so far and will try to get more detailed logs with an increased debug level, but this is what I have from the users affected so far. Note that NUT does not report anything else above the first line, so the problem starts with that. The users reported to hear the UPS inverter clicking during self-testing, these logs appearing and the system shutting down. This can be re-produced by triggering a manual self-test on the affected APC series.

```
Oct  8 02:35:51 UNRAID upsmon[27897]: UPS qnapups@127.0.0.1: administratively OFF or asleep
Oct  8 02:35:51 UNRAID upsd[27893]: Client qnapuser@127.0.0.1 set FSD on UPS [qnapups]
Oct  8 02:35:51 UNRAID upsmon[27897]: Executing automatic power-fail shutdown
Oct  8 02:35:51 UNRAID upsmon[27897]: Auto logout and shutdown proceeding
Oct  8 02:35:56 UNRAID shutdown[3133]: shutting down for system halt
```

[image]
Here we can see the UPS being on OL CHRG OFF, as well as FSD being set right before the system goes down.

I'll try to get more diagnostic information from the users (hopefully, they've already downgraded to 2.8.0 stable), but I thought I'd report this here nonetheless, maybe other users have the same problem and I think unexpected shutdowns are a larger issue.

jimklimov, 2023-10-12:
Thanks for the report! By message wording, seems this is the result of logic changes with PR #2055 for issue #2044, merged indeed less than a month ago. It dealt with UPSes that can be sent to administrative OFF mode and no longer feed their load, but in terms of management connection to the UPS device remain talkative and alive. Reminds me of roads paved with good intentions...

If those users can run an older NUT version (from before 2023-09-17) - does their UPS driver report that it in "CAL" mode when it is calibrating? Trying to figure out if there is a problem with upsmon (e.g. insufficient situation checks before acting - such as not doing rash things if the UPS is in CAL mode... I don't see such checks with a quick skim now), or with the driver (says it goes into administrative-OFF when in fact just calibrating, so it is more of a ONBATT in practice - the OL CHRG is a bit unexpected here). Also whether it takes time for the UPS (or driver) to report it is calibrating - maybe it does first say it goes OFF, and then discovers it is in fact CAL, but for the new upsmon logic it is too late by then... (`usbhid-ups.c::ups_status_set()` does have a stack of `status_set()`'s where `CAL` is last in the long list, `OFF` is just before it).

In either case, before a next NUT release perhaps an `upsmon.conf` keyword would be in order to disable this new logic, if it misbehaves for some other drivers in the field.

desertwitch, 2023-10-12:
Thanks for the quick response, my feeling (but will investigate more with the users) is the latter, that it's an ONBATT misreported as an OFF by the driver. I was expecting an ONBATT somewhere in the logs, the OL CHRG OFF also caught me off guard there. It didn't seem to go ONBATT for the users, as this would've been reported both in UNRAID (via notification services) and - I think - upsmon in the SYSLOG unless the switch between OL, ONBATT and back was so short that it wasn't picked up by both of them? Do take this with a grain of salt as it's just guessing on my part, I'll see if I can get in touch with the users.

desertwitch, 2023-10-12:
Jim, I managed to get a level 6 driver log for you - no CAL seen, reporting OL OFF on 2.8.0 stable (without FSD).
The affected user did the following resulting in that log: _"start with level 6 debug log > manual selftest  > NUT stop. "_ 

[Back-UPS_RS_900MI.txt](https://github.com/networkupstools/nut/files/12886428/Back-UPS_RS_900MI.txt)

Here's a few lines from that log regarding the status changes:
```
Oct 12 21:04:11 UNRAID-SERVER usbhid-ups[9578]: [D6] send_to_all: write 28 bytes to socket 16 succeeded (ret=28): SETINFO ups.status "OL OFF"
Oct 12 21:04:21 UNRAID-SERVER usbhid-ups[9578]: [D5] send_to_all: SETINFO ups.status "OL DISCHRG"
Oct 12 21:04:31 UNRAID-SERVER usbhid-ups[9578]: [D5] send_to_all: SETINFO ups.status "OL CHRG"
Oct 12 21:04:49 UNRAID-SERVER usbhid-ups[9578]: [D5] send_to_all: SETINFO ups.test.result "Done and passed"
```

Hope that helps, Rysz

desertwitch, 2023-10-12:
The user just sent me another one with driver + upsmon on level 6: [Back-UPS_RS_900MI-upsmon.txt](https://github.com/networkupstools/nut/files/12886568/Back-UPS_RS_900MI-upsmon.txt)

jimklimov, 2023-10-15:
The problem is, an administratively-OFF UPS may still be on-line (getting wall power)... I wonder if an UPS being calibrated would, in some implementations, show up as "OL OB" (getting wall power, but chose to disconnect from it to time the draining of battery under load).

jimklimov, 2023-10-15:
@nbriggs : would you have a moment to look at the reports attached above to this issue? Maybe the APC UPS does say it is calibrating, just we don't interpret the relevant bits?

In particular, this looks fishy and repetitive in the debug log:
````
Oct 12 21:05:39 UNRAID-SERVER usbhid-ups[9578]: [D1] upsdrv_updateinfo...
Oct 12 21:05:39 UNRAID-SERVER usbhid-ups[9578]: [D2] file_report_buffer: expected 4 bytes, but got 512 instead
Oct 12 21:05:39 UNRAID-SERVER usbhid-ups[9578]: [D3] Report[err]: (512 bytes) => 06 01 00 08 00 00 00 00 ff 26 41 00 00 00 00 00
Oct 12 21:05:39 UNRAID-SERVER usbhid-ups[9578]: [D3]  90 cf 79 ce fc 7f 00 00 36 00 fa dd 5a 14 00 00 60 cd 79 ce fc 7f 00 00 00
Oct 12 21:05:39 UNRAID-SERVER usbhid-ups[9578]: [D3]  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 c0 0a
Oct 12 21:05:39 UNRAID-SERVER usbhid-ups[9578]: [D3]  44 00 00 00 00 00 10 00 00 00 00 00 00 00 00 ce 79 ce fc 7f 00 00 ff ff ff
Oct 12 21:05:39 UNRAID-SERVER usbhid-ups[9578]: [D3]  ff 00 00 00 00 60 d1 79 ce fc 7f 00 00 00 00 00 00 00 00 00 00 80 62 08 de
Oct 12 21:05:39 UNRAID-SERVER usbhid-ups[9578]: [D3]  5a 14 00 00 35 fc 40 00 00 00 00 00 bf 0b 44 00 00 00 00 00 01 00 00 00 00
Oct 12 21:05:39 UNRAID-SERVER usbhid-ups[9578]: [D3]  00 00 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Oct 12 21:05:39 UNRAID-SERVER usbhid-ups[9578]: [D3]  00 00 00 00 00 00 00 00 00 00 3b 77 41 00 00 00 00 00 08 00 00 00 00 00 00
Oct 12 21:05:39 UNRAID-SERVER usbhid-ups[9578]: [D3]  00 81 43 28 65 00 00 00 00 97 b7 0d 00 00 00 00 00 00 00 00 00 00 00 00 00
Oct 12 21:05:39 UNRAID-SERVER usbhid-ups[9578]: [D3]  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Oct 12 21:05:39 UNRAID-SERVER usbhid-ups[9578]: [D3]  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Oct 12 21:05:39 UNRAID-SERVER usbhid-ups[9578]: [D3]  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Oct 12 21:05:39 UNRAID-SERVER usbhid-ups[9578]: [D3]  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Oct 12 21:05:39 UNRAID-SERVER usbhid-ups[9578]: [D3]  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ff 00 00 00 00
Oct 12 21:05:39 UNRAID-SERVER usbhid-ups[9578]: [D3]  00 00 00 e0 f7 42 00 00 00 00 00 1d 05 00 00 00 00 00 00 d1 74 41 00 00 00
Oct 12 21:05:39 UNRAID-SERVER usbhid-ups[9578]: [D3]  00 00 7c 27 43 0a 00 00 00 00 18 00 00 00 00 00 00 00 60 cd 79 ce fc 7f 00
Oct 12 21:05:39 UNRAID-SERVER usbhid-ups[9578]: [D3]  00 00 00 00 00 00 00 00 00 32 5d 20 50 61 74 68 3a 00 00 00 00 00 00 20 40
Oct 12 21:05:39 UNRAID-SERVER usbhid-ups[9578]: [D3]  1d 05 00 00 00 00 00 00 01 00 00 00 00 00 00 00 1d 05 00 00 00 00 00 00 02
Oct 12 21:05:39 UNRAID-SERVER usbhid-ups[9578]: [D3]  00 00 00 00 00 00 00 b8 ea 42 00 00 00 00 00 88 25 41 00 00 00 00 00 6d 69
Oct 12 21:05:39 UNRAID-SERVER usbhid-ups[9578]: [D3]  74 2c 08 00 00 00 90 cf 79 ce fc 7f 00 00 98 74 41 00 00 00 00 00 ef b5 41
Oct 12 21:05:39 UNRAID-SERVER usbhid-ups[9578]: [D3]  00 00 00 00 00 0c 00 00 00 00 00 00 00 08 00 00 00 00 00 00 00
````

jimklimov, 2023-10-15:
As for the calibration start (I suppose), there are several lines it reports the OFF status at, like this:
````
Oct 12 21:04:11 UNRAID-SERVER usbhid-ups[9578]: [D2] Path: UPS.PowerSummary.APCStatusFlag, Type: Feature, ReportID: 0x06, Offset: 16, Size: 8, Value: 0
Oct 12 21:04:11 UNRAID-SERVER usbhid-ups[9578]: [D5] hu_find_infoval: found off (value: 0)
Oct 12 21:04:11 UNRAID-SERVER usbhid-ups[9578]: [D5] process_boolean_info: off
Oct 12 21:04:11 UNRAID-SERVER usbhid-ups[9578]: [D5] send_to_all: SETINFO ups.status "OL OFF"
Oct 12 21:04:11 UNRAID-SERVER usbhid-ups[9578]: [D6] send_to_all: write 28 bytes to socket 16 succeeded (ret=28): SETINFO ups.status "OL OFF"
```

in the stretch between :11, :13, :15, :17 and :19 seconds. Other times it is `!off`.

During these seconds, the UPS is online, not charging nor discharging, not low battery, not shutdown-imminent...

Possibly the calibration-not-interpreted hides here?
````
Oct 12 21:04:17 UNRAID-SERVER usbhid-ups[9578]: [D4] Entering libusb_get_report
Oct 12 21:04:17 UNRAID-SERVER usbhid-ups[9578]: [D3] Report[get]: (2 bytes) => 36 0c
Oct 12 21:04:17 UNRAID-SERVER usbhid-ups[9578]: [D5] PhyMax = 0, PhyMin = 0, LogMax = 13, LogMin = 0
Oct 12 21:04:17 UNRAID-SERVER usbhid-ups[9578]: [D5] Unit = 00000000, UnitExp = 0
Oct 12 21:04:17 UNRAID-SERVER usbhid-ups[9578]: [D5] Exponent = 0
Oct 12 21:04:17 UNRAID-SERVER usbhid-ups[9578]: [D2] Path: UPS.Input.APCLineFailCause, Type: Feature, ReportID: 0x36, Offset: 0, Size: 8, Value: 12
Oct 12 21:04:17 UNRAID-SERVER usbhid-ups[9578]: [D3] hu_find_infoval: no matching INFO_* value for this HID value (12)
Oct 12 21:04:17 UNRAID-SERVER usbhid-ups[9578]: [D5] Lookup [12] failed for [BOOL]
Oct 12 21:04:17 UNRAID-SERVER usbhid-ups[9578]: [D3] Report[buf]: (2 bytes) => 36 0c
Oct 12 21:04:17 UNRAID-SERVER usbhid-ups[9578]: [D5] PhyMax = 0, PhyMin = 0, LogMax = 13, LogMin = 0
Oct 12 21:04:17 UNRAID-SERVER usbhid-ups[9578]: [D5] Unit = 00000000, UnitExp = 0
Oct 12 21:04:17 UNRAID-SERVER usbhid-ups[9578]: [D5] Exponent = 0
Oct 12 21:04:17 UNRAID-SERVER usbhid-ups[9578]: [D2] Path: UPS.Input.APCLineFailCause, Type: Feature, ReportID: 0x36, Offset: 0, Size: 8, Value: 12
Oct 12 21:04:17 UNRAID-SERVER usbhid-ups[9578]: [D3] hu_find_infoval: no matching INFO_* value for this HID value (12)
Oct 12 21:04:17 UNRAID-SERVER usbhid-ups[9578]: [D5] Lookup [12] failed for [BOOL]
````

This one is seen at:
* 21:03:45 (initial walk);
* 21:04:17 (a few cycles after "OFF" flag got raised);
* 21:04:49 (all seems OK, not "OFF")
* 21:05:21 (all seems OK, not "OFF")

So it seems not related to the emergency state processing, in fact is logged as part of an `upsdrv_updateinfo()` "Full Update" walk done every 30 sec here.

jimklimov, 2023-10-15:
Actually just AFTER that stretch of "OFF" the driver suggests it is calibrating:
````
Oct 12 21:04:21 UNRAID-SERVER usbhid-ups[9578]: [D2] Path: UPS.PowerSummary.APCStatusFlag, Type: Feature, ReportID: 0x06, Offset: 16, Size: 8, Value: 16
Oct 12 21:04:21 UNRAID-SERVER usbhid-ups[9578]: [D5] hu_find_infoval: found !off (value: 16)
Oct 12 21:04:21 UNRAID-SERVER usbhid-ups[9578]: [D5] process_boolean_info: !off
Oct 12 21:04:21 UNRAID-SERVER usbhid-ups[9578]: ups_status_set: seems that UPS [qnapups] is in OL+DISCHRG state now. Is it calibrating or do you perhaps want to set 'onlinedischarge' option? Some UPS models (e.g. CyberPower UT series) emit OL+DISCHRG when offline.
Oct 12 21:04:21 UNRAID-SERVER usbhid-ups[9578]: [D5] send_to_all: SETINFO ups.status "OL DISCHRG"
Oct 12 21:04:21 UNRAID-SERVER usbhid-ups[9578]: [D6] send_to_all: write 32 bytes to socket 16 succeeded (ret=32): SETINFO ups.status "OL DISCHRG"
````

So maybe the decision to raise the (administrative?-)OFF flag should be staggered, and come into force e.g. after `pollfreq` elapses if the condition is still there?.. Alternatively, with existing code, the `upssched` could be used for inducing such delays to filter out flukes.

desertwitch, 2023-10-15:
> During these seconds, the UPS is online, not charging nor discharging, not low battery, not shutdown-imminent...

> So it seems not related to the emergency state processing, in fact is logged as part of an `upsdrv_updateinfo()` "Full Update" walk done every 30 sec here.

Yes, but note this was on 2.8.0 stable, with the current implementation in master the system goes FSD the second OFF is raised.

So might indeed be worth waiting if OFF is reported by the driver for more than x-number of refresh cycles before effectively setting it as UPS status, if the current implementation (resulting in FSD on OFF) is to be kept.

jimklimov, 2023-10-15:
Posted PR #2108 as a tentative fix for one aspect of the puzzle - `upsmon` (so the "workaround" for "OFF" state handling is available to any driver). Would you be able to check with that source branch if NUT behaves better this way (and no new bugs appeared regarding shutting down when things do really go critical)?

I intended to also address `usbhid-ups` to add a toggle for "OL+DISCHRG=CAL" but currently ran out of my time slot :)

jimklimov, 2023-10-17:
So the `usbhid-ups` part was addressed in same PR (merged today); issue closure now pending some practical testing with end-user devices, I guess.

desertwitch, 2023-10-22:
Just heard back from the affected user who provided the debug information and it's fixed for him now.

closed 2023-10-22 by jimklimov

nbriggs, 2023-10-22:
@jimklimov - back online after a period of limited connectivity, and I see you've got this one solved.  Good work!