networkupstools/nut #2106: "HCL"  Serioux UPS LCD600LI supported by blazer_usb, langid_fix="0x4095"

ami3go, 2023-10-13:
**Working ups.conf settings:** 

Device link: 
[https://www.serioux.ro/en/product/ups-serioux-line-interactive-600li-ecran-lcd-capacitate-600va-360w-2-prize-schuko/](url)
Connected to ProxMox, based on Debian 
```
[myups]
    driver = "blazer_usb"
    protocol=megatec
    port = auto
    langid_fix="0x4095"
    desc="UPS Serioux LCD600LI"
    vendorid = "0001"
    productid = "0000"
```
when device is connected it report following parameters: 
```
root@HomeVM:~# upsc myups
Init SSL without certificate database
battery.charge: 100
battery.voltage: 13.50
battery.voltage.high: 13.00
battery.voltage.low: 10.40
battery.voltage.nominal: 12.0
device.mfr: 
device.model: 
device.type: ups
driver.name: blazer_usb
driver.parameter.langid_fix: 0x4095
driver.parameter.pollinterval: 15
driver.parameter.port: auto
driver.parameter.productid: 0000
driver.parameter.protocol: megatec
driver.parameter.synchronous: auto
driver.parameter.vendorid: 0001
driver.version: 2.8.0
driver.version.internal: 0.14
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.current.nominal: 0.0
input.frequency: 49.9
input.frequency.nominal: 0
input.voltage: 237.0
input.voltage.fault: 0.0
input.voltage.nominal: 0
output.voltage: 237.0
ups.beeper.status: disabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.firmware: V3.9
ups.load: 33
ups.mfr: 
ups.model: 
ups.productid: 0000
ups.status: OL
ups.temperature: 29.0
ups.type: offline / line interactive
ups.vendorid: 0001
```

ami3go, 2023-10-13:
```
root@HomeVM:~# upscmd -l myups
Instant commands supported on UPS [myups]:

beeper.toggle - Toggle the UPS beeper
load.off - Turn off the load immediately
load.on - Turn on the load immediately
shutdown.return - Turn off the load and return when power is back
shutdown.stayoff - Turn off the load and remain off
shutdown.stop - Stop a shutdown in progress
test.battery.start - Start a battery test
test.battery.start.deep - Start a deep battery test
test.battery.start.quick - Start a quick battery test
test.battery.stop - Stop the battery test
```

CristiXP, 2024-03-16:
I have the same UPS, but I cannot connect to it with blazer_usb. Can anyone assist?

[myups...
[13:09:00] INFO: Starting the UPS drivers...
Please note that this driver is deprecated and will not receive
new development. If it works for managing your devices - fine,
but if you are running it to try setting up a new device, please
consider the newer nutdrv_qx instead, which should handle all 'Qx'
protocol variants for NUT. (Please also report if your device works
with this driver, but nutdrv_qx would not actually support it with
any subdriver!)

No supported UPS detected
Network UPS Tools - Megatec/Q1 protocol USB driver 0.14 (2.8.0)
Network UPS Tools - UPS driver controller 2.8.0
Driver failed to start (exit status=1)
cont-init: info: /etc/cont-init.d/nut.sh exited 1
cont-init: info: running /etc/cont-init.d/nutclient.sh
cont-init: info: /etc/cont-init.d/nutclient.sh exited 0
cont-init: warning: some scripts exited nonzero
s6-rc: warning: unable to start service legacy-cont-init: command exited 1
/run/s6/basedir/scripts/rc.init: warning: s6-rc failed to properly bring all the services up! Check your logs (in /run/uncaught-logs/current if you have in-container logging) for more information.]]())

jimklimov, 2024-03-17:
@CristiXP : is your "same" UPS branded the same, or just has same IDs? As you can guess, 0000 and 0001 are not registered numbers delegated to a particular vendor for device models talking a particular protocol.

FWIW, NUT has a driver `nutdrv_atcl_usb` for those ID numbers; maybe your device is among the few that fit it?..

See wiki about bumping `debug_min` setting in a driver configuration. Not sure if HA passes that through though; maybe you'd have to investigate with NUT on the host OS first (see wiki about building for "in-place" testing, no obligation to install that build).

CristiXP, 2024-03-17:
@jimklimov it is exactly the same one as stated above. Also on the computer, with the UPSmart app on USB it works. I tried also in a virtual machine on a computer, and still doesn't work. On lsusb it shows the 0001 0000 connected.

It doesn't work with your suggested driver also... don't know how to try other things to test or debug more :(

ami3go, 2024-04-11:
Hello. 
I have tested on ProxMox Debian. 
My main problem was that something takes the connection you I need to reboot to free up connection. Try to reboot and then connect again

ami3go, 2024-04-22:
also tested with blazer_usb, as support for blazer_usb will finish soon. 
Name "ups" is for compatibility with Synology NAS
```
[ups]           
   driver = "nutdrv_qx"
   port = auto
   langid_fix="0x4095"
   desc="UPS Serioux LCD600LI"
   vendorid = 0001
   productid = 0000

```
same result
```
root@HomeVM:~# upsc ups
Init SSL without certificate database
battery.charge: 96
battery.voltage: 12.90
battery.voltage.high: 13.00
battery.voltage.low: 10.40
battery.voltage.nominal: 12.0
device.type: ups
driver.name: nutdrv_qx
driver.parameter.langid_fix: 0x4095
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 15
driver.parameter.port: auto
driver.parameter.productid: 0000
driver.parameter.synchronous: auto
driver.parameter.vendorid: 0001
driver.version: 2.8.0
driver.version.data: Megatec 0.06
driver.version.internal: 0.32
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.current.nominal: 0.0
input.frequency: 50.0
input.frequency.nominal: 0
input.voltage: 234.0
input.voltage.fault: 0.0
input.voltage.nominal: 0
output.voltage: 235.0
ups.beeper.status: disabled
ups.delay.shutdown: 30
ups.delay.start: 600
ups.firmware: V3.9
ups.load: 35
ups.productid: 0000
ups.status: OL
ups.temperature: 29.0
ups.type: offline / line interactive
ups.vendorid: 0001
```

```
root@HomeVM:~# upscmd -l ups
Instant commands supported on UPS [ups]:

beeper.toggle - Toggle the UPS beeper
load.off - Turn off the load immediately
load.on - Turn on the load immediately
shutdown.return - Turn off the load and return when power is back
shutdown.stayoff - Turn off the load and remain off
shutdown.stop - Stop a shutdown in progress
test.battery.start - Start a battery test
test.battery.start.deep - Start a deep battery test
test.battery.start.quick - Start a quick battery test
test.battery.stop - Stop the battery test
```
It able to shutdown my proxmox served connected via USB directly not PC, and also Synology NAS connected via LAN