networkupstools/nut #2110: Turn off the fritzboz router

tonygh77, 2023-10-16:
Is it possible somehow? 
Thank you for your time

jimklimov, 2023-10-16:
First time I hear about these (googling brought up "[FRITZ!Box](https://en.avm.de/products/fritzbox/)") :)

If they run Linux (e.g. OpenWRT, DD-WRT, Tomato, some vendors like Asus base off those projects out of the box), it could probably be possible to build NUT (`upsmon`) for that platform. It could be compromised however by the little space available on the router device itself and in its RAM. So another option could be to SSH/Telnet into the device to call `shutdown` there, as part of a `SHUTDOWNCMD` script that you would configure on a more capable system nearby which monitors the UPS (directly, also running the NUT drivers and connected by cable to the power device and `upsd` data server, or indirectly by being a NUT client to another system that has the driver and data server)

tonygh77, 2023-10-16:
I did some research online
Maybe I could succeed in SSH with special firmware
A thousand thanks