networkupstools/nut #2119: apc_modbus: Add support for commands

EchterAgo, 2023-10-18:
This is for after merging #2063 .

We need to add support for UPS commands. This is what we can add:

- UPSCommand_BF
  - RestoreFactorySettings

- OutletCommand_BF
  - Cancel
  - OutputOn
  - OutputOff
  - OutputShutdown
  - OutputReboot

- SimpleSignalingCommand_BF
  - RequestShutdown
  - RemoteOff
  - RemoteOn

- ReplaceBatteryTestCommand_BF (incorrectly named UPSStatus_BF in docs)
  - start test

- RunTimeCalibrationCommand_BF
  - start calibration

- UserInterfaceCommand_BF
  - mute
  - cancel mute
  - acknowledge battery alarm

  - args:
    - ColdBootAllowed
    - UseOnDelay
    - UseOffDelay

  - target:
    - UnswitchedOutletGroup
    - SwitchedOutletGroup0
    - SwitchedOutletGroup1
    - SwitchedOutletGroup2

EchterAgo, 2023-10-18:
There are also quite a few values that can not only be read but also set. Things like the shutdown delay.

EchterAgo, 2023-10-20:
We also should add a `SIZEOF_ARRAY` helper macro to simplify the `sizeof(x)/sizeof(x[0])` situation, see #2063

jimklimov, 2023-10-21:
@EchterAgo : do you expect to proceed with this immediately/soon? As in, should we delay NUT v2.8.1 release in anticipation of this addition (so as to publish it with a fully functional `apc_modbus` driver, at least to the extent that systems shutdown can be set up)?

EchterAgo, 2023-10-21:
Sorry, but it will have to wait a bit, I currently have some other things I need to get sorted out first, but I expect to have this working sometime in November.

EchterAgo, 2023-10-21:
Although maybe if you can tell me what you'd consider the minimum amount of support and which commands that would be, I might be able to do a quick implementation of these. I'm really new to actually using commands / writing values in nut, I have only used it for monitoring so far.

I'm not sure how I'd actually do system shutdown from the UPS, do you mean just turning the output off? Load off / cycling is something that APC supports for what it calls outlet groups.

EchterAgo, 2023-10-21:
Implementing the commands should actually be just a simple register write. I think for power on/off/reboot we just need to write the right registers. We might need only `OutletCommand_BF` and `SimpleSignalingCommand_BF`. The documentation for these are here if anyone wants to check: https://download.schneider-electric.com/files?p_enDocType=Application+Notes&p_File_Name=MPAO-98KJ7F_R1_EN.pdf&p_Doc_Ref=SPD_MPAO-98KJ7F_EN

I'd need some advice what to write for which command, then I think I could do an implementation quickly.

EchterAgo, 2023-10-21:
I just checked and for example `modbus_write_register(modbus_ctx, 1541, 1);` (`ReplaceBatteryTestCommand_BF`) works to start a battery test. I just need to know when to call which function, implement some error handling and maybe things also need some status polling (for example `RunTimeCalibrationCommand_BF` has a `RunTimeCalibrationStatus_BF`).

jimklimov, 2023-10-21:
As for the "bare minimum" - well, monitoring the device to the point that we know it is discharging and in a low-battery state is already quite a lot :)

Actual shutdown routines usually end with a command to the device to turn off the load or completely the UPS (or ideally - power-cycle if the wall power came back while we were shutting down, possibly turn on after getting recharged to a safe level that ensures another graceful shutdown if needed). These relate to several commands and variables about "load.*" and "shutdown.*" (see `docs/nut-names.txt`) , further depending on what this or that hardware/firmware actually allows to do.

Numerous other things like setting delays for these power operations, starting calibration, managing separate outlet groups, popularly the beeper state, etc. are useful cherries on the cake - welcome generally but might wait if not a low-hanging fruit.

EchterAgo, 2023-10-21:
Thanks for the information Jim. I'll try to look into this tomorrow (23rd, it is the 22nd already here). @PrplHaze4, do you mind sharing what commands you consider most important and how you'd expect them to work?

EchterAgo, 2023-10-21:
Also @PrplHaz4, I'd like some feedback on increasing the timeout setting I added called `response_timeout_ms` (see #2063). I think your timeout issue will likely be resolved or at least much better by setting it to 1000 or higher. You can just set the option in `ups.conf`. If it does not improve things, you might want to make an issue and make sure to ping me by mentioning `@EchterAgo`.

@jimklimov If you can, please also mention me if there are any issues with the `apc_modbus` driver. I'd like to fix whatever comes up.

jimklimov, 2023-10-21:
Sure thing, thanks!

jimklimov, 2023-10-22:
@EchterAgo : I believe in one of these new driver threads you've asked about reconnection - and whether the exact matcher should be used?.. A discussion in #1843 suggested that when devices go AWOL, the USB stack can re-enumerate them - so assigning a new "device" number at least (and an old exact matcher would no longer fit). Probably at least making sure we do fall back to same regex matcher as used for the initial connection, if the cached exact one did not work for a re-connect, is a good idea after all :)

PrplHaz4, 2023-10-26:
> Also @PrplHaz4, I'd like some feedback on increasing the timeout setting I added called `response_timeout_ms` (see #2063). I think your timeout issue will likely be resolved or at least much better [...]

@EchterAgo Sorry for taking so long to look at this.  I let it run for a while with `response_timeout_ms = 1000` and this was the only timeout I saw, and it seems to have recovered quickly and just kept on going, so nice work!  I'll be sure to open a new issue if I see anything that stands out.

```
134.399286     [D5] send_to_all: SETINFO driver.state "quiet"
 135.385731     [D5] send_to_all: SETINFO driver.state "updateinfo"
 135.997512     [D5] send_to_all: SETINFO battery.runtime "16196"
 135.997562     [D5] send_to_all: SETINFO battery.voltage "54.59"
 135.997571     [D5] send_to_all: SETINFO battery.temperature "28.83"
 135.997575     [D5] send_to_all: SETINFO ups.load "15.48"
 135.997578     [D5] send_to_all: SETINFO ups.realpower "185.77"
 135.997581     [D5] send_to_all: SETINFO ups.power "196.88"
 135.997604     [D5] send_to_all: SETINFO output.current "1.62"
 135.997612     [D5] send_to_all: SETINFO input.voltage "120.59"
 136.368536     [D5] send_to_all: SETINFO driver.state "quiet"
 137.386940     [D5] send_to_all: SETINFO driver.state "updateinfo"
 138.669650     _apc_modbus_read_registers: Read of 128:160 failed: Connection timed out (192.168.4.27:502)
 139.669994     [D5] send_to_all: DATASTALE
 139.670054     [D5] send_to_all: SETINFO driver.state "quiet"
 139.670066     [D5] send_to_all: SETINFO driver.state "updateinfo"
 140.320091     [D5] send_to_all: SETINFO battery.runtime "17148"
 140.320134     [D5] send_to_all: SETINFO ups.load "13.10"
 140.320137     [D5] send_to_all: SETINFO ups.realpower "157.17"
 140.320139     [D5] send_to_all: SETINFO ups.power "172.41"
 140.320157     [D5] send_to_all: SETINFO output.current "1.41"
 140.320162     [D5] send_to_all: SETINFO ups.efficiency "92.8"
 140.683053     [D5] send_to_all: DATAOK
 140.683093     [D5] send_to_all: SETINFO driver.state "quiet"
 141.671352     [D5] send_to_all: SETINFO driver.state "updateinfo"
```

EchterAgo, 2023-10-26:
@PrplHaz4 In that case you might want to further increase the timeout, the TCP connection is clearly still working, maybe try 1500.

jimklimov, 2023-11-12:
@EchterAgo : cheers, just inquiring if you had (or will soon have) time to look at this?
It seems I would be releasing NUT 2.8.2 with some follow-up fixes for 2.8.1 sooner rather than later - should I hold the door open here for `upscmd` and `upsrw` via `apc_modbus`, or is it likely to follow up just later?

EchterAgo, 2023-11-13:
Yea, I'm still working on things, but I can't promise when I have an update, I'm busy with work. I do hope to have something soon.

Right now I'm still working on variable writes, I can already set some. I also did some research about APCs outlet groups and how to properly use them.

I also did some work on commands, I have a working `upsdrv_shutdown` and am implementing an `instcmd` handler.

EchterAgo, 2023-11-13:
I'm also still investigating some weird communication issues I have sometimes since I started reading more data, but that is likely a change in `libmodbus`. I currently don't have my equipment to capture USB/RS232 traffic, so that will also need to wait until maybe next week.

jimklimov, 2023-11-13:
Thanks for the updates, and for the continued effort :)

EchterAgo, 2023-11-13:
I just put in some effort and I actually already have most commands implemented, just need to do some testing. Currently I can only mute the beeper, not toggle/enable/disable. I think I might be able to add toggle.

For testing this I'd need to have a way to send commands to the driver when running in the debugger. I'm running the driver standalone right now, I guess I'll have to look into starting `upsd` or is there an easier way for testing? Actually, I think I can just call the `instcmd` function in gdb.

EchterAgo, 2023-11-13:
Is there a difference between `load.off.delay` and `shutdown.stayoff`?

jimklimov, 2023-11-13:
> a way to send commands to the driver when running in the debugger

jimklimov, 2023-11-13:
> Is there a difference between load.off.delay and shutdown.stayoff?

https://github.com/networkupstools/nut/blob/master/docs/nut-names.txt#L774-L785 :)

EchterAgo, 2023-11-13:
> > Is there a difference between load.off.delay and shutdown.stayoff?
> 
> https://github.com/networkupstools/nut/blob/master/docs/nut-names.txt#L774-L785 :)

I know the doc. Let me rephrase. Isn't one of `load.off.delay`, `shutdown.return` and `shutdown.stayoff` redundant? `load.off.delay` is the same as the two `shutdown`s, but `load.off.delay` does not specify whether to return.

jimklimov, 2023-11-13:
Hm, if you put it this way... I think this historically depends on hardware support. Not all UPSes allow the granularity of explicitly returning or not after power is back. Having a delay support also varies between devices. I suppose the forefathers left the differentiation in place so end-users would not be mis-led by a single command that never does what people want of it on average (like, "I asked to wait 30 sec and power off, but it went down instantly") :) That said, there are fallbacks in some UPS shutdown methods to try what was requested, and if it is still alive - go over other variants.

Wondering if there was also a connotation of turning off the UPS as a device completely (so it can be turned on by a button or returning power, but not via protocol) vs. "just" turning off its load (effectively outlet groups, maybe the inverter too) while the UPS controller remains manageable.

EchterAgo, 2023-11-13:
Turning the device is something the APC units can do, but also the description says "the load".

EchterAgo, 2023-11-13:
I'll also have to figure out if I can reduce or take out the `usleep(100000);` in `setvar`. I seem to be getting very persistent connection problems even after restarting when I don't have that. Maybe I'll need to poll something, or maybe this is specified somewhere in APCs Modbus documentation, I'll have to check.

EchterAgo, 2023-11-13:
Some of the commands won't work, I forgot to account for commands that take multiple 16-bit registers.

EchterAgo, 2023-11-13:
We also need to use `dstate_addcmd` / `dstate_setflags`

EchterAgo, 2023-11-14:
EchterAgo@e49fe7e0fe40f9c149ec77096157adc2c038075d

Pushed an update, make a command table. I tested the `test.battery.*`, `test.panel.start`, `calibrate.*` and `beeper.mute` commands. `test.battery.stop` for some reason returns an error on my device even though we send the correct command, maybe it is not supported?

It would be great if someone could test the `load` / `shutdown` commands as I can not currently turn off my UPSs load.

I will continue working on `setvar` now.

EchterAgo, 2023-11-14:
EchterAgo@8c50ac205500c2aefa168656334f7d1d0950e293

- Removed incorrect RW flag from `ups.timer.reboot`, the manual is incorrect here and it does not even make sense to have this RW.
- Outlet groups should be working properly now, no outlet group specific commands yet. MOG (Main outlet group) that switches everything, including the other outlet groups is `outlet.group.0`, SOG0 (labeled as Group 1 on the device) is `outlet.group.1`, SOG1 is `outlet.group.2`, etc...
- Made most of the static map RW, you can now configure transfer voltages and delays for each outlet group.
- Added `outlet.group.count`

Missed from last update:

- Correctly set `ST_FLAG_STRING`, `ST_FLAG_RW` and use `dstate_setaux` to set length for writable strings.
- Use `dstate_addcmd` to register all the commands we support.

EchterAgo, 2023-11-14:
@jimklimov 

we also have a duplicate description for `outlet.n.shutdown.return`:

```
| outlet.n.shutdown.return | Turn off the outlet possibly after a delay
                             and return when power is back
...
| outlet.n.shutdown.return | Turn off the outlet and return when power is back
```

EchterAgo, 2023-11-14:
EchterAgo@7def65995800715a96aca9d09d12e93c80e2fc36

- Added commands for `outlet.x.*`, this currently does not control individual outlets but outlet groups. `nut-names.txt` does not list commands for `outlet.group.x.*` so we use this instead. Should we change this to `outlet.group` ?
- Added warning about SOG3 detection. This is part of the relay config. AFAIK there are no models with 4 SOGs and all the rest of the values / commands only list SOG0-SOG2.

EchterAgo, 2023-11-14:
EchterAgo@37504d11756d61ebe33ced8b9fea40b43802c8c6

Fixed some warnings

EchterAgo, 2023-11-14:
Created PR, see #2184

closed 2024-03-02 by jimklimov