networkupstools/nut #2121: After months of no problems, CyberPower UPS  won't stay connected -"failed - Data stale"

Jsalas424, 2023-10-19:
NUT was working perfectly for years until yesterday.

[image]

[image]

```
Oct 17 13:23:22 Server upsd[1039]: Data for UPS [Power] is stale - check driver
Oct 17 13:23:22 Server nut_exporter[802]: time="2023-10-17T13:23:22-04:00" level=error msg="Failure getting the list of UPS devices: upsd is connected to the driver for the UPS, but that driver isn’t providing regular updates or has specifically marked the data as stale. upsd refuses to provide variables on stale units to avoid false readings. This generally means that the driver is running, but it has lost communications with the hardware. Check the physical connection to the equipment" source="nut_collector.go:70"
```

I can still see the UPS listed:
```
root@Server:~# lsusb
Bus 002 Device 004: ID 0764:0501 Cyber Power System, Inc. CP1500 AVR UPS
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

and if I run it directly, no problems

```
root@Server:~# /lib/nut/usbhid-ups -a Power
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
Duplicate driver instance detected! Terminating other driver!
Using subdriver: CyberPower HID 0.4
                                                                               
Broadcast message from nut@Server (somewhere) (Wed Oct 18 21:20:03 2023):  
                                                                               
Communications with UPS Power@10.0.0.3 established                       
```

I have nut installed directly onto my Proxmox hypervisor. No changes have been made to the hypervisor between when it was last working and now, haven't even updated, nor any changes to the nut config file. Thoughts?

Jsalas424, 2023-10-19:
After looking around some more, I think this thread applies: https://github.com/networkupstools/nut/issues/2017

I found `Parse error on sock: Invalid ctx buffer` while digging through my logs

Will report back

jimklimov, 2023-10-19:
Good luck! Check also the wiki-links (and other sites they lead to) posted in that issue. At the moment I've got nothing to add :}

Jsalas424, 2023-10-30:
I upgraded my Hypervisor to Debian Bookworm which ships with NUT v2.8. Full symptomatic relief: 
[image]

Will hopefully be able to close out ticket #2001 once that system gets updated as well.

closed 2023-10-30 by Jsalas424