networkupstools/nut #2122: apc_modbus: truncation compiler warning

desertwitch, 2023-10-21:
Good morning!

@EchterAgo & @jimklimov & others :
**THANK YOU** for the immense efforts with `apc_modbus` - great to have this included in NUT.

Building on my Slackware 15.0/gcc11.2.0 environment:
```
In function â€˜_apc_modbus_parse_host_portâ€™,
    inlined from â€˜upsdrv_initupsâ€™ at apc_modbus.c:1246:8:
apc_modbus.c:1137:17: warning: â€˜strncpyâ€™ specified bound 256 equals destination size [-Wstringop-truncation]
 1137 |                 strncpy(host, start, host_buf_size);
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
apc_modbus.c: At top level:
cc1: note: unrecognized command-line option â€˜-Wno-unknown-warning-optionâ€™ may have been intended to silence earlier diagnostics
cc1: note: unrecognized command-line option â€˜-Wno-reserved-identifierâ€™ may have been intended to silence earlier diagnostics
```

EchterAgo, 2023-10-21:
Looks like we should pass `host_buf_size - 1` to account for the NULL character, otherwise we won't get zero termination if the string is too large.

EchterAgo, 2023-10-21:
Or maybe we should just use `snprintf`, which returns an error, instead

desertwitch, 2023-10-21:
Builds without warning if I patch this in, but I'm not a C expert (so not sure if this would have fallout):
```
diff --git a/drivers/apc_modbus.c b/drivers/apc_modbus.c
index dc401ca6c..f5a2f1f48 100644
--- a/drivers/apc_modbus.c
+++ b/drivers/apc_modbus.c
@@ -1134,7 +1134,7 @@ static int _apc_modbus_parse_host_port(const char *input, char *host, size_t hos

        if (!end) {
                /* Port is missing, use the default port */
-               strncpy(host, start, host_buf_size);
+               snprintf(host, host_buf_size, "%s", start);
                snprintf(port, port_buf_size, "%u", default_port);
                return 0;
        }

```

EchterAgo, 2023-10-21:
I also made it use `snprintf` in #2123, along with some other string safety changes. Thanks for the report!

jimklimov, 2023-10-21:
Maybe you can't even imagine how pleasant it is to see issues discovered and solved without my involvement, after a few years of feeling like being the main spinning cog for everything :)

jimklimov, 2023-10-21:
@desertwitch : by the way, can you please elaborate on how you build NUT to see those warnings? Do you use plain `./autogen.sh && ./configure && make` or some specific `configure` arguments, the `ci_build.sh` script features, etc.?

These and other warnings you've recently posted seem like something that should have popped up in CI, similar gcc versions are in use after all, so wondering what we missed... does that build of GCC have some different defaults for `-Wall` etc?...

EchterAgo, 2023-10-21:
@jimklimov maybe if you explicitly add `-Wstringop-truncation` for gcc 11?

desertwitch, 2023-10-21:
> @desertwitch : by the way, can you please elaborate on how you build NUT to see those warnings? Do you use plain `./autogen.sh && ./configure && make` or some specific `configure` arguments, the `ci_build.sh` script features, etc.?
> 
> These and other warnings you've recently posted seem like something that should have popped up in CI, similar gcc versions are in use after all, so wondering what we missed... does that build of GCC have some different defaults for `-Wall` etc?...
> 
> (There is also some ongoing bug-fix for the CI jenkins-dynamatrix library setup which caused it to not try all the compilers it is supposed to loop against, for a couple of months since August or so; supposed to have got fixed during past weeks but maybe that is still related too)

Sure - here's how I build NUT on my Slackware 15.0 with gcc 11.2.0:
```
./autogen.sh
./configure --build="$ARCH-slackware-linux" \
            --host="$ARCH-slackware-linux" \
            --prefix=/usr \
            --libdir=/usr/lib$LIBDIRSUFFIX \
            --sysconfdir=/etc/nut \
            --localstatedir=/var \
            --datadir=/usr/share/nut \
            --mandir=/usr/man \
            --enable-shared=yes \
            --enable-static=no \
            --enable-strip \
            --with-all \
            --with-linux-i2c=no \
            --with-gpio=no \
            --with-docs=no \
            --with-ssl \
            --with-user=root \
            --with-group=root \
            --with-cgipath=/var/www/nut-cgi-bin \
            --with-htmlpath=/var/www/nut \
            --with-drvpath=/usr/libexec/nut \
            --with-statepath=/var/run/nut \
            --with-pidpath=/var/run/nut \
            --with-altpidpath=/var/run/nut || exit 4

# Build & install
make || exit 5
make DESTDIR=$PKG install || exit 6
```

```
NUT Compiler settings:
----------------------

* CC      	: x86_64-slackware-linux-gcc
* CFLAGS  	: -isystem /usr/local/include -g -O2 -Wno-reserved-identifier -Wno-unknown-warning-option -std=gnu99 -s -Wno-system-headers -Wall -Wextra -Wsign-compare -pedantic -Wno-error
* CXX     	: x86_64-slackware-linux-g++
* CXXFLAGS	: -isystem /usr/local/include -g -O2 -Wno-reserved-identifier -Wno-unknown-warning-option -std=gnu++11 -Wno-system-headers -Wall -Wextra -Wno-error
* CPP     	: x86_64-slackware-linux-gcc -E
* CPPFLAGS	: 
```

PR #2123 (2023-10-21): apc_modbus: Use snprintf instead of strncpy
`strncpy` does not necessarily zero terminate the buffer so we replace it with `snprintf` and add error checking.

This also fixes an off-by-one in the `snprintf` error checking of the `_apc_modbus_double_to_nut` and `_apc_modbus_power_to_nut` functions and adds error checking to every `snprintf` call.

fixes #2122 

## General points

- [x] Described the changes in the PR submission or a separate issue, e.g.
  known published or discovered protocols, applicable hardware (expected
  compatible and actually tested/developed against), limitations, etc. [... cut]
closed 2023-10-21 by jimklimov