networkupstools/nut #2125: [HCL] Exegate SpecialPro Smart

dsenash, 2023-10-22:
Hi,
Which driver need to use for:
ID 0925:1234 Lakeview Research UPS USB Mon V2.0
iManufacturer           1 RICHCOMM
iProduct                2 UPS USB Mon V2.0

Driver richcomm_usb doesn't work. Driver failed to start (exit status=1)
Looks like this driver is for UPS USB MON 1.4
Which driver need to use with UPS USB Mon V2.0?
Thanks.

dsenash, 2023-10-22:
USB device [0925:1234] matches, but driver callback failed: could not clear/halt ep 129: No such file or directory
Driver failed to start (exit status=1)
Network UPS Tools - Richcomm dry-contact to USB driver 0.04 (2.7.4)

jimklimov, 2023-10-23:
Try `nutdrv_qx`, this ID recently came up a lot with `armac` subdriver that got many updates in the NUT master branch, so you might benefit from a custom build while 2.8.1 is not released yet: https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests

At the very least, you can run newest drivers right from the build workspace to test if they support the device, without necessarily changing the installed NUT files until you're satisfied with the new ones.

dsenash, 2023-10-23:
nut-scanner showed me the following:
[nutdev1]
	driver = "richcomm_usb"
	port = "auto"
	vendorid = "0925"
	productid = "1234"
	product = "UPS USB Mon V2.0"
	vendor = "RICHCOMM"
	bus = "007"

Looks like driver richcomm_usb is valid.
But as soon as I start upsdrvctl I am getting:
root@orange5:/tmp/nut/drivers# upsdrvctl start
Network UPS Tools - UPS driver controller 2.7.4
Network UPS Tools - Richcomm dry-contact to USB driver 0.04 (2.7.4)
Warning: This is an experimental driver.
Some features may not function correctly.

USB device [0925:1234] matches, but driver callback failed: could not clear/halt ep 129: No such file or directory
Driver failed to start (exit status=1)

Is any ideas how to fix this?
Rg, Dmitry

jimklimov, 2023-10-23:
Well, `nut-scanner` discovery is mostly based on identifiers; just for a few driver types can it dig deeper. Many USB-connected vendors use same chips with unrelated protocols (some with bogus unlicensed IDs like 0000, 0001, ffff). Some vendors seem to own or license their IDs but put out unrelated products to the market under the same brand and identification.

I believe `richcomm_usb` describes itself as a "dry-contact to USB solution". If your UPS talks an actual protocol (likely Qx) then a driver like `nutdrv_qx` should be the first choice - it is supposed to include and override all previously made separate Qx driver efforts.

Also note that NUT v2.7.4 is about 7.5 years old now, so newer code may behave better...

DIMMonchik, 2024-04-18:
Has anyone found a suitable driver for this UPS yet?

jimklimov, 2024-04-18:
Hello. Which drivers did you try, and with which NUT version?

DIMMonchik, 2024-04-18:
> Hello. Which drivers did you try, and with which NUT version?

Network UPS Tools - 2.8.0

arsonx, 2024-07-01:
Hello. I have the same issue trying to setup Nut for Armac UPS (ARMAC UPS HOME 650E LED V2). I've compiled nut-2.8.2 on Debian 12. I was trying both richcomm_usb and nutdrv_qx.

**nutdrv_qx**
```
drivers/nutdrv_qx -a nutdrv -D

Network UPS Tools - Generic Q* USB/Serial driver 0.36 (2.8.2)
USB communication driver (libusb 0.1) 0.47
   0.000000     [D1] upsdrv_makevartable...
   0.000156     [D1] Using USB implementation: libusb-0.1 (or compat)
   0.000509     [D1] Network UPS Tools version 2.8.2 (release/snapshot of 2.8.2) built with gcc (Debian 12.2.0-14) 12.2.0 and configured with flags: --with-drivers=richcomm_usb,nutdrv_qx --with-statepath=/var/run/nut --with-pidpath=/var/run/nut --with-user=root --with-group=root --sysconfdir=/etc/nut
   0.000617     [D1] debug level is '1'
   0.002873     [D1] Succeeded to become_user(root): now UID=0 GID=0
   0.002969     [D1] upsdrv_initups...
   0.010568     [D1] interrupt read error: error submitting URB: No such file or directory (-2)
   0.017408     [D1] interrupt read error: error submitting URB: No such file or directory (-2)
   0.024096     [D1] interrupt read error: error submitting URB: No such file or directory (-2)
   0.030572     [D1] interrupt read error: error submitting URB: No such file or directory (-2)
   0.036698     [D1] interrupt read error: error submitting URB: No such file or directory (-2)
   0.043001     [D1] interrupt read error: error submitting URB: No such file or directory (-2)
   0.049476     [D1] interrupt read error: error submitting URB: No such file or directory (-2)
   0.056389     [D1] interrupt read error: error submitting URB: No such file or directory (-2)
   0.063118     [D1] interrupt read error: error submitting URB: No such file or directory (-2)
   0.069594     [D1] interrupt read error: error submitting URB: No such file or directory (-2)
   0.076262     [D1] interrupt read error: error submitting URB: No such file or directory (-2)
   0.082711     [D1] interrupt read error: error submitting URB: No such file or directory (-2)
   0.089008     [D1] interrupt read error: error submitting URB: No such file or directory (-2)
   0.095352     [D1] interrupt read error: error submitting URB: No such file or directory (-2)
   0.101893     [D1] interrupt read error: error submitting URB: No such file or directory (-2)
   0.108356     [D1] interrupt read error: error submitting URB: No such file or directory (-2)
   0.114546     [D1] interrupt read error: error submitting URB: No such file or directory (-2)
   0.120961     [D1] interrupt read error: error submitting URB: No such file or directory (-2)
   0.127458     [D1] interrupt read error: error submitting URB: No such file or directory (-2)
   0.133902     [D1] interrupt read error: error submitting URB: No such file or directory (-2)
   0.140679     [D1] interrupt read error: error submitting URB: No such file or directory (-2)
   0.147354     [D1] interrupt read error: error submitting URB: No such file or directory (-2)
   0.153634     [D1] interrupt read error: error submitting URB: No such file or directory (-2)
   0.160420     [D1] interrupt read error: error submitting URB: No such file or directory (-2)
   0.160587     Device not supported!
   0.160617     Device not supported!
```

**richcomm_usb**
```
drivers/richcomm_usb -a richcomm -D

Network UPS Tools - Richcomm dry-contact to USB driver 0.13 (2.8.2)
Warning: This is an experimental driver.
Some features may not function correctly.

   0.000000     [D1] Network UPS Tools version 2.8.2 (release/snapshot of 2.8.2) built with gcc (Debian 12.2.0-14) 12.2.0 and configured with flags: --with-drivers=richcomm_usb,nutdrv_qx --with-statepath=/var/run/nut --with-pidpath=/var/run/nut --with-user=root --with-group=root --sysconfdir=/etc/nut
   0.000120     [D1] debug level is '1'
   0.002875     [D1] Succeeded to become_user(root): now UID=0 GID=0
   0.028376     USB device [0925:1234] matches, but driver callback failed: could not clear/halt ep 129: No such file or directory
   0.028440     upsnotify: failed to notify about state 4: no notification tech defined, will not spam more about it
```

**/etc/nut/ups.conf**
```
[richcomm]
        driver = "richcomm_usb"
        port = "auto"
[nutdrv]
        driver = nutdrv_qx
        port = auto
        vendorid = "0925"
        productid = "1234"
        product = "UPS USB Mon V2.0"
        vendor = "RICHCOMM"
        bus = "001"
```

Did anyone managed to make it run?
Thanks!

pr0fetul, 2024-08-15:
Him,,
It seems that I have a similar issue. I have a TedElectric UPS, powered by a Richcomm driver. When I put _the richcomm_usb_ driver, I have the following output:
```
-----------------------------------------------------------
 Add-on: Network UPS Tools
 Manage battery backup (UPS) devices
-----------------------------------------------------------
 Add-on version: 0.13.1
 You are running the latest version of this add-on.
 System: Home Assistant OS 12.4  (amd64 / qemux86-64)
 Home Assistant Core: 2024.8.1
 Home Assistant Supervisor: 2024.08.0
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-timezone: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
[15:17:47] INFO: Configuring timezone (Europe/Bucharest)...
Log level is set to TRACE
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service base-addon-timezone successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/nut.sh
[15:02:57] INFO: Setting mode to netserver...
[15:02:58] INFO: Connected USB devices:
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 006: ID 0925:1234 Lakeview Research UPS USB Mon V2.0
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 0627:0001 Adomax Technology Co., Ltd QEMU Tablet
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
[15:02:58] INFO: Generating /etc/nut/upsd.users...
[15:02:59] INFO: Configuring user: kiri
[15:02:59] DEBUG: Password SHA1: 40D504F096532AFB7D65D495D9AC882E54D6CB7B
[15:02:59] DEBUG: Requested API resource: https://api.pwnedpasswords.com/range/40D50
[15:02:59] DEBUG: API HTTP Response code: 200
[15:02:59] INFO: Password is NOT in the Have I Been Pwned database! Nice!
[15:03:00] INFO: Configuring Device named TE_1800W...
[15:18:09] INFO: Starting the UPS drivers...
   0.000000	[D1] Starting UPS: TE_1800W
USB device [0925:1234] matches, but driver callback failed: Entity not found
Network UPS Tools - Richcomm dry-contact to USB driver 0.10 (2.8.0)
Warning: This is an experimental driver.
Some features may not function correctly.

Network UPS Tools - UPS driver controller 2.8.0
   0.065558	Driver failed to start (exit status=1)
cont-init: info: /etc/cont-init.d/nut.sh exited 1
cont-init: info: running /etc/cont-init.d/nutclient.sh
[15:18:09] TRACE: bashio::config.equals: mode netclient
[15:18:09] TRACE: bashio::config: mode
[15:18:09] TRACE: bashio::addon.config
[15:18:09] TRACE: bashio::cache.exists: addons.self.options.config
[15:18:09] TRACE: bashio::fs.file_exists: /tmp/.bashio/addons.self.options.config.cache
[15:18:09] TRACE: bashio::cache.get: addons.self.options.config
[15:18:09] TRACE: bashio::cache.exists: addons.self.options.config
[15:18:09] TRACE: bashio::fs.file_exists: /tmp/.bashio/addons.self.options.config.cache
[15:18:09] TRACE: bashio::jq: {"users":[{"username":"xxx","password":"xxxxxx","instcmds":["all"],"actions":[]}],"devices":[{"name":"TE_1800W","driver":"richcomm_usb","port":"auto","config":[]}],"mode":"netserver","shutdown_host":false,"list_usb_devices":true,"log_level":"trace","leave_front_door_open":false} if (.mode == null) then
            null
        elif (.mode | type == "string") then
            .mode // empty
        elif (.mode | type == "boolean") then
            .mode // false
        elif (.mode | type == "array") then
            if (.mode == []) then
                empty
            else
[... 182 lines cut ...]
                empty
            else
                .upsmon_deadtime
            end
        else
            .upsmon_deadtime
        end
cont-init: info: /etc/cont-init.d/nutclient.sh exited 0
cont-init: warning: some scripts exited nonzero
s6-rc: warning: unable to start service legacy-cont-init: command exited 1
/run/s6/basedir/scripts/rc.init: warning: s6-rc failed to properly bring all the services up! Check your logs (in /run/uncaught-logs/current if you have in-container logging) for more information.
/run/s6/basedir/scripts/rc.init: fatal: stopping the container.
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service base-addon-timezone: stopping
s6-rc: info: service base-addon-log-level: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service base-addon-timezone successfully stopped
s6-rc: info: service base-addon-log-level successfully stopped
s6-rc: info: service base-addon-banner: stopping
s6-rc: info: service base-addon-banner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
```

Any idea on how can I make it work?

romale, 2024-12-23:
I've issue to. 
UPS: Exegate UHB-3000

```
lsusb |grep Lake
Bus 003 Device 004: ID 0925:1234 Lakeview Research UPS USB Mon V2.0

./richcomm_usb -a ups -u root
Network UPS Tools 2.8.2.1755-1755-gf1bc634ab (development iteration after 2.8.2) - Richcomm dry-contact to USB driver 0.14
Warning: This is an experimental driver.
Some features may not function correctly.

USB device [0925:1234] matches, but driver callback failed: could not clear/halt ep 129: No such file or directory
upsnotify: failed to notify about state NOTIFY_STATE_STOPPING: no notification tech defined, will not spam more about it

./nutdrv_qx -a ups -u root
Network UPS Tools 2.8.2.1755-1755-gf1bc634ab (development iteration after 2.8.2) - Generic Q* USB/Serial driver 0.39
USB communication driver (libusb 0.1) 0.50
Device not supported!
Device not supported!
upsnotify: failed to notify about state NOTIFY_STATE_STOPPING: no notification tech defined, will not spam more about it
```

Stasiz, 2024-12-25:
UPS ExeGate SpecialPro Smart LLB-650.LCD.AVR.2SH.RJ.USB works fine with nutdrv_qx.
**lsusb**
```
Bus 005 Device 002: ID 0925:1234 Lakeview Research
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x0925 Lakeview Research
  idProduct          0x1234
  bcdDevice            0.00
  iManufacturer           1 RICHCOMM
  iProduct                2 UPS USB Mon V2.0
```
**upsc**
```
Init SSL without certificate database
battery.voltage: 13.6
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: nutdrv_qx
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: auto
driver.state: quiet
driver.version: 2.8.2
driver.version.data: Q1 0.08
driver.version.internal: 0.36
driver.version.usb: libusb-1.0.23 (API: 0x1000107)
input.frequency: 49.0
input.voltage: 244.0
input.voltage.fault: 0.0
output.voltage: 243.0
ups.beeper.status: enabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.load: 3
ups.productid: 1234
ups.status: OL
ups.temperature: 30.8
ups.type: offline / line interactive
ups.vendorid: 0925
```
**os**
```
Oracle Linux Server release 8.10
nut-2.8.2-1.el8.x86_64
```

MauroMP1, 2024-12-26:
> UPS ExeGate SpecialPro Smart LLB-650.LCD.AVR.2SH.RJ.USB works fine with nutdrv_qx. **lsusb**
>
> ```
> Bus 005 Device 002: ID 0925:1234 Lakeview Research
> Device Descriptor:
>   bLength                18 [...]

Hello, i have the same iusse with ups hikvision ds-ups2000, can you share the ups.conf?, for this case, thanks

Stasiz, 2024-12-26:
> can you share the ups.conf?
Nothing special:
```
[myups]
  driver = nutdrv_qx
  port = auto
```
But I used the rules for udev:
`cp /lib/udev/rules.d/62-nut-usbups.rules /etc/udev/rules.d/`
There is:
```
#  Sweex 1000VA  - nutdrv_qx richcomm_usb
ATTR{idVendor}=="0925", ATTR{idProduct}=="1234", MODE="664", GROUP="dialout"
```

MauroMP1, 2024-12-26:
i have a other ups with nutdrv_qx (Forza ds500), i think this generate conflict, because with this rule disappears the ups Forza and show the hikvision but with vendorid of forza

pr0fetul, 2024-12-27:
> Nothing special:
>
> ```
> [myups]
>   driver = nutdrv_qx
>   port = auto
> ```
>
> But I used the rules for udev: `cp /lib/udev/rules.d/62-nut-usbups.rules /etc/udev/rules.d/` There is:
>
> ```
> #  Sweex 1000VA  - [...]

Hi Stasiz, 

When I try with the driver nutdrv_qx - I get the following in HA addon:
`s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/nut.sh
[11:25:14] INFO: Setting mode to netserver...
[11:25:14] INFO: Connected USB devices:
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 002: ID 0925:1234 Lakeview Research UPS USB Mon V2.0
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 0627:0001 Adomax Technology Co., Ltd QEMU Tablet
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
[11:25:15] INFO: Generating /etc/nut/upsd.users...
[11:25:15] INFO: Configuring user: kiri
[11:25:15] INFO: Password is NOT in the Have I Been Pwned database! Nice!
[11:25:16] INFO: Configuring Device named tedelectric...
[11:25:16] INFO: Starting the UPS drivers...
Network UPS Tools - Generic Q* USB/Serial driver 0.32 (2.8.0)
USB communication driver (libusb 1.0) 0.43
Device not supported!
Device not supported!
Driver failed to start (exit status=1)
Network UPS Tools - UPS driver controller 2.8.0
cont-init: info: /etc/cont-init.d/nut.sh exited 1
cont-init: info: running /etc/cont-init.d/nutclient.sh
cont-init: info: /etc/cont-init.d/nutclient.sh exited 0
cont-init: warning: some scripts exited nonzero
s6-rc: warning: unable to start service legacy-cont-init: command exited 1
/run/s6/basedir/scripts/rc.init: warning: s6-rc failed to properly bring all the services up! Check your logs (in /run/uncaught-logs/current if you have in-container logging) for more information.
/run/s6/basedir/scripts/rc.init: fatal: stopping the container.
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service base-addon-timezone: stopping
s6-rc: info: service base-addon-log-level: stopping
s6-rc: info: service base-addon-timezone successfully stopped
s6-rc: info: service base-addon-log-level successfully stopped`

When I try with the richcomm_usb driver, I get:
`s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/nut.sh
[11:30:16] INFO: Setting mode to netserver...
[11:30:16] INFO: Connected USB devices:
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 002: ID 0925:1234 Lakeview Research UPS USB Mon V2.0
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 0627:0001 Adomax Technology Co., Ltd QEMU Tablet
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
[11:30:16] INFO: Generating /etc/nut/upsd.users...
[11:30:17] INFO: Configuring user: kiri
[11:30:17] INFO: Password is NOT in the Have I Been Pwned database! Nice!
[11:30:18] INFO: Configuring Device named tedelectric...
[11:30:18] INFO: Starting the UPS drivers...
USB device [0925:1234] matches, but driver callback failed: Entity not found
Network UPS Tools - Richcomm dry-contact to USB driver 0.10 (2.8.0)
Warning: This is an experimental driver.
Some features may not function correctly.
Driver failed to start (exit status=1)
Network UPS Tools - UPS driver controller 2.8.0
cont-init: info: /etc/cont-init.d/nut.sh exited 1
cont-init: info: running /etc/cont-init.d/nutclient.sh
cont-init: info: /etc/cont-init.d/nutclient.sh exited 0
cont-init: warning: some scripts exited nonzero
s6-rc: warning: unable to start service legacy-cont-init: command exited 1
/run/s6/basedir/scripts/rc.init: warning: s6-rc failed to properly bring all the services up! Check your logs (in /run/uncaught-logs/current if you have in-container logging) for more information.
/run/s6/basedir/scripts/rc.init: fatal: stopping the container.
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service base-addon-timezone: stopping
s6-rc: info: service base-addon-log-level: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service base-addon-timezone successfully stopped
s6-rc: info: service base-addon-log-level successfully stopped`

I don't know how to solve this at all :( 

How did you manage or where, the rules? maybe I am missing a file or a setup....

Stasiz, 2024-12-27:
> Network UPS Tools - Generic Q* USB/Serial driver 0.32 (2.8.0)
> USB communication driver (libusb 1.0) 0.43

Try updating NUT addon. I have a version:
```
nutdrv_qx -V

Network UPS Tools - Generic Q* USB/Serial driver 0.36 (2.8.2)
USB communication driver (libusb 1.0) 0.47
```

MauroMP1, 2025-01-06:
Hello again, i tried this:
Install rpi4 how in this post https://github.com/networkupstools/nut/issues/1754
Continued of issue:
I have 3 ups, APC Back-UPS CS 500 (works), Forza ds500(works), Hikvision DS-UPS2000(not work)
```shell
lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 0925:1234 Lakeview Research UPS USB Mon V2.0
Bus 001 Device 004: ID 0665:5161 Cypress Semiconductor USB to Serial
Bus 001 Device 003: ID 051d:0002 American Power Conversion Uninterruptible Power Supply
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
``` 

```shell
nut-scanner
Scanning USB bus.
No IP range(s) requested, skipping NUT bus (old libupsclient connect method)
Scanning NUT simulation devices.
[nutdev-usb1]
        driver = "usbhid-ups"
        port = "auto"
        vendorid = "051D"
        productid = "0002"
        # bus = "001"
        # device = "003"
[nutdev-usb2]
        driver = "nutdrv_qx"    # alternately: blazer_usb
        port = "auto"
        vendorid = "0665"
        productid = "5161"
        # bus = "001"
        # device = "004"
[nutdev-usb3]
        driver = "nutdrv_qx"    # alternately: richcomm_usb
        port = "auto"
        vendorid = "0925"
        productid = "1234"
        # bus = "001"
        # device = "005"
``` 

Tried nutdrv_qx and richcomm_usb
with richcomm_usb
ups.conf
```shell
[apc]
        driver = usbhid-ups
        port = auto
        vendorid = 051D
        productid = 0002
        # bus = "001"
        # device = "003"
[forza]
        driver = nutdrv_qx
        port = auto
        vendorid = 0665
        productid = 5161
        # bus = 001
        # device = 004
[hikvision]
        driver = richcomm_usb
        port = auto
        vendorid = 0925
        productid = 1234
        # bus = "001"
        # device = "005"
``` 
```shell
Jan 05 22:36:22 nutserverpi upsd[698]: Can't connect to UPS [hikvision] (richcomm-usb-hikvision): No such file or directory
[... 100 lines cut ...]
Jan 05 23:06:51 nutserverpi nut-driver@apc[477]: Network UPS Tools upsdrvctl - UPS driver controller 2.8.2.1827-1827-g99ea9729a (development iteration after 2.8.2)
Jan 05 23:08:02 nutserverpi upsd[1213]: WARNING: /etc/nut/upsd.conf is world readable (hope you don't have passwords there)
Jan 05 23:08:02 nutserverpi nut-server[1213]: Network UPS Tools upsd 2.8.2.1827-1827-g99ea9729a (development iteration after 2.8.2)
Jan 05 23:08:02 nutserverpi nut-server[1213]: WARNING: /etc/nut/upsd.conf is world readable (hope you don't have passwords there)
Jan 05 23:08:02 nutserverpi upsd[1213]: listening on 0.0.0.0 port 3493
Jan 05 23:08:02 nutserverpi upsd[1213]: Can't connect to UPS [hikvision] (nutdrv_qx-hikvision): No such file or directory
Jan 05 23:08:02 nutserverpi upsd[1213]: Connected to UPS [forza]: nutdrv_qx-forza
Jan 05 23:08:02 nutserverpi upsd[1213]: Connected to UPS [apc]: usbhid-ups-apc
Jan 05 23:08:02 nutserverpi upsd[1213]: Found 3 UPS defined in ups.conf
Jan 05 23:08:02 nutserverpi nut-server[1213]: WARNING: /etc/nut/upsd.users is world readable (hope you don't have passwords there)
Jan 05 23:08:02 nutserverpi upsd[1213]: WARNING: /etc/nut/upsd.users is world readable (hope you don't have passwords there)
Jan 05 23:08:02 nutserverpi upsd[1213]: Running as foreground process, not saving a PID file
Jan 05 23:08:02 nutserverpi upsd[1213]: upsnotify: failed to notify about state NOTIFY_STATE_READY_WITH_PID: no notification tech defined, will not spam more about it
Jan 05 23:08:02 nutserverpi nut-monitor[1216]: Network UPS Tools upsmon 2.8.2.1827-1827-g99ea9729a (development iteration after 2.8.2)
Jan 05 23:08:02 nutserverpi upsd[1213]: User monitor@127.0.0.1 logged into UPS [apc]
Jan 05 23:08:02 nutserverpi upsd[1213]: User monitor@127.0.0.1 logged into UPS [forza]
Jan 05 23:08:02 nutserverpi upsd[1213]: User monitor@127.0.0.1 logged into UPS [hikvision]
Jan 05 23:08:17 nutserverpi nut-driver@hikvision[1282]: Network UPS Tools upsdrvctl - UPS driver controller 2.8.2.1827-1827-g99ea9729a (development iteration after 2.8.2)
Jan 05 23:09:22 nutserverpi nut-driver@hikvision[1431]: Network UPS Tools upsdrvctl - UPS driver controller 2.8.2.1827-1827-g99ea9729a (development iteration after 2.8.2)
Jan 05 23:10:28 nutserverpi nut-driver@hikvision[1547]: Network UPS Tools upsdrvctl - UPS driver controller 2.8.2.1827-1827-g99ea9729a (development iteration after 2.8.2)
``` 

I guess not supported yet
If someone have a idea to try....
Thanks

Stasiz, 2025-01-06:
The idea is to find the right driver for Hikvision DS-UPS2000.
First you need to stop all nut-driver@*.services.  Then try the driver. For example for my UPS:
`nutdrv_qx -s testhik -DDD -x vendorid=0925 -x productid=1234 -x port=auto -x protocol=Q1 -x subdriver=armac`

MauroMP1, 2025-01-06:
> The idea is to find the right driver for Hikvision DS-UPS2000. First you need to stop all nut-driver@*.services. Then try the driver. For example for my UPS: `nutdrv_qx -s testhik -DDD -x vendorid=0925 -x productid=1234 -x port=auto -x protocol=Q1 -x subdriver=armac` or wiithout protocol `nutdrv_qx -s testhik -DDD -x vendorid=0925 -x productid=1234 -x port=auto` or richcomm_usb driver `richcomm_usb -s testhik -DDD -x vendorid=0925 -x productid=1234 -x port=auto`

Hello, yes try all subdrivers and protocols, but "Device not supported"

```shell
sudo ./nutdrv_qx -s hikvision -u root -DDDDDD -x productid=1234 -x vendorid=0925 -x subdriver=fabula -x port=auto
sudo ./nutdrv_qx -s hikvision -u root -DDDDDD -x productid=1234 -x vendorid=0925 -x subdriver=armac -x port=auto
cypress, phoenixtec, phoenix, ippon, krauler, fabula, hunnox, fuji, sgs, snr, ablerex, armac
Acceptable values for 'protocol' via -x or ups.conf in this driver: voltronic, voltronic-qs, voltronic-qs-hex, mustek, megatec/old, bestups, mecer, megatec, zinto, masterguard, hunnox, ablerex, innovart31, q1
``` 
All saied same 
```shell
 Checking device (0925/1234) (001/005)
   3.026478     [D2] - VendorID: 0925
   3.026514     [D2] - ProductID: 1234
   3.026532     [D2] - Manufacturer: RICHCOMM
   3.026548     [D2] - Product: UPS USB Mon V2.0
   3.026564     [D2] - Serial Number: unknown
   3.026581     [D2] - Bus: 001
   3.026597     [D2] - Device: 005
   3.026613     [D2] - Device release number: 0000
   3.026629     [D2] Trying to match device
   3.026646     [D3] match_function_exact: matching a device...
   3.026665     [D3] match_function_regex: matching a device...
   3.026703     [D2] Device matches
   3.026771     [D3] nut_libusb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   3.026803     [D5] send_to_all: SETINFO driver.state "reconnect.updateinfo"
   3.026825     [D4] armac command Q1
   3.031314     [D1] interrupt read error: error submitting URB: No such file or directory (-2)
   3.031360     [D5] send_to_all: SETINFO driver.state "reconnect.trying"
   3.031435     [D4] qx_process: failed to preprocess answer [input.voltage]
   3.031456     Device not supported!
   3.031473     Device not supported!
   3.031506     [D5] send_to_all: SETINFO driver.state "cleanup.exit"
   3.031530     upsnotify: failed to notify about state NOTIFY_STATE_STOPPING: no notification tech defined, will not spam more about it
   3.031550     [D1] On systems without service units, consider `export NUT_QUIET_INIT_UPSNOTIFY=true`
``` 
I will try write to support of hikvision, maybe tell me something that help

Thanks for reply

MauroMP1, 2025-05-05:
> > Hikvision DS-UPS2000
> 
> Any news about this? have the same issue with Hikvision DS-UPS2000 :(

Hello, nothing yet, if you can find something, share
thanks
regards

jmaloway, 2025-06-28:
I'm having the same issue with the Hikvision DS-UPS3000, which I suppose makes sense since presumably they all use the same interface. Fixing it is over my head, but I can definitely test and provide feedback if it's helpful.