networkupstools/nut #2134: snmp-ups: time-intensive, futile attempts on non-existent/closed hosts

desertwitch, 2023-10-28:
Reading yesterday's e-mail in the mailing list regarding a more or less imminent 2.8.1. release, I've decided to do some more testing. I've actually run into this problem (with a stuck system boot) before but haven't been able to trace it back to NUT and the SNMP driver.

At present the SNMP driver will start time-intensive, futile attempts to walk all kinds of MIBs on target hosts which are non-existent or do not even have a SNMP server listening. It will spend almost 4 minutes attemping SNMP operations on a dead/unresponsive target host before determining there's no connection possible or readable SNMP available and exiting. 

This can (especially at system boot and with home-brewed startup scripts or badly implemented service environments) cause an unwanted delay in the whole boot routine, made much harder to trace by the fact that the `snmp-ups` driver itself does not present any log (only debug > 1) messages during these futile attempts... only after it eventually gives up after the ~4 minutes (in my case).

### Regarding `upsdrvctl`

I've - in particular - noticed this line in the log file (see below for full log):
```
Oct 28 02:47:16 Tower root: Startup timer elapsed, continuing...
```
Is it intended that `upsdrvctl` let the driver continue (which at present it does) after this point or should it kill it?

I have RTFM but I didn't find a startup timer option after which (I presume) the `upsdrvctl` should give up on a driver?
**If there isn't one - would this be a useful addition worth looking into for `upsdrvctl`?**

### Regarding `snmp-ups`

**Not sure how, but possibly we can (sanity-)check if the host and/or SNMP ports are connectable on driver launch?**

These are (debug level 4) log messages having to driver attempt to connect to a non-existent IP on my network:
```
Oct 28 02:46:31 Tower root:    0.000000#011[D1] debug level is '4'
Oct 28 02:46:31 Tower root:    0.001365#011[D1] Succeeded to become_user(root): now UID=0 GID=0
Oct 28 02:46:31 Tower root:    0.001415#011[D1] Saving PID 21139 into /var/run/nut/snmp-ups-ups.pid
Oct 28 02:46:31 Tower root:    0.001634#011[D1] SNMP UPS driver: entering upsdrv_initups()
Oct 28 02:46:31 Tower root:    0.001659#011[D2] SNMP UPS driver: entering nut_snmp_init(snmp-ups)
Oct 28 02:46:31 Tower root:    0.016486#011[D2] Setting SNMP retries to 5
Oct 28 02:46:31 Tower root:    0.016496#011[D2] Setting SNMP timeout to 1 second(s)
Oct 28 02:46:31 Tower root:    0.016625#011[D1] SNMP UPS driver: entering load_mib2nut(auto) to detect proper MIB for device [ups] (host 192.168.0.238)
Oct 28 02:46:31 Tower root:    0.016646#011[D2] load_mib2nut: trying the new match_sysoid() method with auto
Oct 28 02:46:31 Tower root:    0.016661#011[D3] load_mib2nut: trying the new match_sysoid() method: attempt #1
Oct 28 02:46:31 Tower root:    0.016676#011[D3] Entering nut_snmp_get_oid()
Oct 28 02:46:31 Tower root:    0.016690#011[D3] nut_snmp_get(.1.3.6.1.2.1.1.2.0)
Oct 28 02:46:31 Tower root:    0.016704#011[D3] nut_snmp_walk(.1.3.6.1.2.1.1.2.0)
Oct 28 02:46:31 Tower root:    0.016717#011[D4] nut_snmp_walk: max. iteration = 1
Oct 28 02:46:37 Tower root:    6.023129#011[D2] Can't get sysOID value (using nut_snmp_get_oid())
Oct 28 02:46:37 Tower root:    6.023145#011[D3] Entering nut_snmp_get_str()
Oct 28 02:46:37 Tower root:    6.023148#011[D3] nut_snmp_get(.1.3.6.1.2.1.1.2.0)
Oct 28 02:46:37 Tower root:    6.023153#011[D3] nut_snmp_walk(.1.3.6.1.2.1.1.2.0)
Oct 28 02:46:37 Tower root:    6.023157#011[D4] nut_snmp_walk: max. iteration = 1
Oct 28 02:46:43 Tower root:   12.029002#011[D2] Can't get sysOID value (using nut_snmp_get_str())
Oct 28 02:46:43 Tower root:   12.029018#011[D3] load_mib2nut: failed with new match_sysoid() method
Oct 28 02:46:43 Tower root:   12.029023#011[D3] load_mib2nut: trying the new match_sysoid() method: attempt #2
Oct 28 02:46:43 Tower root:   12.029027#011[D3] Entering nut_snmp_get_oid()
Oct 28 02:46:43 Tower root:   12.029031#011[D3] nut_snmp_get(.1.3.6.1.2.1.1.2.0)
Oct 28 02:46:43 Tower root:   12.029034#011[D3] nut_snmp_walk(.1.3.6.1.2.1.1.2.0)
Oct 28 02:46:43 Tower root:   12.029038#011[D4] nut_snmp_walk: max. iteration = 1
Oct 28 02:46:49 Tower root:   18.035416#011[D2] Can't get sysOID value (using nut_snmp_get_oid())
Oct 28 02:46:49 Tower root:   18.035431#011[D3] Entering nut_snmp_get_str()
Oct 28 02:46:49 Tower root:   18.035436#011[D3] nut_snmp_get(.1.3.6.1.2.1.1.2.0)
Oct 28 02:46:49 Tower root:   18.035440#011[D3] nut_snmp_walk(.1.3.6.1.2.1.1.2.0)
Oct 28 02:46:49 Tower root:   18.035445#011[D4] nut_snmp_walk: max. iteration = 1
Oct 28 02:46:55 Tower root:   24.041775#011[D2] Can't get sysOID value (using nut_snmp_get_str())
Oct 28 02:46:55 Tower root:   24.041790#011[D3] load_mib2nut: failed with new match_sysoid() method
Oct 28 02:46:55 Tower root:   24.041795#011[D3] load_mib2nut: trying the new match_sysoid() method: attempt #3
Oct 28 02:46:55 Tower root:   24.041799#011[D3] Entering nut_snmp_get_oid()
Oct 28 02:46:55 Tower root:   24.041803#011[D3] nut_snmp_get(.1.3.6.1.2.1.1.2.0)
Oct 28 02:46:55 Tower root:   24.041807#011[D3] nut_snmp_walk(.1.3.6.1.2.1.1.2.0)
Oct 28 02:46:55 Tower root:   24.041810#011[D4] nut_snmp_walk: max. iteration = 1
Oct 28 02:47:01 Tower root:   30.048193#011[D2] Can't get sysOID value (using nut_snmp_get_oid())
Oct 28 02:47:01 Tower root:   30.048209#011[D3] Entering nut_snmp_get_str()
Oct 28 02:47:01 Tower root:   30.048213#011[D3] nut_snmp_get(.1.3.6.1.2.1.1.2.0)
Oct 28 02:47:01 Tower root:   30.048217#011[D3] nut_snmp_walk(.1.3.6.1.2.1.1.2.0)
Oct 28 02:47:01 Tower root:   30.048221#011[D4] nut_snmp_walk: max. iteration = 1
Oct 28 02:47:07 Tower root:   36.054560#011[D2] Can't get sysOID value (using nut_snmp_get_str())
Oct 28 02:47:07 Tower root:   36.054576#011[D3] load_mib2nut: failed with new match_sysoid() method
Oct 28 02:47:07 Tower root:   36.054581#011[D4] load_mib2nut: checking against mapping table entry #0 "apc_ats"
Oct 28 02:47:07 Tower root:   36.054585#011[D2] load_mib2nut: trying classic sysOID matching method with 'apc_ats' mib
[... 333 lines cut ...]
Oct 28 02:50:19 Tower root:  228.250259#011[D2] load_mib2nut: trying classic sysOID matching method with 'tripplite' mib
Oct 28 02:50:19 Tower root:  228.250272#011[D3] su_find_info: unknown info type (device.model)
Oct 28 02:50:19 Tower root:  228.250277#011[D3] su_find_info: "ups.model" found
Oct 28 02:50:19 Tower root:  228.250281#011[D2] Found entry, not a template 1.3.6.1.2.1.33.1.1.2.0
Oct 28 02:50:19 Tower root:  228.250285#011[D2] Testing ups.model using OID 1.3.6.1.2.1.33.1.1.2.0
Oct 28 02:50:19 Tower root:  228.250289#011[D3] Entering nut_snmp_get_str()
Oct 28 02:50:19 Tower root:  228.250292#011[D3] nut_snmp_get(1.3.6.1.2.1.33.1.1.2.0)
Oct 28 02:50:19 Tower root:  228.250296#011[D3] nut_snmp_walk(1.3.6.1.2.1.33.1.1.2.0)
Oct 28 02:50:19 Tower root:  228.250300#011[D4] nut_snmp_walk: max. iteration = 1
Oct 28 02:50:25 Tower root:  234.256638#011[D3] load_mib2nut: testOID provided and doesn't match MIB 'tripplite'!
Oct 28 02:50:25 Tower root:  234.256655#011[D4] load_mib2nut: checking against mapping table entry #33 "ietf"
Oct 28 02:50:25 Tower root:  234.256660#011[D2] load_mib2nut: trying classic sysOID matching method with 'ietf' mib
Oct 28 02:50:25 Tower root:  234.256674#011[D3] su_find_info: unknown info type (device.model)
Oct 28 02:50:25 Tower root:  234.256678#011[D3] su_find_info: "ups.model" found
Oct 28 02:50:25 Tower root:  234.256682#011[D2] Found entry, not a template 1.3.6.1.2.1.33.1.1.2.0
Oct 28 02:50:25 Tower root:  234.256686#011[D2] Testing ups.model using OID 1.3.6.1.2.1.33.1.1.2.0
Oct 28 02:50:25 Tower root:  234.256690#011[D3] Entering nut_snmp_get_str()
Oct 28 02:50:25 Tower root:  234.256694#011[D3] nut_snmp_get(1.3.6.1.2.1.33.1.1.2.0)
Oct 28 02:50:25 Tower root:  234.256697#011[D3] nut_snmp_walk(1.3.6.1.2.1.33.1.1.2.0)
Oct 28 02:50:25 Tower root:  234.256701#011[D4] nut_snmp_walk: max. iteration = 1
Oct 28 02:50:31 Tower root:  240.262902#011[D3] load_mib2nut: testOID provided and doesn't match MIB 'ietf'!
Oct 28 02:50:31 Tower root:  240.262918#011No supported device detected at [ups] (host 192.168.0.238)
Oct 28 02:50:31 Tower root:  240.262987#011upsnotify: failed to notify about state 4: no notification tech defined, will not spam more about it
Oct 28 02:50:31 Tower root: Network UPS Tools - Generic SNMP UPS driver 1.30 (2.8.0.1)
```

desertwitch, 2023-10-28:
Also relevant:
https://github.com/networkupstools/nut/blame/858db30fef2e6854b114eeb1306381b59c05f848/drivers/snmp-ups.c#L618

jimklimov, 2023-10-28:
As far as I know, with at least classic UDP-based SNMP the lack of server or wrong password are indistinguishable: you fire a best effort UDP packet, no reply comes in, that's it. After an arbitrary timeout you decide to give up and do not know if the server is not available, or chose to not reply (bad auth, stressed, etc.), or either packet got lost in transit... More or less the same can be seen with standard tools like `snmpwalk`.

Just to clarify: in the environments you work with, are init-scripts still the "king of the day" (so e.g. NUT drivers' long init times can be blocking the appearance of interactive console login), or are service frameworks like systemd or SMF available? In particular, what I'd expect of "frameworks" includes:
* dependency definition (this starts after that) which in particular can untie NUT "branch" of the system services from the rest such as interactive environments
* auto-revival of dead service instances

So how much are you limited by requiring `upsdrvctl` as a single program to start *all* drivers successfully and in one shot (or fail on any hiccup) as a unitary step in the system boot-up?

Regarding configurability for this, check https://github.com/networkupstools/nut/blob/master/conf/ups.conf.sample - in particular: `maxretry`, `retrydelay` and overall `maxstartdelay`.

In trench experience of the 42ity project for massive-monitoring appliances, the burst of driver inits could overwhelm the appliance CPU to the point that it timed out every driver across the 3-minute boot allowance. Ended up making a special systemd unit that wrapped `nut-driver@instancename.service` logic to keep those per-driver units off by default and to "manually" start them in batches of 30 or so.

desertwitch, 2023-10-28:
Thanks for getting back to me in light speed (again), Jim.

> As far as I know, with at least classic UDP-based SNMP, the lack of server or wrong password are indistinguishable: you fire a best effort UDP packet, no reply comes in, that's it. After an arbitrary timeout you decide to give up and do not know if the server is not available, or chose to not reply (bad auth, stressed, etc.), or either packet got lost in transit... There is no "session" whose state you can track to gauge success of the attempt, like with TCP. More or less the same can be seen with standard tools like snmpwalk.

This is also what I've gathered from my investigations into this - so it seems to be a _cul-de-sac_ (protocol-wise).

> Just to clarify: in the environments you work with, are init-scripts still the "king of the day" (so e.g. NUT drivers' long init times can be blocking the appearance of interactive console login), [...]

Init-scripts - but makes one get creative with such things... :-)

> So how much are you limited by requiring upsdrvctl as a single program to start all drivers successfully and in one shot (or fail on any hiccup) as a unitary step in the system boot-up?

Not massively, but it's certainly a convenience not having to evaluate which drivers my end-users have configured.

> Regarding configurability for this, check https://github.com/networkupstools/nut/blob/master/conf/ups.conf.sample - in particular: maxretry, retrydelay and overall maxstartdelay.

I had the feeling I missed something, sorry about that...
But reading up on that does that make me wonder about that timeout.

```
# maxstartdelay: OPTIONAL.  This can be set as a global variable
#                above your first UPS definition and it can also be
#                set in a UPS section.  This value controls how long
#                upsdrvctl will wait for the driver to finish starting.
#                This keeps your system from getting stuck due to a
#                broken driver or UPS.
#                The default is 45 seconds.
```
Default (when not explicitly set) seems 45 seconds, resulting (probably) in this log line (45s after start):
```
Oct 28 02:47:16 Tower root: Startup timer elapsed, continuing...
```
Following that logic shouldn't the SNMP driver have been killed (here) after these 45 seconds elapsed?

jimklimov, 2023-10-28:
Looking at code, seems it would retry (3 times by default in the `ups.conf.sample`, 1 in code) so forking off new drivers. In fact, there does not seem to be any reaping for drivers that just timed out but did not die during `forkexec()` - though usually they should avoid stepping on each other's toes by killing off an earlier instance themselves.

At least, if the timeouts you see are like 3*45 sec - that may be it.

desertwitch, 2023-10-28:
> Looking at code, seems it would retry (3 times by default in the `ups.conf.sample`, 1 in code) so forking off new drivers. In fact, there does not seem to be any reaping for drivers that just timed [...]

I'll investigate some more into this, because I don't have `maxretry` set in my `ups.conf`.

So it should default into 1(default `maxretry`)*45(default `maxstartdelay`) = 45 seconds

Those 45 seconds being the first and only time that log line appears:
`Oct 28 02:47:16 Tower root: Startup timer elapsed, continuing...`

But even with  3(`maxretry`)*45(`maxstartdelay`) = ~2,25 minutes (2,5 minutes with 3x the default `retrydelay` of 5s)

... something doesn't add up (if my logic is not flawed somehow here... which is always possible 😆 )

```
#    maxretry: OPTIONAL.  Specify the number of attempts to start the driver(s),
#              in case of failure, before giving up. A delay of 'retrydelay' is
#              inserted between each attempt. Caution should be taken when using
#              this option, since it can impact the time taken by your system to
#              start.
#
#              The built-in default is 1 attempt.
```

In my logs it seems the driver is only started once, and kills **itself** after the 4 minutes, rather than being killed by any `upsdrvctl` timeouts - so perhaps `upsdrvctl` is already considering the driver started at that point, but more likely it doesn't kill it as it should after this log line: `Oct 28 02:47:16 Tower root: Startup timer elapsed, continuing...`. That is under the assumption that "continuing..." in this case means continuing to the next defined UPS (none, in my case) rather than continuing with the already running driver despite the "timer elapsed".

desertwitch, 2023-10-28:
Diving into the code all these cases (except for the first) already have a dead fork at that point (and no need to terminate it)

https://github.com/networkupstools/nut/blob/858db30fef2e6854b114eeb1306381b59c05f848/drivers/upsdrvctl.c#L668-L693

Upstream function evaluating the error counter doesn't terminate, just checks whether to spawn new forks based on the options and whether an error was encountered (which it was, just not resulting in fork termination by itself):
https://github.com/networkupstools/nut/blob/858db30fef2e6854b114eeb1306381b59c05f848/drivers/upsdrvctl.c#L893-L903

So it seems the fork never gets killed (because in all other cases it's already dead), so perhaps needs explicit killing here?
https://github.com/networkupstools/nut/blob/858db30fef2e6854b114eeb1306381b59c05f848/drivers/upsdrvctl.c#L668-L672

desertwitch, 2023-10-28:
I'm running some tests based on my logic above and will report back here or with a PR.

desertwitch, 2023-10-28:
I've tested this some more and it seems my assumption is right that an elapsed maxstartdelay does report an error to the upstream fork-spawning function (resulting in spawning of additional forks when maxretry > 1) but not in termination of the current time-elapsed fork. With the default setting of maxretry=1 the maxstartdelay basically does elapse (and does log it) but that driver instance is never effectively terminated and runs forever until self-termination (as seen in the logs).

jimklimov, 2023-10-28:
My expectation was that the newly forked+execed driver would find (e.g. by PID file) that a predecessor exists, and would kill it then. On `upsdrvctl` side that previous attempt (if it did not fork off into the background again - something the debug flags can prevent by default) would probably become a zombie child process to be reaped explicitly (or eventually as `upsdrvctl` itself ends and the OS reaps its resources).

As for a SIGTERM - I think it is evaluated between loops in `main.c`, so if the driver is stuck in some time-consuming method for initialization, it might at best raise the "I'm terminated" flag with the signal handler, but never get to check/process it...

desertwitch, 2023-10-28:
> My expectation was that the newly forked+execed driver would find (e.g. by PID file) that a predecessor exists, and would kill it then. On `upsdrvctl` side that previous attempt (if it did not fork [...]

That it does, but with a default maxretry=1 a second fork (killing the first one by detection) is never spawned (as intended), and the first fork just keeps running way past our maxstartdelay until self-termination (as it's never killed by a second fork). So methinks every such error (possibly resulting or not resulting in fork-spawning) should finish with a check if the fork is actually terminated. Otherwise with maxretry=1 in its default state the option becomes useless?

jimklimov, 2023-10-28:
I think the idea is to try and start the driver after all :) So if it takes too long, `upsdrvctl` takes note and will probably return an error exit-code eventually, but marches on to start another driver (if there are many) while the attempted driver's forked process is on its own and no longer monitored to succeed or fail the start-up.

It would be counter-productive here to be pedantic and kill drivers just because they did not talk to some remote honeypot quickly enough.

desertwitch, 2023-10-28:
> I think the idea is to try and start the driver after all :) So if it takes too long, `upsdrvctl` takes note and will probably return an error exit-code eventually, but marches on to start another driver (if there are many) while the attempted driver's forked process is on its own and no longer monitored to succeed or fail the start-up.
> 
> 
> 
> It would be counter-productive here to be pedantic and kill drivers just because they did not talk to some remote honeypot quickly enough.
> 
> 
> 
> BTW if wrapping into systemd/SMF units, keep in mind the framework settings for a process to start up (90 sec by default in systemd) and possibly tune that up for the slower remote devices. Especially the SNMP NMCs tended to time out or reboot under stress for us (when e.g. 30 clients bombard the same poor card with a full walk as they boot up).

Makes sense when you say it like that. I'm good with keeping this as is with likely only one not immediately error-concious driver affected. Wondering if it'd make sense to put a side-note as part of the maxstartdelay setting description that the timeout is more intended towards ensuring successive driver startup rather than cleaning up stale driver instances. But I'm good as far as this issue is concerned, thanks for the time as usual, Jim - sorry if it caused any inconvenience.

jimklimov, 2023-10-28:
A docs update may be worthwhile. After all, it is from outside that the missing bits are best visible :)

desertwitch, 2023-10-28:
Thanks for the commit, one last thing I was wondering about when reading your side-note:

> Note that after this time upsdrvctl would just move
> along with its business (whether retrying the same
> driver if `maxretry>1`, or trying another driver if
> starting them all, **or just exit**); however the most
> recently started "stuck" driver process may be further
> initializing **in the background**, and might even succeed
> eventually.

`upsdrvctl` doesn't actually **exit** until the SNMP driver self-terminates after the 4 minutes.
It doesn't let the driver do its hopeless thing in the background either, as I understand from the above paragraph.
That is with 1 driver configured and default settings of maxretry=1, maxstartdelay=45 seconds - this is still "normal", right?

jimklimov, 2023-10-28:
Well, normally the drivers initialize, fork to background (unless told not to), and begin the infinite polling loop. So if the driver would succeed eventually -

jimklimov, 2023-10-28:
Not sure what to state about upsdrvctl not exiting until child processes actually finish (fail or fork)... I suppose, that is correct (and notes can be further reworded about "eventual" exit). Fir a start of single driver might as well wait and pass its result code.

desertwitch, 2023-10-28:
> Not sure what to state about upsdrvctl not exiting until child processes actually finish (fail or fork)... I suppose, that is correct (and notes can be further reworded about "eventual" exit). For a start of single driver might as well wait and pass its result code.

All good then, many thanks for going down that rabbit hole with me.
As far as I'm concerned, issue can be closed since there seems little room for improvement SNMP-wise.

jimklimov, 2023-10-28:
Update about not blocking the system startup due to `upsdrvctl` - check the `nowait` flag too.

jimklimov, 2023-11-13:
@desertwitch : do you think solvable issues here have been solved?

I guess that for SNMP over UDP the inability to discern un-handled requests from dead hosts is more or less fundamental, so we have to try all detection OIDs etc. and come to conclusions after receiving no replies.

Looking at `nut_snmp_init()` we do not currently customize ports (non-161) nor protocol types (TCP); perhaps lack-of-host could be more visible with TCP - if the device talks that, and more so if it does not :) But I guess someone with actual devices to tap would have to implement this, as well as config file syntax changes.

desertwitch, 2023-11-13:
Yep, I'm good here - thanks - feel free to close!

jimklimov, 2023-11-13:
CC @arnaudquette-eaton @dzomaya : got any interest in enhancing SNMP low-level support per comments above?

I guess I'll offload the specific idea into the issue tracker as a separate thing.

closed 2023-11-13 by jimklimov