networkupstools/nut #2139: [HCL] Prolink PRO1201SFCU

aaronntw, 2023-10-30:
root@TRUENAS-SCALE[/]# upsc -l
Init SSL without certificate database
ups
root@TRUENAS-SCALE[/]# date
Mon Oct 30 11:24:19 +08 2023
root@TRUENAS-SCALE[/]# upsc ups@localhost
Init SSL without certificate database
battery.charge: 100
battery.voltage: 27.10
battery.voltage.high: 26.00
battery.voltage.low: 20.80
battery.voltage.nominal: 24.0
device.mfr: Prolink
device.model: PRO1201SFC
device.type: ups
driver.name: blazer_usb
driver.parameter.pollinterval: 2
driver.parameter.port: /dev/uhid
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.internal: 0.12
input.current.nominal: 4.0
input.frequency: 50.0
input.frequency.nominal: 50
input.voltage: 238.3
input.voltage.fault: 238.3
input.voltage.nominal: 230
output.voltage: 238.3
ups.beeper.status: enabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.firmware:
ups.load: 0
ups.mfr: Prolink
ups.model: PRO1201SFC
ups.productid: 5161
ups.status: OL
ups.type: offline / line interactive
ups.vendorid: 0665

Broadcast message from nut@TRUENAS-SCALE (somewhere) (Mon Oct 30 11:25:41 2023)

Executing automatic power-fail shutdown

2023 Oct 30 11:25:41 TRUENAS-SCALE Executing automatic power-fail shutdown

aaronntw, 2023-10-30:
Product Page
https://prolink2u.com/products/sfcu-series?variant=40573922902089

jimklimov, 2023-10-30:
Thank you for the report!