networkupstools/nut #2140: apc_modbus manpage

desertwitch, 2023-10-30:
I'm travelling these next days but wanted to throw this in before the release (sorry, no means to create a PR at the moment)

`apc_modbus` manpage should probably also include the usual modbus driver paragraph:
```
This driver is not built by default. You can build it by installing libmodbus and running configure --with-modbus=yes.

You also need to give proper permissions on the local serial device file (/dev/ttyUSB0 for example) to allow the run-time NUT driver user account to access it.
```
As well as a small note that for USB to actually work you'll need Axel's `rtu_usb` patched `libmodbus` (perhaps with a link to it). I've had one or two users ask me about this (who had the `apc_modbus` driver recommended by `nut-scanner`) and as far as I remember I couldn't find it anywhere apart from the `NEWS.adoc` so far - probably not the first place to look or to be recommended by Google. 

Happy releasing and thanks to everyone involved for the hard work and tons of improvements since the 2.8.0 release! 💯

desertwitch, 2023-10-31:
resolved as of https://github.com/networkupstools/nut/commit/cddf83b4283c644b2dfee06e816a265aec742678

jimklimov, 2023-10-31:
Follows up for #2063 and #139

jimklimov, 2023-10-31:
Did not make note about permissions on the device-FS node, because the same applies to most other (non-networked) drivers.