networkupstools/nut #2142: Compiling 2.8.1 from sources, windows.h fatal error

teoberi, 2023-11-01:
SO: Slackware64-current
`./configure --with-user=ups --with-group=nut`

> NUT Configuration summary:
> ==========================
> * configured version:   2.8.1
> * build serial drivers: yes
> * build USB drivers:    yes (libusb-1.0)
> * build neon based XML driver:  yes
> * enable Avahi support: no
> * build Powerman PDU client driver:     no
> * build Modbus driver:  no
> * build IPMI driver:    no
> * build GPIO driver:    no
> * build Mac OS X meta-driver:   no
> * build i2c based drivers:      no
> * enable SSL support:   yes (OpenSSL)
> * enable libwrap (tcp-wrappers) support:        yes
> * enable libltdl (Libtool dlopen abstraction) support:  yes
> * build nut-scanner:    yes
> * build CGI programs:   no
> * install NUT-Monitor desktop application:      no
> * install PyNUT binding module: yes
> * use default  Python  interpreter:     /usr/bin/python
> * use specific Python2 interpreter:     /usr/bin/python2.7
> * use specific Python3 interpreter:     /usr/bin/python3
> * build and install documentation:      man=auto
> * build specific documentation format(s):       yes  man
> * build and install the development files:      no
> * consider basic SMF support:   no
> * consider basic systemd support:       no
> * build with tighter systemd support:   no
> * build C++11 codebase (client library, etc.):  yes
> * build C++ tests with CPPUNIT: no
> * build SNMP drivers:   yes
> * build SNMP drivers with statically linked lib(net)snmp:       no
> * User to run as:       ups
> * Group of user to run as:      nu

> NUT Build/Target system info:
> -----------------------------
> * Compact version of C compiler:        gcc (GCC) 13.2.0
> * Compact version of C++ compiler:      g++ (GCC) 13.2.0
> * Compact version of C preprocessor:    gcc (GCC) 13.2.0
> * host env spec we run on:      x86_64-pc-linux-gnu
> * host env spec we built on:    x86_64-pc-linux-gnu
> * host env spec we built for:   x86_64-pc-linux-gnu
> * host OS short spec we run on: x86_64-linux-gnu
> * host OS short spec we built on:       x86_64-linux-gnu
> * host OS short spec we built for:      x86_64-linux-gnu

`make -j3`

> Making all in Windows
> make[2]: Entering directory '/usr/local/src/nut-2.8.1/scripts/Windows'
>   CC       halt.o
>   CCLD     nut
> halt.c:30:10: fatal error: windows.h: No such file or directory
>    30 | #include <windows.h>
>       |          ^~~~~~~~~~~
> compilation terminated.
> make[2]: *** [Makefile:599: halt.o] Error 1
> make[2]: *** Waiting for unfinished jobs....
> /usr/bin/ld: /usr/lib64/gcc/x86_64-slackware-linux/13.2.0/../../../../lib64/crt1.o: in function `_start':
> /tmp/glibc-2.38/csu/../sysdeps/x86_64/start.S:103:(.text+0x17): undefined reference to `main'
> collect2: error: ld returned 1 exit status
> make[2]: *** [Makefile:581: nut] Error 1
> make[2]: Leaving directory '/usr/local/src/nut-2.8.1/scripts/Windows'
> make[1]: *** [Makefile:558: all-recursive] Error 1
> make[1]: Leaving directory '/usr/local/src/nut-2.8.1/scripts'
> make: *** [Makefile:781: all-recursive] Error 1
>

jimklimov, 2023-11-01:
CC @desertwitch - did you see anything like this on your Slackware?

@teoberi : Looking at `scripts/Windows/Makefile.am`, this component gets built `if HAVE_MINGW_RESGEN`... can you check in `config.log` whether `windmc` and `windres` were found on your system (and if they have to do with MinGW and/or with some Windows resource file processing at all?

desertwitch, 2023-11-01:
Yes, I've also encountered this the first time setting my build environment up on Slackware (but forgot about it after)
I'll have to dig up what caused this (it's something to do with windres) - will report back ASAP

teoberi, 2023-11-01:
> CC @desertwitch - did you see anything like this on your Slackware?
>
> @teoberi : Looking at `scripts/Windows/Makefile.am`, this component gets built `if HAVE_MINGW_RESGEN`... can you check in [...]

From config.log:

> configure:45579: checking for windmc
> configure:45600: found /usr/bin/windmc
> configure:45611: result: windmc
> configure:45681: checking for windres
> configure:45702: found /usr/bin/windres
> configure:45713: result: windres

The two files are included in the [binutils-2.41-x86_64-1](https://packages.slackware.com/?r=slackware64-current&p=binutils-2.41-x86_64-1.txz&f) package.

desertwitch, 2023-11-01:
> From config.log:
>
>
> The two files are included in the [binutils-2.41-x86_64-1](https://packages.slackware.com/?r=slackware64-current&p=binutils-2.41-x86_64-1.txz&f) package.

Yes, I remember this now... "full" Slackware contains those Windows binaries but not the required development files. You can workaround this by renaming /usr/bin/windres and /usr/bin/windmc to something else (e.g. windres.bak/windmc.bak) for the duration that you are building NUT. I would also recommend to just "make" instead of "make -j" or "make --jobs", simultaneous make processes introduced some problems for me last year as far as I remember (as opposed to just slow "make").

teoberi, 2023-11-01:
> Yes, I remember this now... "full" Slackware contains those Windows binaries but not the required development files. You can workaround this by renaming /usr/bin/windres and /usr/bin/windmc to [...]

That's what I thought too, although it's not a very elegant solution.
Now it can be used to test the new version, but the final solution should be different, if a Linux operating system is found.

desertwitch, 2023-11-01:
> That's what I thought too, although it's not a very elegant solution. Now it can be used to test the new version, but the final solution should be different, if a Linux operating system is found.

Probably something along the lines of `configure` also checking for some header files (which are only part of MinGW but not GNU Binutils) to determine a MinGW installation rather than just `windres` + `windmc` (present in both MinGW and GNU Binutils). But this is just my theory - Slackware seems to have GNU Binutils but not MinGW in a default "full" installation.

jimklimov, 2023-11-01:
Well, finding these on Linux is perfectly fine... if you've set it up for cross-builds - that is how NUT for Windows originally happened :)

Here it seems to be half-set-up for something of the sort, and that causes the problem. Still, I guess the Makefile clause could use some different toggle to check if we are actually building for Windows (on whatever host platform).

jimklimov, 2023-11-01:
Thanks for the report and quick checks; PR brewing.

desertwitch, 2023-11-01:
> Well, finding these on Linux is perfectly fine... if you've set it up for cross-builds - that is how NUT for Windows originally happened :)
>
> Here it seems to be half-set-up for something of the [...]

What I'm curious about is that GNU Binutils seem to by default include both `windres` and `windmc`, the presence of which seems to trigger NUT `configure` into initiating a Windows build (even when not explicitly cross-building for it). Do other distributions usually ship with both GNU Binutils + MinGW so that this is/was not an issue on other distributions?

jimklimov, 2023-11-01:
Can't really say. On illumos I see it was present in at least binutils-2.25.1 sources back in 2016, but not built or installed. Probably not picked for build due to some run-time `configure` choices (e.g. lack of Windows headers? ;) ), don't see any particular prohibition in recipe.

````
configure:9371: checking for windres                                                              
configure:9401: result: no                                                                        
configure:9512: checking for windmc                                                               
configure:9542: result: no      
...
configure:13456: checking for windres                                                             
configure:13486: result: no                                                                       
configure:13686: checking for windmc                                                              
configure:13716: result: no    
...
configure:14462: checking where to find the target windres                                        
configure:14485: result: just compiled                                                            
configure:14504: checking where to find the target windmc                                         
configure:14527: result: just compiled    
````

Never got built though. Odd...

jimklimov, 2023-11-01:
Can you please give it a shot, if the changes from PR #2143 fix it for you?

teoberi, 2023-11-01:
> Making all in Windows
> make[2]: Entering directory '/tmp/nut-2.8.1/scripts/Windows'
> windmc winevent.mc
> windmc winevent.mc
>   CC       halt.o
> windres winevent.rc winevent.o
>   CC       wininit.o
> windres: can't get BFD_RELOC_RVA relocation type: cause of error unknown
> make[2]: *** [Makefile:851: winevent.o] Error 1
> make[2]: *** Waiting for unfinished jobs....
> halt.c:30:10: fatal error: windows.h: No such file or directory
>    30 | #include <windows.h>
>       |          ^~~~~~~~~~~
> compilation terminated.
> make[2]: *** [Makefile:599: halt.o] Error 1
> make[2]: Leaving directory '/tmp/nut-2.8.1/scripts/Windows'
> make[1]: *** [Makefile:558: all-recursive] Error 1
> make[1]: Leaving directory '/tmp/nut-2.8.1/scripts'
> make: *** [Makefile:781: all-recursive] Error 1

jimklimov, 2023-11-01:
Hoping to reproduce locally, without a "broken phone" back and forth... but hey, while I'm no stranger to interesting systems a bit stuck in their 90's, getting to https://mirrors.slackware.com/slackware/slackware-iso/slackware64-15.0-iso/ from http://www.slackware.com/ was a bit of an adventure. Took a while to notice http://www.slackware.com/getslack/ link there.

Side note: Firefox asks for an external handler to open `ftp://...` URLs. Sigh.

jimklimov, 2023-11-01:
I suppose at least `HAVE_WINDOWS` should certainly mean we *can* build for it to the best of our knowledge (that's what `HAVE_*` are supposed to mean at least; at most that the user also asked for it and we can fulfill - but that "can not" case should be filtered out by `configure` as a `--enable-xxx=yes` vs. inability).

jimklimov, 2023-11-01:
As for (not-)configuring for something specific, that's a mixed bag. Normally `configure.ac` scripts check the current system circumstances (what we can) and their CLI flags for caller's desires (what we explicitly want) or lack thereof (don't clearly care), and try to match the two or three inputs. Roughly speaking, if we had an `--enable-windows` flag (but then why not `--enable-linux`, `--enable-solaris`, `--enable-bsd`, or deeper into distros and releases?) that could be a divergence point for this.

Until this target, all others (regularly hit by CI farm) could differentiate "POSIX" vs "non-POSIX must mean WIN32" sort of builds. Not perfectly as the dichotomy goes, but practically successfully.

desertwitch, 2023-11-01:
Makes sense as far as configuration and checking if we can build for it goes, but should it actually start building Windows files (as it did for the user) merely due to MinGW presence (HAVE_WINDOWS) on the building system, despite the actual build/host/target all being - say - Linux i386?

PR #2143 (2023-11-02): Adapt to builds with Slackware 15
Started as `scripts/Windows/Makefile.am: limit binary builds to cases where we HAVE_WINDOWS` and as such -

Closes: #2142
closed 2023-11-02 by jimklimov