networkupstools/nut #2145: [Question] Is it possible to change the POLLFREQ on the official docker image?

sblantipodi, 2023-11-03:
As title.
I'm using the official docker image:

https://hub.docker.com/r/instantlinux/nut-upsd

when I run this command:
`upsc Cyberpower_PR1500ELCD@localhost`

I get this output:
```
Init SSL without certificate database
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 20
battery.mfr.date: CPS
battery.runtime: 5730
battery.runtime.low: 300
battery.type: PbAcid
battery.voltage: 24.0
battery.voltage.nominal: 24
device.mfr: CPS
device.model: PR1500ELCD
device.serial: PRTGN2000005
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: /dev/bus/usb/001/003
driver.parameter.synchronous: auto
driver.version: 2.8.0
driver.version.data: CyberPower HID 0.6
driver.version.internal: 0.47
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.voltage: 234.0
input.voltage.nominal: 230
output.voltage: 265.0
ups.beeper.status: disabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 17
ups.mfr: CPS
ups.model: PR1500ELCD
ups.productid: 0601
ups.realpower.nominal: 1350
ups.serial: PRTGN2000005
ups.status: OL
ups.test.result: Done and passed
ups.timer.shutdown: -60
ups.timer.start: -60
ups.vendorid: 0764
```
Can I change the pollfreq that is set at 30?

Even if my pollinterval is set at 2, it seems that I get only 1 new reading every 30 seconds...

Thanks!

jimklimov, 2023-11-03:
Better ask with instantlinux - it is *their* (probably-)official image; NUT so far does not publish any.

Also as docs note, e.g. https://github.com/networkupstools/nut/blob/master/docs/man/ups.conf.txt#L95-L105 - `pollinterval` sets the light-weight "status" update frequency, vs. possibly heavier (on processing side) `pollfreq` defined by some but not all drivers.

sblantipodi, 2023-11-07:
hope to see an official container from NUT in the future :)

closed 2023-11-07 by sblantipodi