networkupstools/nut #2150: Debian 12 nut-monitor reports connection refused on reboot - requires a nut-server restart to resolve.

pjkerly, 2023-11-07:
After rebooting Debian 12 Bookworm, my nut-server appears to be unreachable from the nut-client until I restart the nut-server

From the output from below.  I show the status of the nut-server.  It appears to be running correctly.  Using upsc, I get the info from the UPS system and the results appear correct.  Then getting the status from nut-client, you can see repeated Connection Refused messages.    After I restart the nut-server then check the nut-client status, the communications are  established and my other remote systems get the correct updates.  NUT is setup as a MODE=netserver

I'm using the same configuration for NUT config files that I used on Debian 11 and it worked correctly.  Debian 12 is a new installation with no explicit changes to the NUT services or configurations from my working version.

Any pointers, feedback would be appreciated.

sudo service nut-server status

● nut-server.service - Network UPS Tools - power devices information server
     Loaded: loaded (/lib/systemd/system/nut-server.service; enabled; preset: enabled)
     Active: active (running) since Mon 2023-11-06 19:25:05 EST; 51s ago
   Main PID: 1038 (upsd)
      Tasks: 1 (limit: 18981)
     Memory: 1.2M
     CGroup: /system.slice/nut-server.service
             └─1038 /lib/nut/upsd -F

Nov 06 19:25:05 homeassistant nut-server[1038]: listening on ::1 port 3493
Nov 06 19:25:05 homeassistant nut-server[1038]: not listening on 10.0.1.8 port 3493
Nov 06 19:25:05 homeassistant nut-server[1038]: listening on 127.0.0.1 port 3493
Nov 06 19:25:05 homeassistant upsd[1038]: listening on ::1 port 3493
Nov 06 19:25:05 homeassistant nut-server[1038]: Connected to UPS [cyberpower]: usbhid-ups-cyberpower
Nov 06 19:25:05 homeassistant upsd[1038]: not listening on 10.0.1.8 port 3493
Nov 06 19:25:05 homeassistant upsd[1038]: listening on 127.0.0.1 port 3493
Nov 06 19:25:05 homeassistant upsd[1038]: Connected to UPS [cyberpower]: usbhid-ups-cyberpower
Nov 06 19:25:05 homeassistant upsd[1038]: Running as foreground process, not saving a PID file
Nov 06 19:25:05 homeassistant nut-server[1038]: Running as foreground process, not saving a PID file

upsc cyberpower

Init SSL without certificate database
battery.charge: 100
battery.charge.low: 20
battery.charge.warning: 40
battery.mfr.date: CPS
battery.runtime: 8880
battery.runtime.low: 300
battery.type: PbAcid
battery.voltage: 24.0
battery.voltage.nominal: 24
device.mfr: CPS
device.model: CP1500PFCLCD
device.serial: 000000000000
device.type: ups
driver.name: usbhid-ups
driver.parameter.offdelay: 600
driver.parameter.ondelay: 900
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 15
driver.parameter.port: auto
driver.parameter.synchronous: auto
driver.parameter.vendorid: 0764
driver.version: 2.8.0
driver.version.data: CyberPower HID 0.6
driver.version.internal: 0.47
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.transfer.high: 139
input.transfer.low: 88
input.voltage: 121.0
input.voltage.nominal: 120
output.voltage: 121.0
ups.beeper.status: enabled
ups.delay.shutdown: 600
ups.delay.start: 900
ups.load: 3
ups.mfr: CPS
ups.model: CP1500PFCLCD
ups.productid: 0501
ups.realpower.nominal: 900
ups.serial: 000000000000
ups.status: OL
ups.test.result: Done and passed
ups.timer.shutdown: -60
ups.timer.start: -60
ups.vendorid: 0764

sudo service nut-client status

● nut-monitor.service - Network UPS Tools - power device monitor and shutdown controller
     Loaded: loaded (/lib/systemd/system/nut-monitor.service; enabled; preset: enabled)
     Active: active (running) since Mon 2023-11-06 19:25:05 EST; 2min 9s ago
   Main PID: 1039 (upsmon)
      Tasks: 2 (limit: 18981)
     Memory: 3.5M
     CGroup: /system.slice/nut-monitor.service
             ├─1039 /lib/nut/upsmon -F
             └─1041 /lib/nut/upsmon -F

Nov 06 19:26:25 homeassistant nut-monitor[1041]: UPS [cyberpower@10.0.1.8]: connect failed: Connection failure: Connection refused
Nov 06 19:26:30 homeassistant nut-monitor[1041]: UPS [cyberpower@10.0.1.8]: connect failed: Connection failure: Connection refused
Nov 06 19:26:35 homeassistant nut-monitor[1041]: UPS [cyberpower@10.0.1.8]: connect failed: Connection failure: Connection refused
Nov 06 19:26:40 homeassistant nut-monitor[1041]: UPS [cyberpower@10.0.1.8]: connect failed: Connection failure: Connection refused
Nov 06 19:26:45 homeassistant nut-monitor[1041]: UPS [cyberpower@10.0.1.8]: connect failed: Connection failure: Connection refused
Nov 06 19:26:50 homeassistant nut-monitor[1041]: UPS [cyberpower@10.0.1.8]: connect failed: Connection failure: Connection refused
Nov 06 19:26:55 homeassistant nut-monitor[1041]: UPS [cyberpower@10.0.1.8]: connect failed: Connection failure: Connection refused
Nov 06 19:27:00 homeassistant nut-monitor[1041]: UPS [cyberpower@10.0.1.8]: connect failed: Connection failure: Connection refused
Nov 06 19:27:05 homeassistant nut-monitor[1041]: UPS [cyberpower@10.0.1.8]: connect failed: Connection failure: Connection refused
Nov 06 19:27:10 homeassistant nut-monitor[1041]: UPS [cyberpower@10.0.1.8]: connect failed: Connection failure: Connection refused

sudo service nut-server restart

sudo service nut-server status

● nut-server.service - Network UPS Tools - power devices information server
     Loaded: loaded (/lib/systemd/system/nut-server.service; enabled; preset: enabled)
     Active: active (running) since Mon 2023-11-06 19:31:10 EST; 24s ago
   Main PID: 10862 (upsd)
      Tasks: 1 (limit: 18981)
     Memory: 584.0K
     CGroup: /system.slice/nut-server.service
             └─10862 /lib/nut/upsd -F

Nov 06 19:31:10 homeassistant nut-server[10862]: listening on 127.0.0.1 port 3493
Nov 06 19:31:10 homeassistant upsd[10862]: listening on ::1 port 3493
Nov 06 19:31:10 homeassistant nut-server[10862]: Connected to UPS [cyberpower]: usbhid-ups-cyberpower
Nov 06 19:31:10 homeassistant upsd[10862]: listening on 10.0.1.8 port 3493
Nov 06 19:31:10 homeassistant upsd[10862]: listening on 127.0.0.1 port 3493
Nov 06 19:31:10 homeassistant upsd[10862]: Connected to UPS [cyberpower]: usbhid-ups-cyberpower
Nov 06 19:31:10 homeassistant upsd[10862]: Running as foreground process, not saving a PID file
Nov 06 19:31:10 homeassistant nut-server[10862]: Running as foreground process, not saving a PID file
Nov 06 19:31:15 homeassistant nut-server[10862]: User upsmon@10.0.1.8 logged into UPS [cyberpower]
Nov 06 19:31:15 homeassistant upsd[10862]: User upsmon@10.0.1.8 logged into UPS [cyberpower]

sudo service nut-client status

● nut-monitor.service - Network UPS Tools - power device monitor and shutdown controller
     Loaded: loaded (/lib/systemd/system/nut-monitor.service; enabled; preset: enabled)
     Active: active (running) since Mon 2023-11-06 19:25:05 EST; 7min ago
   Main PID: 1039 (upsmon)
      Tasks: 2 (limit: 18981)
     Memory: 3.5M
     CGroup: /system.slice/nut-monitor.service
             ├─1039 /lib/nut/upsmon -F
             └─1041 /lib/nut/upsmon -F

Nov 06 19:30:35 homeassistant nut-monitor[1041]: UPS [cyberpower@10.0.1.8]: connect failed: Connection failure: Connection refused
Nov 06 19:30:40 homeassistant nut-monitor[1041]: UPS [cyberpower@10.0.1.8]: connect failed: Connection failure: Connection refused
Nov 06 19:30:45 homeassistant nut-monitor[1041]: UPS [cyberpower@10.0.1.8]: connect failed: Connection failure: Connection refused
Nov 06 19:30:50 homeassistant nut-monitor[1041]: UPS [cyberpower@10.0.1.8]: connect failed: Connection failure: Connection refused
Nov 06 19:30:55 homeassistant nut-monitor[1041]: UPS [cyberpower@10.0.1.8]: connect failed: Connection failure: Connection refused
Nov 06 19:31:00 homeassistant nut-monitor[1041]: UPS [cyberpower@10.0.1.8]: connect failed: Connection failure: Connection refused
Nov 06 19:31:05 homeassistant nut-monitor[1041]: UPS [cyberpower@10.0.1.8]: connect failed: Connection failure: Connection refused
Nov 06 19:31:10 homeassistant nut-monitor[1041]: UPS [cyberpower@10.0.1.8]: connect failed: Connection failure: Connection refused
Nov 06 19:31:15 homeassistant nut-monitor[1041]: Communications with UPS cyberpower@10.0.1.8 established
Nov 06 19:31:15 homeassistant nut-monitor[10864]: Network UPS Tools upsmon 2.8.0

pjkerly, 2023-11-07:
I just noticed in the reboot nut-server status ... says that it is NOT listening on my ip address.  It is only listening when I do the restart of the nut-server.  I don't know how to fix this issue or if it is a bug or why I have to restart to get it to work.  Is it possible the network is not fully enabled or having received the dhcpd ip address?  Again, my previous Debian 11 was configured has a dhcp for ip address.

My other question is ... should the process be running as a foreground process?

Nov 06 19:25:05 homeassistant nut-server[1038]: listening on ::1 port 3493
Nov 06 19:25:05 homeassistant nut-server[1038]: not listening on 10.0.1.8 port 3493
Nov 06 19:25:05 homeassistant nut-server[1038]: listening on 127.0.0.1 port 3493
Nov 06 19:25:05 homeassistant upsd[1038]: listening on ::1 port 3493
Nov 06 19:25:05 homeassistant nut-server[1038]: Connected to UPS [cyberpower]: usbhid-ups-cyberpower
Nov 06 19:25:05 homeassistant upsd[1038]: not listening on 10.0.1.8 port 3493
Nov 06 19:25:05 homeassistant upsd[1038]: listening on 127.0.0.1 port 3493
Nov 06 19:25:05 homeassistant upsd[1038]: Connected to UPS [cyberpower]: usbhid-ups-cyberpower
Nov 06 19:25:05 homeassistant upsd[1038]: Running as foreground process, not saving a PID file
Nov 06 19:25:05 homeassistant nut-server[1038]: Running as foreground process, not saving a PID file

pjkerly, 2023-11-07:
Resolved ... This allows the network to be up and running before trying to start the nut server

created the file -> `/etc/systemd/system/nut-server.service.d/network.conf` with:
[Unit]
Requires=network-online.target
After=network-online.target

closed 2023-11-07 by pjkerly

jimklimov, 2023-11-07:
Interestingly, this question crosses paths with a recently revived #723 which regards what deterministic behavior we want NUT data server `upsd` to follow when an IP address from the LISTEN directive is busy or absent. It is a bit borked so far even in master (note that the current NUT release is 2.8.1 by the way) but would likely cause the data server to not start in such situation when fixed. By coincidence, systemd would revive it until an IP address appears and the daemon no longer aborts, I suppose.

"Foregrounding" allows the NUT daemons to not fork away so far that systemd has a hard time following which processes it has to watch for the health of a service.

What can be done quickly is a multi-layer question/answer:
* Why not connect with the local(?) `upsmon` just to the `upsname@localhost`?
* Check how the service dependencies of `nut-server.service` are defined in your system - does it request to only start after `network-online.service` or some such guarantee that your system has the IP address?
    * Whether a server should rely on DHCP-assigned addresses is an argument for another day.
* Packaging defaults may differ from what NUT "vanilla" offers

jimklimov, 2023-11-07:
Congrats that you've solved this. I see there's a comment to this effect in the unit file (vanilla at least) - did it help?

https://github.com/networkupstools/nut/blob/master/scripts/systemd/nut-server.service.in#L14-L24

pjkerly, 2023-11-07:
It absolutely did.  I was initially lost in exactly what the problem was and I missed that it wasn't listening on my ip address.  But once I realized the cause it helped narrow my googling ... and then found this suggestion.   But think you for taking the time to read and offer suggestions.

xaxexa, 2024-03-25:
What helped me was adding “LISTEN to localhost 3493” to the /etc/nut/upsd.conf file