networkupstools/nut #2151: GS-1KVAS-RK - Vultech Server Series RACK 1000VA

grydan, 2023-11-07:
Hi , 
I have a problem with this model. Here the URL [https://www.vultech.it/it/serie-server/972-gruppo-di-continuit%C3%A0-server-series-rack-1000va-gs-1kvas-rk-onda-sinusoidale-8052780306003.html](url)

I have a new rasperry pi4 and I have installed 

**lsusb**

```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
**Bus 001 Device 010: ID 0925:1234 Lakeview Research STD UPS MON V2.0**
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
**lsusb -vvv -d0925:**

```
Bus 001 Device 010: ID 0925:1234 Lakeview Research STD UPS MON V2.0
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0         8
  idVendor           0x0925 Lakeview Research
  idProduct          0x1234
  bcdDevice            0.01
  iManufacturer           0
  iProduct                1 STD UPS MON V2.0
  iSerial                 2 J9552E6E43FBC05
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0022
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0
      bInterfaceProtocol      0
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      78
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0006  1x 6 bytes
        bInterval               1
```

**/etc/nut/ups.conf**

```
[vultech]
driver = nutdrv_qx
desc = "Vultech GS-1KVAS-RK"
port = auto
vendorid = "0925"
productid = "1234"
subdriver = "armac"
product = "STD UPS MON V2.0"
bus = 001

```

**sudo /usr/lib/nut/nutdrv_qx -DDDD -a vultech**

```
Network UPS Tools - Generic Q* USB/Serial driver 0.32 (2.8.0)
USB communication driver (libusb 1.0) 0.43
   0.000000     [D3] do_global_args: var='maxretry' val='3'
   0.000704     [D3] main_arg: var='driver' val='nutdrv_qx'
   0.000904     [D3] main_arg: var='desc' val='Vultech GS-1KVAS-RK'
   0.001083     [D3] main_arg: var='port' val='auto'
   0.001307     [D3] main_arg: var='vendorid' val='0925'
   0.001510     [D3] main_arg: var='productid' val='1234'
   0.001698     [D3] main_arg: var='subdriver' val='armac'
   0.001916     [D3] main_arg: var='product' val='STD UPS MON V2.0'
   0.002113     [D3] main_arg: var='bus' val='001'
   0.002357     [D1] debug level is '4'
   0.003861     [D1] upsdrv_initups...
   0.029145     [D2] Checking device 1 of 4 (1D6B/0003)
   0.029357     [D1] Failed to open device (1D6B/0003), skipping: Access denied (insufficient permissions)
   0.029812     [D2] Checking device 2 of 4 (0925/1234)
   0.037222     [D2] - VendorID: 0925
   0.037302     [D2] - ProductID: 1234
   0.037709     [D2] - Manufacturer: unknown
   0.037779     [D2] - Product: STD UPS MON V2.0
   0.038156     [D2] - Serial Number: J9552E6E43FBC05
   0.038225     [D2] - Bus: 001
   0.038595     [D2] - Device: unknown
   0.038662     [D2] - Device release number: 0001
   0.039036     [D2] Trying to match device
   0.039105     [D3] match_function_regex: matching a device...
   0.039696     [D2] Device matches
   0.039769     [D2] Reading first configuration descriptor
   0.039892     [D3] libusb_kernel_driver_active() returned 0
   0.039938     [D2] failed to claim USB device: Resource busy
   0.039979     [D2] Kernel driver already detached
   0.040019     [D2] failed to claim USB device: Resource busy
   0.040058     [D2] Kernel driver already detached
   0.040097     [D2] failed to claim USB device: Resource busy
   0.040139     [D2] Kernel driver already detached
   0.040178     [D2] failed to claim USB device: Resource busy
   0.040218     [D2] Kernel driver already detached
   0.040313     Can't claim USB device [0925:1234]@0/0: Entity not found
```

I have found a similar thread problem here [https://github.com/networkupstools/nut/issues/1978](url)

grydan, 2023-11-07:
**sudo /usr/lib/nut/nutdrv_qx -DDDD -a vultech**

```
Network UPS Tools - Generic Q* USB/Serial driver 0.32 (2.8.0)
USB communication driver (libusb 1.0) 0.43
   0.000000     [D3] do_global_args: var='maxretry' val='3'
   0.000704     [D3] main_arg: var='driver' val='nutdrv_qx'
   0.000904     [D3] main_arg: var='desc' val='Vultech GS-1KVAS-RK'
   0.001083     [D3] main_arg: var='port' val='auto'
   0.001307     [D3] main_arg: var='vendorid' val='0925'
   0.001510     [D3] main_arg: var='productid' val='1234'
   0.001698     [D3] main_arg: var='subdriver' val='armac'
   0.001916     [D3] main_arg: var='product' val='STD UPS MON V2.0'
   0.002113     [D3] main_arg: var='bus' val='001'
   0.002357     [D1] debug level is '4'
   0.003861     [D1] upsdrv_initups...
   0.029145     [D2] Checking device 1 of 4 (1D6B/0003)
   0.029357     [D1] Failed to open device (1D6B/0003), skipping: Access denied (insufficient permissions)
   0.029812     [D2] Checking device 2 of 4 (0925/1234)
   0.037222     [D2] - VendorID: 0925
   0.037302     [D2] - ProductID: 1234
   0.037709     [D2] - Manufacturer: unknown
   0.037779     [D2] - Product: STD UPS MON V2.0
   0.038156     [D2] - Serial Number: J9552E6E43FBC05
   0.038225     [D2] - Bus: 001
   0.038595     [D2] - Device: unknown
   0.038662     [D2] - Device release number: 0001
   0.039036     [D2] Trying to match device
   0.039105     [D3] match_function_regex: matching a device...
   0.039696     [D2] Device matches
   0.039769     [D2] Reading first configuration descriptor
   0.039892     [D3] libusb_kernel_driver_active() returned 0
   0.039938     [D2] failed to claim USB device: Resource busy
   0.039979     [D2] Kernel driver already detached
   0.040019     [D2] failed to claim USB device: Resource busy
   0.040058     [D2] Kernel driver already detached
   0.040097     [D2] failed to claim USB device: Resource busy
   0.040139     [D2] Kernel driver already detached
   0.040178     [D2] failed to claim USB device: Resource busy
   0.040218     [D2] Kernel driver already detached
   0.040313     Can't claim USB device [0925:1234]@0/0: Entity not found
```

After remove USB device and re-connect I re-rerun the command and obtain this
**sudo /usr/lib/nut/nutdrv_qx -DDDD -a vultech**
```
Network UPS Tools - Generic Q* USB/Serial driver 0.32 (2.8.0)
USB communication driver (libusb 1.0) 0.43
   0.000000     [D3] do_global_args: var='maxretry' val='3'
   0.002399     [D3] main_arg: var='driver' val='nutdrv_qx'
   0.002559     [D3] main_arg: var='desc' val='Vultech GS-1KVAS-RK'
   0.003260     [D3] main_arg: var='port' val='auto'
   0.003900     [D3] main_arg: var='vendorid' val='0925'
   0.004032     [D3] main_arg: var='productid' val='1234'
   0.004328     [D3] main_arg: var='subdriver' val='armac'
   0.004486     [D3] main_arg: var='product' val='STD UPS MON V2.0'
   0.004621     [D3] main_arg: var='bus' val='001'
   0.004745     [D1] debug level is '4'
   0.006250     [D1] upsdrv_initups...
   0.021625     [D2] Checking device 1 of 4 (1D6B/0003)
   0.021853     [D1] Failed to open device (1D6B/0003), skipping: Access denied (insufficient permissions)
   0.022324     [D2] Checking device 2 of 4 (0925/1234)
   0.030507     [D2] - VendorID: 0925
   0.030629     [D2] - ProductID: 1234
   0.031061     [D2] - Manufacturer: unknown
   0.031106     [D2] - Product: STD UPS MON V2.0
   0.031146     [D2] - Serial Number: J9552E6E43FBC05
   0.031184     [D2] - Bus: 001
   0.031222     [D2] - Device: unknown
   0.031260     [D2] - Device release number: 0001
[... 232 lines cut ...]
  24.405862     [D4] read: ret 6 buf a6: 28 32 34 30 2e  >(240.<
  24.426901     [D4] read: ret 6 buf 86: 37 20 30 30 30  >7 000<
  24.447943     [D4] read: ret 6 buf a6: 2e 30 20 32 31  >.0 21<
  24.468877     [D4] read: ret 6 buf 86: 39 2e 37 20 30  >9.7 0<
  24.489873     [D4] read: ret 6 buf a6: 32 30 20 35 30  >20 50<
  24.509908     [D4] read: ret 6 buf 86: 2e 30 20 32 2e  >.0 2.<
  24.530880     [D4] read: ret 6 buf a6: 32 31 20 34 34  >21 44<
  24.551909     [D4] read: ret 6 buf 86: 2e 30 20 30 30  >.0 00<
  24.572886     [D4] read: ret 6 buf a6: 30 30 30 30 30  >00000<
000<.076066     [D4] read: ret 6 buf 83: 31 0d 30 30 30  >1
  26.076783     [D1] interrupt read error: Operation timed out (-7)
  26.076902     [D4] qx_process: failed to preprocess answer [input.voltage]
  26.076986     [D4] armac command Q1
  26.150104     [D4] read: ret 6 buf a6: 28 32 34 30 2e  >(240.<
  26.171102     [D4] read: ret 6 buf 86: 36 20 30 30 30  >6 000<
  26.192136     [D4] read: ret 6 buf a6: 2e 30 20 32 31  >.0 21<
  26.213146     [D4] read: ret 6 buf 86: 39 2e 36 20 30  >9.6 0<
  26.234145     [D4] read: ret 6 buf a6: 32 30 20 34 39  >20 49<
  26.254110     [D4] read: ret 6 buf 86: 2e 39 20 32 2e  >.9 2.<
  26.275142     [D4] read: ret 6 buf a6: 32 31 20 34 34  >21 44<
  26.296117     [D4] read: ret 6 buf 86: 2e 30 20 30 30  >.0 00<
  26.317121     [D4] read: ret 6 buf a6: 30 30 30 30 30  >00000<
000<.816256     [D4] read: ret 6 buf 83: 31 0d 30 30 30  >1
  27.816894     [D1] interrupt read error: Operation timed out (-7)
```

jimklimov, 2023-11-07:
I am not sure the issues are that similar, but it should not hurt to try newer NUT in any case. See e.g. https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests for notes on building a configuration similar to your current (e.g. packaged) installation which you can test right from the build workspace, and if it works - replace the existing setup by `make install`.

For context, in #1978 the problem is about vendors being creative with the protocol (something your second post with its "screenshots" probably confirms as a related problem, solved there and included in NUT v2.8.1 release).

Your original post however, about "device busy", is something different. Not sure OTOH why that happens - maybe some `udev` rules were not right and USB VID:PID remained occupied by the system or some other program. Maybe the system-started driver instance kept it busy and could not be killed (or was killed and the OS could not reap the resources back somehow)?..

grydan, 2023-11-07:
OK I have follow the url, thank... but If now I run command from /tmp/ directory:

 **sudo ./drivers/nutdrv_qx -a vultech**
```

Network UPS Tools - Generic Q* Serial driver 0.36 (2.8.1-31-ge451a97bc)

Fatal error: 'vendorid' is not a valid variable name for this driver.

Look in the man page or call this driver with -h for a list of
valid variable names and flags.
upsnotify: failed to notify about state 4: no notification tech defined, will not spam more about it
```

jimklimov, 2023-11-07:
That's strange :\ Did you get libusb dev libraries+headers package? Is USB support reported as enabled after NUT configuration?

jimklimov, 2023-11-07:
"No notification tech" suggests you lack libsystemd-dev too?

grydan, 2023-11-08:
Ok now I have redone all steps...and I'm occur in the problem **Resource busy** .. argggggh

```
sudo ./drivers/nutdrv_qx -s tempups     -d1 -DDDDDD -x port=auto     -x vendorid=0925 -x productid=1234     -x subdriver=armac
Network UPS Tools - Generic Q* USB/Serial driver 0.36 (2.8.1-35-g8c913eda4)
USB communication driver (libusb 1.0) 0.46
   0.000000     [D3] main_arg: var='port' val='auto'
   0.000105     [D6] testinfo_reloadable: var=port, infoname=driver.parameter.port, newval=auto, reloadable=0, reload_flag=0
   0.000162     [D6] testinfo_reloadable: verdict for (re)loading var=port value: 1
   0.000219     [D5] send_to_all: SETINFO driver.parameter.port "auto"
   0.000303     [D3] main_arg: var='vendorid' val='0925'
   0.000369     [D5] send_to_all: SETINFO driver.parameter.vendorid "0925"
   0.000424     [D3] main_arg: var='productid' val='1234'
   0.000499     [D5] send_to_all: SETINFO driver.parameter.productid "1234"
   0.000552     [D3] main_arg: var='subdriver' val='armac'
   0.000610     [D5] send_to_all: SETINFO driver.parameter.subdriver "armac"
   0.000662     [D1] Network UPS Tools version 2.8.1-35-g8c913eda4 (release/snapshot of 2.8.1.1) built with gcc (Debian 12.2.0-14) 12.2.0 and configured with flags: --prefix=/usr --sbindir=/usr/sbin --bindir=/usr/bin --sysconfdir=/etc/nut --with-group=nut --with-user=nut --enable-Wcolor --enable-warnings --enable-Werror --enable-keep_nut_report_feature --with-all=auto --with-cgi=auto --with-serial=auto --with-dev=auto --with-doc=skip --with-nut_monitor=auto --with-pynut=auto --disable-force-nut-version-header --enable-check-NIT --enable-maintainer-mode
   0.000747     [D1] debug level is '6'
   0.000799     [D5] send_to_all: SETINFO driver.debug "6"
   0.000852     [D5] send_to_all: SETFLAGS driver.debug RW NUMBER
   0.002153     [D1] Succeeded to become_user(nut): now UID=112 GID=122
   0.002230     [D5] send_to_all: SETINFO device.type "ups"
   0.002285     [D5] send_to_all: SETINFO driver.state "init.device"
   0.002334     [D1] upsdrv_initups...
   0.027277     [D2] Checking device 1 of 4 (1D6B/0003)
   0.027451     [D1] Failed to open device (1D6B/0003), skipping: Access denied (insufficient permissions)
   0.027510     [D2] Checking device 2 of 4 (0925/1234)
   0.036862     [D2] - VendorID: 0925
   0.036925     [D2] - ProductID: 1234
   0.036974     [D2] - Manufacturer: unknown
   0.037021     [D2] - Product: STD UPS MON V2.0
   0.037068     [D2] - Serial Number: J9552E6E43FBC05
   0.037115     [D2] - Bus: 001
   0.037161     [D2] - Bus Port: 004
   0.037210     [D2] - Device: 008
   0.037256     [D2] - Device release number: 0001
   0.037302     [D2] Trying to match device
   0.037351     [D3] match_function_regex: matching a device...
   0.037526     [D2] Device matches
   0.037577     [D2] Reading configuration descriptor 1 of 1
   0.037655     [D3] libusb_kernel_driver_active() returned 0: Success
   0.037719     [D2] failed to claim USB device: Resource busy
   0.037785     [D2] Kernel driver already detached
   0.037841     [D2] failed to claim USB device: Resource busy
   0.037897     [D2] Kernel driver already detached
   0.037952     [D2] failed to claim USB device: Resource busy
   0.038008     [D2] Kernel driver already detached
   0.038064     [D2] failed to claim USB device: Resource busy
   0.038124     [D2] Kernel driver already detached
   0.038183     Can't claim USB device [0925:1234]@0/0/0: Entity not found
   0.038250     [D5] send_to_all: SETINFO driver.state "cleanup.exit"
```

grydan, 2023-11-08:
If I re-run the command in another time seem that this work with 2.8.1
What do you think?
I need only solve the problem of **Resource busy** (_It seems that the UPS is constantly connecting and disconnecting....but could it be that I am wrong?_)

**sudo ./drivers/nutdrv_qx -s tempups  -d1 -DDDDDD -x port=auto     -x vendorid=0925 -x productid=1234     -x 
subdriver=armac**
```
Network UPS Tools - Generic Q* USB/Serial driver 0.36 (2.8.1-35-g8c913eda4)
USB communication driver (libusb 1.0) 0.46
   0.000000     [D3] main_arg: var='port' val='auto'
   0.000153     [D6] testinfo_reloadable: var=port, infoname=driver.parameter.port, newval=auto, reloadable=0, reload_flag=0
   0.000210     [D6] testinfo_reloadable: verdict for (re)loading var=port value: 1
   0.000265     [D5] send_to_all: SETINFO driver.parameter.port "auto"
   0.000350     [D3] main_arg: var='vendorid' val='0925'
   0.000414     [D5] send_to_all: SETINFO driver.parameter.vendorid "0925"
   0.000468     [D3] main_arg: var='productid' val='1234'
   0.000577     [D5] send_to_all: SETINFO driver.parameter.productid "1234"
   0.000784     [D3] main_arg: var='subdriver' val='armac'
   0.001060     [D5] send_to_all: SETINFO driver.parameter.subdriver "armac"
   0.001280     [D1] Network UPS Tools version 2.8.1-35-g8c913eda4 (release/snapshot of 2.8.1.1) built with gcc (Debian 12.2.0-14) 12.2.0 and configured with flags: --prefix=/usr --sbindir=/usr/sbin --bindir=/usr/bin --sysconfdir=/etc/nut --with-group=nut --with-user=nut --enable-Wcolor --enable-warnings --enable-Werror --enable-keep_nut_report_feature --with-all=auto --with-cgi=auto --with-serial=auto --with-dev=auto --with-doc=skip --with-nut_monitor=auto --with-pynut=auto --disable-force-nut-version-header --enable-check-NIT --enable-maintainer-mode
   0.001700     [D1] debug level is '6'
   0.001908     [D5] send_to_all: SETINFO driver.debug "6"
   0.002115     [D5] send_to_all: SETFLAGS driver.debug RW NUMBER
   0.003651     [D1] Succeeded to become_user(nut): now UID=112 GID=122
   0.003728     [D5] send_to_all: SETINFO device.type "ups"
   0.003783     [D5] send_to_all: SETINFO driver.state "init.device"
   0.003871     [D1] upsdrv_initups...
   0.028964     [D2] Checking device 1 of 4 (1D6B/0003)
   0.029277     [D1] Failed to open device (1D6B/0003), skipping: Access denied (insufficient permissions)
   0.029487     [D2] Checking device 2 of 4 (0925/1234)
   0.038016     [D2] - VendorID: 0925
   0.038201     [D2] - ProductID: 1234
   0.038349     [D2] - Manufacturer: unknown
   0.038399     [D2] - Product: STD UPS MON V2.0
   0.038475     [D2] - Serial Number: J9552E6E43FBC05
   0.038674     [D2] - Bus: 001
   0.038811     [D2] - Bus Port: 004
   0.038901     [D2] - Device: 008
   0.038951     [D2] - Device release number: 0001
   0.038997     [D2] Trying to match device
   0.039075     [D3] match_function_regex: matching a device...
   0.039551     [D2] Device matches
   0.039632     [D2] Reading configuration descriptor 1 of 1
   0.039744     [D3] libusb_kernel_driver_active() returned 0: Success
   0.040147     [D2] Claimed interface 0 successfully
   0.040427     [D3] nut_usb_set_altinterface: skipped libusb_set_interface_alt_setting(udev, 0, 0)
   0.040540     [D5] send_to_all: SETINFO ups.vendorid "0925"
   0.040672     [D5] send_to_all: SETINFO ups.productid "1234"
   0.040925     [D2] armac: unsupported cmd: QGS
   0.041160     [D2] qx_process_answer: short reply (input.voltage)
   0.041409     [D2] armac: unsupported cmd: QGS
   0.041603     [D2] qx_process_answer: short reply (input.voltage)
   0.041778     [D2] armac: unsupported cmd: QGS
   0.042038     [D2] qx_process_answer: short reply (input.voltage)
   0.042120     [D2] armac: unsupported cmd: M
   0.042174     [D2] voltronic_qs_protocol: invalid protocol [M]
   0.042224     [D4] ups_infoval_set: failed to preprocess value [ups.firmware.aux: M]
   0.042274     [D2] armac: unsupported cmd: M
   0.042404     [D2] voltronic_qs_protocol: invalid protocol [M]
   0.042623     [D4] ups_infoval_set: failed to preprocess value [ups.firmware.aux: M]
   0.042679     [D2] armac: unsupported cmd: M
   0.042782     [D2] voltronic_qs_protocol: invalid protocol [M]
   0.042998     [D4] ups_infoval_set: failed to preprocess value [ups.firmware.aux: M]
   0.043255     [D2] armac: unsupported cmd: M
   0.043480     [D2] voltronic_qs_hex_protocol: invalid protocol [M]
   0.043598     [D4] ups_infoval_set: failed to preprocess value [ups.firmware.aux: M]
   0.043652     [D2] armac: unsupported cmd: M
   0.043704     [D2] voltronic_qs_hex_protocol: invalid protocol [M]
   0.043785     [D4] ups_infoval_set: failed to preprocess value [ups.firmware.aux: M]
   0.044046     [D2] armac: unsupported cmd: M
[... 335 lines cut ...]
driver.version: 2.8.1-35-g8c913eda4
driver.version.data: Megatec 0.07
driver.version.internal: 0.36
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.current.nominal: 4.0
input.frequency: 50.0
input.frequency.nominal: 50
input.voltage: 238.4
input.voltage.fault: 0.0
input.voltage.nominal: 220
output.voltage: 220.0
ups.beeper.status: enabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.firmware: R1.00.47
ups.load: 20
ups.productid: 1234
ups.status: OL
ups.temperature: 43.0
ups.type: online
ups.vendorid: 0925
  13.745162     [D5] send_to_all: SETINFO driver.state "cleanup.upsdrv"
  13.745215     [D1] upsdrv_cleanup...
  13.745978     [D5] send_to_all: SETINFO driver.state "cleanup.exit"
```

grydan, 2023-11-08:
I also have one last question for you....
I installed on my rasperry PI4 nut with the command:
*******
sudo apt install nut
*****
the version that downloads from the official repositories is version 2.8.0.7 ...

Is there an alternative repository from which I can get version 2.8.1-35 which I just tested and recompiled from github?

I don't understand what is the procedure to replace NUT with the most up-to-date version?

grydan, 2023-11-08:
Hi, I have follow After a step
[here ](https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests#replacing-a-systemd-deployment)

after a complete rasperry reboot now work.

Missing only battery information: is normal?

I have also installed the http CGI and this is the result:
[image]
[image]
[image]

grydan, 2023-11-08:
solved this is the way to fix also this:
[battery charge](https://networkupstools.org/docs/man/nutdrv_qx.html#_battery_charge_guesstimation)

jimklimov, 2023-11-09:
Congratulations with your progress and new learnings :)

As for repositories with recent NUT builds packaged - I am not aware of any, at least the NUT CI efforts do not currently produce packages nor containers.

There is a lingering hope to provide "reference" packaging definitions in the source tree and integrate perhaps with OBS or GitHub actions to build them into deb/rpm/... archives. But so far there was not much progress about that.