networkupstools/nut #2162: apcsmart : Inappropriate ioctl for device

ketomagasaki, 2023-11-09:
Hello,

The hardware:
Device: Raspberry Pi 4 Model B
UPS: SMT1500RMI2U

Initially, I connected the UPS to the Raspberry Pi via USB without a card. With the usbhid-ups driver, I managed to make it work, but with limited information provided by the UPS.

I read that I need to add a card: AP9620. I added it and modified the configuration:

ups.conf :
```
pollinterval = 1
maxretry = 3
[ups]
    driver = apcsmart
    port = /dev/bus/usb/001/003
    desc = "Main UPS"

```
But when I start the driver, I get:

```
Network UPS Tools - UPS driver controller 2.8.1
Network UPS Tools - APC Smart protocol driver 3.32 (2.8.1)
APC command table version 3.1
tcgetattr(/dev/bus/usb/001/003): Inappropriate ioctl for device
upsnotify: failed to notify about state 4: no notification tech defined, will not spam more about it
Driver failed to start (exit status=1)
```

Do you have a solution?

Thanks

jimklimov, 2023-11-09:
I believe this is a serial-port driver, so for USB connections (assuming they went for serial-over-usb shim) you'd need something like `port=/dev/ttyISB0`

ketomagasaki, 2023-11-11:
Okay, thanks. I didn't realize that I needed to use the serial port on this expansion card because it also had a USB port. I'm ordering a serial-USB adapter, and I'll test it.

jimklimov, 2023-11-13:
For the USB port(s), can you please also check if it supports the protocol handled by `apc_modbus` introduced with NUT v2.8.1?

ketomagasaki, 2023-11-14:
Ok,

So, first test list on the AP9620 card:

1)
```
[ups]
    driver = apc_modbus
    port = auto
    desc = "Main UPS"
```
Result :
`modbus_connect: unable to connect: No such file or directory`

2)
```
[ups] 
    driver = apcsmart
    port = /dev/bus/usb/001/003
    desc = "Main UPS"
```
Result :
tcgetattr(/dev/bus/usb/001/003): Inappropriate ioctl for device

3)
```
[ups]
    driver = usbhid-ups
    port = auto 
    desc = "Main UPS"
    vendorid = 051d
```
Result :
```
Connected to NUT server localhost in SSL
Certificate verification is disabled
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 50
battery.mfr.date: 2022/09/22
battery.runtime: 3720
battery.runtime.low: 120
battery.temperature: 19.9
battery.type: PbAc
battery.voltage: 27.3
battery.voltage.nominal: 24.0
device.mfr: American Power Conversion
device.model: Legacy Communication Card
device.serial: xxxxxxxxxxxxxxxxx
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 1
driver.parameter.port: auto
driver.parameter.synchronous: auto
driver.parameter.vendorid: 051d
driver.state: quiet
driver.version: 2.8.1
driver.version.data: APC HID 0.100
driver.version.internal: 0.52
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.sensitivity: high
input.transfer.high: 253
input.transfer.low: 207
input.voltage: 231.7
output.current: 1.53
output.frequency: 50.0
output.voltage: 231.7
output.voltage.nominal: 230.0
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.firmware: LCC 03.1 / ID=5004
ups.load: 22.0
ups.mfr: American Power Conversion
ups.mfr.date: 2022/09/22
ups.model: Legacy Communication Card
ups.productid: 0002
ups.serial: xxxxxxxxxxxxx
ups.status: OL
ups.test.result: No test initiated
ups.timer.reboot: -1
ups.timer.shutdown: -1
ups.timer.start: -1
ups.vendorid: 051d
```

Then, the second test list directly on the UPS:
(It's absolutely necessary to remove the AP9620 card; otherwise, the direct USB port is disabled)

1)
```
[ups]
    driver = apc_modbus
    port = auto
    desc = "Main UPS"
```
Result :
`modbus_connect: unable to connect: No such file or directory`

2)
```
[ups] 
    driver = apcsmart
    port = /dev/bus/usb/001/003
    desc = "Main UPS"
```
Result :
tcgetattr(/dev/bus/usb/001/003): Inappropriate ioctl for device

3)
```
[ups]
    driver = usbhid-ups
    port = auto 
    desc = "Main UPS"
    vendorid = 051d
```
Result :
```
Connected to NUT server localhost in SSL
Certificate verification is disabled
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 50
battery.runtime: 3720
battery.runtime.low: 120
battery.type: PbAc
battery.voltage: 27.3
battery.voltage.nominal: 24.0
device.mfr: American Power Conversion
device.model: Smart-UPS 1500
device.serial: xxxxxxxxxxxxxxxx
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 1
driver.parameter.port: auto
driver.parameter.synchronous: auto
driver.parameter.vendorid: 051d
driver.state: quiet
driver.version: 2.8.1
driver.version.data: APC HID 0.100
driver.version.internal: 0.52
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.firmware: UPS 09.3 / ID=18
ups.mfr: American Power Conversion
ups.mfr.date: 2022/09/22
ups.model: Smart-UPS 1500
ups.productid: 0003
ups.serial: xxxxxxxxxxxxxxxxxxxx
ups.status: OL
ups.timer.reboot: -1
ups.timer.shutdown: -1
ups.vendorid: 051d
```

One of the results provides all the information. Using the usbhid-ups driver and with the AP9620 card connected via USB to USB.

closed 2023-11-26 by ketomagasaki