networkupstools/nut #2168: is_ups_critical: seems that UPS [xxx] is OB+LB now, but it is also calibrating : but it's not (longer) true (from 2.8.0)

GreyWolfBW, 2023-11-11:
Hi everyone,

During a power cut long enough to bring my UPS batteries to a critical level, I noticed that my server was not shut down by NUT as I expected.

I think this is a bug in UPSMON since version 2.8.0, here is what I noticed :

First of all, I'm using an **ONLINE-USV XANTO 2000R UPS**, connected to my server via a USB cable and therefore managed by NUT with the **nutdrv_qs** driver with **cypress voltronic** subdriver protocol. The server's operating system is Debian 12 Bookworm.

I've been using this model of UPS since Debian 11 or Debian 10 and NUT without encountering this problem.

On Debian 11 Bullseye, the NUT version supplied is 2.7.4.

I've already used this UPS on this version and management is very satisfactory. There's no relevant battery runtime feedback. So, I prefer to use the "_ignorelb_" flag and set a low battery threshold via "_override.battery.charge.low_" to trigger server shutdown at a predefined battery level.

With Debian 12 Bookworm, the **NUT version is 2.8.0**.

Unfortunately, it didn't work with this version. So, I tried to figure out what was going on.

The log clearly shows this line :
`is_ups_critical: seems that UPS [Onduleur] is OB+LB now, but it is also calibrating - not declaring a critical state`

At first glance, among the changes between these two versions, UPSMON can now manage the "_calibration_" mode of an UPS.

I think it's something universal, but every UPS I know of performs a ten-second battery self-test every one or two weeks. During a self-test, a call to UPSC indicates the status "_OL CAL_" with this model of UPS. UPSMON notices this because a log entry reads "_**UPS Onduleur: calibration in progress**_".

If I understand the UPSMON source code correctly, it records the calibration status in a variable ( `setflag(&ups->status, ST_CAL);` [on line 864 in nut/clients/upsmon.c](https://github.com/networkupstools/nut/blob/ff16dabca191e5fd8ddc20137317bdebee554d8d/clients/upsmon.c#L852C1-L865C2) ). **But at no time do I see it deleting this status ?** And I think this is a problem.

So, when there's a power cut, the UPS switches to battery power and when the battery is low, the _is_ups_critical_ function thinks that the UPS is in calibration and that the server shouldn't be shut down.

Here's an example :

Version tested = v2.8.0 supplied by Debian 12 Bookworm :

> apt policy nut
> nut:
>   Installé : 2.8.0-7
>   Candidat : 2.8.0-7

Contents of my ups.conf file (for this test) :
```
[Onduleur]
override.battery.charge.low=50
override.battery.charge.warning=70
ignorelb
override.ups.delay.shutdown=40
override.ups.delay.start=20
driver=nutdrv_qx
protocol=voltronic
vendorid=0665
productid=5161
subdriver=cypress
port=auto
```

Driver process in debug mode :
I can see the calibration flag set to 1 during the self-test and then return to 0.
From [line 1064 of nut/drivers/nutdrv_qx_voltronic.c](https://github.com/networkupstools/nut/blob/ff16dabca191e5fd8ddc20137317bdebee554d8d/drivers/nutdrv_qx_voltronic.c#L1064C1-L1069C5), I understand that the calibration flag is managed by the 70st offset ([see line 1089](https://github.com/networkupstools/nut/blob/ff16dabca191e5fd8ddc20137317bdebee554d8d/drivers/nutdrv_qx_voltronic.c#L1089)).
```
 319.167600     [D1] upsdrv_updateinfo...
 319.167655     [D1] Quick update...
 319.168047     [D3] send: QGS
 319.483858     [D3] read: (236.6 50.0 229.8 50.0 000.1 002 375.3 379.5 054.7 ---.- 018.4 100000000001
 321.169388     [D1] upsdrv_updateinfo...
 321.169437     [D1] Quick update...
 321.169788     [D3] send: QGS
 321.483880     [D3] read: (236.8 50.0 229.7 50.0 000.0 002 375.5 379.3 054.7 ---.- 018.4 100000000001
 323.171177     [D1] upsdrv_updateinfo...
 323.171227     [D1] Quick update...
 323.171568     [D3] send: QGS
 323.483904     [D3] read: (236.9 50.0 229.5 50.0 000.1 002 373.1 382.6 054.2 ---.- 018.4 100000010001
[... 463 lines cut ...]
```

After compiling the code with this modification, UPSMON works as I expected even after a battery self-test.

However, I don't know if this could have any other impact ?

I've also tried it with tag version "v2.8.1" and the test is also conclusive after compiling ([line 2128](https://github.com/networkupstools/nut/blob/4ba352d8f82e4c51032d4166d4cc3276b31fb3a5/clients/upsmon.c#L2128C1-L2136C25)) :
```
	/* clear these out early if they disappear */
	if (!strstr(status, "LB"))
		clearflag(&ups->status, ST_LOWBATT);
	if (!strstr(status, "FSD"))
		clearflag(&ups->status, ST_FSD);
	if (!strstr(status, "OFF"))
		ups_is_notoff(ups);
	if (!strstr(status, "BYPASS"))
		ups_is_notbypass(ups);
	if (!strstr(status, "CAL"))
		clearflag(&ups->status, ST_CAL);
```

What do you think about it ?

Thanks for reading,
Fabien

desertwitch, 2023-11-12:
Nice observations and proposed fix, but quite surprised this didn't pop up any earlier... 
Seems a CAL flag never gets cleared indeed, hence your problem when your UPS goes OB+LB.

I also think it'd be the right place for the fix, but let's wait what Jim says... :-)

jimklimov, 2023-11-12:
Looks right :)

jimklimov, 2023-11-13:
PR merged

closed 2023-11-13 by jimklimov

geerlingguy, 2025-02-19:
For those running Debian 12, FYI there's an active bug report against the version (2.8.0) which is currently shipping there—and in unstable, currently: [/sbin/upsmon: CAL flag in UPSMON never cleared, shutdown procedure will not be triggered](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1056190)