networkupstools/nut #2174: Errors compiling pipedebug.c on Slackware 15.0 (GCC 11.2.0)

gdsotirov, 2023-11-13:
I'm having the following errors while trying to build [2.8.1](https://github.com/networkupstools/nut/releases/tag/v2.8.1) on Slackware 15.0 with GCC 11.2.0:

```
Making all in server
make[1]: Entering directory '/usr/src/tmp/nut-2.8.1/server'
depbase=`echo pipedebug.o | sed 's|[^/]*$|.deps/&|;s|\.o$||'`;\
i586-slackware-linux-gcc -DHAVE_CONFIG_H -I. -I../include    -I../include   -isystem /usr/local/include  -O3 -march=i586 -mtune=i686 -Wno-reserved-identifier -Wno-unknown-warning-option -std=gnu99 -s -Wno-system-headers -Wall -Wextra -Wsign-compare -pedantic -Wno-error -MT pipedebug.o -MD -MP -MF $depbase.Tpo -c -o pipedebug.o pipedebug.c &&\
mv -f $depbase.Tpo $depbase.Po
pipedebug.c:40:8: error: unknown type name ‘HANDLE’
   40 | static HANDLE pipe_connect(const char *pipefn)
      |        ^~~~~~
pipedebug.c: In function ‘pipe_connect’:
pipedebug.c:42:9: error: unknown type name ‘HANDLE’
   42 |         HANDLE  fd;
      |         ^~~~~~
pipedebug.c:44:9: error: unknown type name ‘BOOL’
   44 |         BOOL    result = FALSE;
      |         ^~~~
pipedebug.c:44:26: error: ‘FALSE’ undeclared (first use in this function)
   44 |         BOOL    result = FALSE;
      |                          ^~~~~
pipedebug.c:44:26: note: each undeclared identifier is reported only once for each function it appears in
pipedebug.c:48:18: warning: implicit declaration of function ‘WaitNamedPipe’ [-Wimplicit-function-declaration]
   48 |         result = WaitNamedPipe(pipename,NMPWAIT_USE_DEFAULT_WAIT);
      |                  ^~~~~~~~~~~~~
pipedebug.c:48:41: error: ‘NMPWAIT_USE_DEFAULT_WAIT’ undeclared (first use in this function)
   48 |         result = WaitNamedPipe(pipename,NMPWAIT_USE_DEFAULT_WAIT);
      |                                         ^~~~~~~~~~~~~~~~~~~~~~~~
pipedebug.c:51:47: warning: implicit declaration of function ‘GetLastError’ [-Wimplicit-function-declaration]
   51 |                 printf("WaitNamedPipe : %d\n",GetLastError());
      |                                               ^~~~~~~~~~~~
pipedebug.c:55:14: warning: implicit declaration of function ‘CreateFile’ [-Wimplicit-function-declaration]
   55 |         fd = CreateFile(
      |              ^~~~~~~~~~
pipedebug.c:57:25: error: ‘GENERIC_READ’ undeclared (first use in this function)
   57 |                         GENERIC_READ |  // read and write access
      |                         ^~~~~~~~~~~~
pipedebug.c:58:25: error: ‘GENERIC_WRITE’ undeclared (first use in this function)
   58 |                         GENERIC_WRITE,
      |                         ^~~~~~~~~~~~~
pipedebug.c:61:25: error: ‘OPEN_EXISTING’ undeclared (first use in this function)
   61 |                         OPEN_EXISTING,  // opens existing pipe
      |                         ^~~~~~~~~~~~~
pipedebug.c:62:25: error: ‘FILE_FLAG_OVERLAPPED’ undeclared (first use in this function)
   62 |                         FILE_FLAG_OVERLAPPED, //  enable async IO
      |                         ^~~~~~~~~~~~~~~~~~~~
pipedebug.c:65:19: error: ‘INVALID_HANDLE_VALUE’ undeclared (first use in this function)
   65 |         if (fd == INVALID_HANDLE_VALUE) {
      |                   ^~~~~~~~~~~~~~~~~~~~
pipedebug.c: At top level:
pipedebug.c:73:34: error: unknown type name ‘DWORD’
   73 | static void read_buf(char * buf, DWORD num)
      |                                  ^~~~~
pipedebug.c:91:1: error: unknown type name ‘DWORD’
   91 | DWORD WINAPI ReadThread( LPVOID lpParameter )
      | ^~~~~
pipedebug.c:91:14: error: expected ‘=’, ‘,’, ‘;’, ‘asm’ or ‘__attribute__’ before ‘ReadThread’
   91 | DWORD WINAPI ReadThread( LPVOID lpParameter )
      |              ^~~~~~~~~~
pipedebug.c:111:1: error: unknown type name ‘DWORD’
  111 | DWORD WINAPI WriteThread( LPVOID lpParameter )
      | ^~~~~
pipedebug.c:111:14: error: expected ‘=’, ‘,’, ‘;’, ‘asm’ or ‘__attribute__’ before ‘WriteThread’
  111 | DWORD WINAPI WriteThread( LPVOID lpParameter )
      |              ^~~~~~~~~~~
pipedebug.c: In function ‘main’:
pipedebug.c:132:9: error: unknown type name ‘HANDLE’
  132 |         HANDLE  pipefd;
      |         ^~~~~~
pipedebug.c:133:9: error: unknown type name ‘HANDLE’
  133 |         HANDLE  thread[2];
      |         ^~~~~~
pipedebug.c:149:21: warning: implicit declaration of function ‘CreateThread’ [-Wimplicit-function-declaration]
  149 |         thread[0] = CreateThread(
      |                     ^~~~~~~~~~~~
pipedebug.c:152:41: error: ‘ReadThread’ undeclared (first use in this function); did you mean ‘readahead’?
  152 |                                         ReadThread, /* func */
      |                                         ^~~~~~~~~~
      |                                         readahead
pipedebug.c:157:22: warning: comparison between pointer and integer
  157 |         if(thread[0] == NULL) {
      |                      ^~
pipedebug.c:165:41: error: ‘WriteThread’ undeclared (first use in this function)
  165 |                                         WriteThread, /* func */
      |                                         ^~~~~~~~~~~
pipedebug.c:170:22: warning: comparison between pointer and integer
  170 |         if(thread[1] == NULL) {
      |                      ^~
pipedebug.c:175:9: warning: implicit declaration of function ‘WaitForMultipleObjects’ [-Wimplicit-function-declaration]
  175 |         WaitForMultipleObjects(2,thread,TRUE,INFINITE);
      |         ^~~~~~~~~~~~~~~~~~~~~~
pipedebug.c:175:41: error: ‘TRUE’ undeclared (first use in this function)
  175 |         WaitForMultipleObjects(2,thread,TRUE,INFINITE);
      |                                         ^~~~
pipedebug.c:175:46: error: ‘INFINITE’ undeclared (first use in this function)
  175 |         WaitForMultipleObjects(2,thread,TRUE,INFINITE);
      |                                              ^~~~~~~~
At top level:
pipedebug.c:27:13: warning: ‘pipe_arg’ defined but not used [-Wunused-function]
   27 | static void pipe_arg(int numarg, char **arg)
      |             ^~~~~~~~
cc1: note: unrecognized command-line option ‘-Wno-unknown-warning-option’ may have been intended to silence earlier diagnostics
cc1: note: unrecognized command-line option ‘-Wno-reserved-identifier’ may have been intended to silence earlier diagnostics
make[1]: *** [Makefile:693: pipedebug.o] Error 1
make[1]: Leaving directory '/usr/src/tmp/nut-2.8.1/server'
make: *** [Makefile:781: all-recursive] Error 1
```

This is after I worked around #2142, because as I see #2143 was not included and would wait 2.8.2 milestone.

jimklimov, 2023-11-13:
Are you building it manually? It is a Windows program... do you have the mingw build enabled?

jimklimov, 2023-11-13:
Also not sure what to think of "as I see #2143 was not included and would wait 2.8.2 milestone" - included where? in 2.8.1? That PR was made after the release, and release tarballs are published and set in stone after that :)

jimklimov, 2023-11-13:
Or in other words, is the problem reproducible with current master branch? Per those earlier issues, we know Slackware and its seeming support for Windows builds had issues fixed later...

gdsotirov, 2023-11-14:
@jimklimov I'm not sure I understand what is _a Windows program_... NUT 2.8.0 builds just fine and I've been building NUT on Slackware Linux since 2006-th (2.0.x). I'm also not sure what you mean by _manually_... configure && make && make install is what I do (with a script).

> That PR was made after the release, and release tarballs are published and set in stone after that :)

Yes, that was exactly what I've meant as I had to work around this problem before getting to the compilation errors reported here...

> Or in other words, is the problem reproducible with current master branch?

I'm not interested in building master, but only specific stable releases, and as far as I'm aware the latest is 2.8.1. Anyway, I've tried to build from latest commit tarball and pipedebug.c compiles without any warning or error (using the same command line), so the solution should be in the source. I'll try to search for it...

jimklimov, 2023-11-14:
> what is a Windows program?

It is one that uses data types and commands declared by `windows.h` and half a dozen others (for networking in our case) - the "undeclared" stuff from your first post.

>  and pipedebug.c compiles without any warning or error

Are you sure it was building actually `pipedebug.c` - did its filename show up in verbose build logs?

As I said, the recipes call for building a `sockdebug` program. Then it cheats a bit by using platform-specific source files to fulfill that target, see https://github.com/networkupstools/nut/blob/master/server/Makefile.am#L34-L39