networkupstools/nut #2183: PyNUTClient needs `telnetlib` which is deprecated

jimklimov, 2023-11-14:
Raised in #2181 discussion - after Python-3.13 we will need another implementation for the layer of text-based interaction over TCP in Python.

aquette, 2023-12-09:
side notes: while playing with Home Assistant NUT addon, I'm thinking about improving a bit pyNUT... :D
scan, device dump esp., possibly the cmd cookies (not sure it supports...)
cheers @jimklimov ;)

Mausy5043, 2024-01-29:
Although I currently have no time to work on this (again), I'd like to point anyone who does to [this repo](https://github.com/Mausy5043/python-nut3). I've deprecated it a month or so ago. But the implementation of `pexpect` to replace `telnetlib` was completed in that repo. Hope this helps.

opoplawski, 2024-02-25:
We just started development on Fedora 41 which will ship with Python 3.13.  So it would be nice to see this addressed relatively soon.  The above comment seems promising.

Also FWIW:
According to https://docs.python.org/3.13/whatsnew/3.13.html:
PEP 594: Remove the telnetlib module, deprecated in Python 3.11: use the projects telnetlib3 or Exscript instead.
(Contributed by Victor Stinner in gh-104773.)

jimklimov, 2024-02-26:
Took some time to investigate telnetlib3, but it seems too great a paradigm change. With original PyNUT code, we open a client connection to the NUT server once, and in different methods post this or that request and parse replies. With telnetlib3+asyncio, it seems that each interaction must be a completely (pre-)defined I/O loop, so a separate connection, login, query, response, exit for each bit of work. Lots of avoidable extra server stress too, I guess.

jimklimov, 2024-03-04:
It seems that one recommended approach per https://discuss.python.org/t/pep-594-take-2-removing-dead-batteries-from-the-standard-library/13508/59 is to just copy `telnetlib.py` from a distro which shipped one, and use as part of the consuming project that is hard to update otherwise. At least, this can be our first step in the area (until someone comes up with a more modern port).

Although examples at one of the recommended replacements, https://github.com/knipknap/exscript/ / https://exscript.readthedocs.io/en/latest/ seem to be conceptually similar to what PyNUT worked like.

opoplawski, 2024-06-21:
Just a heads up that Python 3.13b2 has now landed in Fedora Rawhide so NUT is failing to build.

jimklimov, 2024-06-21:
@opoplawski : thanks. WDYT : would making a private copy of the telnetlib module from original recent Python distribution be a decent stop-gap solution (try/except to import locally if not available globally)? Their migration docs seem to suggest as much...

opoplawski, 2024-06-21:
Yes, it would be acceptable.  Though I fear that it might become the permanent solution...

jimklimov, 2024-06-21:
Might... maybe a vetter one would be to just go socket programming (same as done in C) - we do not need the full power of telnet/ssh here...

jimklimov, 2024-07-02:
With PR #2501 merged, we now have at least that, a stashed copy. Not perfect, so keeping this ticket open to track a desire for simpler code to work with that socket layer, but good enough to keep on walking.

@opoplawski : can you please check if this passes for builds on systems with "too new" a version of Python?

opoplawski, 2024-07-03:
So, just applying that PR to 2.8.2 I get:
```
+ ./configure --build=x86_64-redhat-linux --host=x86_64-redhat-linux --program-prefix= --disable-dependency-tracking --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --sysconfdir=/etc --datadir=/usr/share --includedir=/usr/include --libdir=/usr/lib64 --libexecdir=/usr/libexec --localstatedir=/var --runstatedir=/run --sharedstatedir=/var/lib --mandir=/usr/share/man --infodir=/usr/share/info --with-all --without-powerman --with-libltdl --without-wrap --with-cgi --with-python=/usr/bin/python3 --with-python3=/usr/bin/python3 --without-python2 --datadir=/usr/share/nut --with-user=nut --with-group=dialout --with-statepath=/run/nut --with-pidpath=/run/nut --with-altpidpath=/run/nut --sysconfdir=/etc/ups --with-cgipath=/var/www/nut-cgi-bin --with-drvpath=/usr/sbin --without-gpio --with-systemdsystemunitdir=/usr/lib/systemd/system --with-systemdshutdowndir=/lib/systemd/system-shutdown --with-pkgconfig-dir=/usr/lib64/pkgconfig --disable-static --with-udev-dir=/usr/lib/udev --libdir=/usr/lib64
checking for CONFIG_FLAGS... --build=x86_64-redhat-linux --host=x86_64-redhat-linux --program-prefix= --disable-dependency-tracking --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --sysconfdir=/etc --datadir=/usr/share --includedir=/usr/include --libdir=/usr/lib64 --libexecdir=/usr/libexec --localstatedir=/var --runstatedir=/run --sharedstatedir=/var/lib --mandir=/usr/share/man --infodir=/usr/share/info --with-all --without-powerman --with-libltdl --without-wrap --with-cgi --with-python=/usr/bin/python3 --with-python3=/usr/bin/python3 --without-python2 --datadir=/usr/share/nut --with-user=nut --with-group=dialout --with-statepath=/run/nut --with-pidpath=/run/nut --with-altpidpath=/run/nut --sysconfdir=/etc/ups --with-cgipath=/var/www/nut-cgi-bin --with-drvpath=/usr/sbin --without-gpio --with-systemdsystemunitdir=/usr/lib/systemd/system --with-systemdshutdowndir=/lib/systemd/system-shutdown --with-pkgconfig-dir=/usr/lib64/pkgconfig --disable-static --with-udev-dir=/usr/lib/udev --libdir=/usr/lib64
...
checking if we can and should install NUT-Monitor desktop application... no: Missing some or all of these Python3 modules: 're,glob,codecs,PyQt5.uic,configparser' and some or all of these Python2 modules: 're,glob,codecs,gtk,gtk.glade,gobject,ConfigParser'
checking if we can and should install PyNUT module (note for warnings from python 3.11 and beyond: we have a fallback nut_telnetlib module just in case)... Traceback (most recent call last):
  File "<string>", line 1, in <module>
    import telnetlib
ModuleNotFoundError: No module named 'telnetlib'
./configure: line 33905: cd: script/python/module: No such file or directory
configure: error: Prerequisites for PyNUT not found, can't install as required
```

Another comment would be to call out that presumably that file is under the Python license rather than the GPL.

If I try to build from git master I get an autoreconf failure:
```
configure.ac:5144: error: required file 'scripts/augeas/nutupsconf.aug.in' not found
configure.ac:5144: error: required file 'scripts/devd/nut-usb.conf.in' not found
configure.ac:5036: error: required file 'scripts/systemd/nut-common-tmpfiles.conf.in' not found
configure.ac:5144: error: required file 'scripts/udev/nut-usbups.rules.in' not found
```
This is with autoconf 2.72.

jimklimov, 2024-07-03:
Interesting point about the license, thanks.

At least the first commit was Dec 1997 https://github.com/python/cpython/commit/b9b50eb7e09ec1b28e8549c2c90d7ed5748b3173 for Python 1.5 beta: https://github.com/python/cpython/tree/b9b50eb7e09ec1b28e8549c2c90d7ed5748b3173 by by Guido van Rossum (Python creator) under CNRI at that time.

According to https://docs.python.org/3/license.html it should have been the "GPL-compatible" version; sources at the time pointed to https://github.com/python/cpython/blob/b9b50eb7e09ec1b28e8549c2c90d7ed5748b3173/Misc/COPYRIGHT

The module is not mentioned among https://docs.python.org/3/license.html#otherlicenses so it is probably part of Python core and marched on with its licensing changes (PSFv2 at the moment, per https://github.com/python/cpython/blob/3.10/LICENSE applicable for the copied file).

jimklimov, 2024-07-03:
Comments should have got addressed by #2505

opoplawski, 2024-07-12:
That looks good to me.  A release at this point would be very helpful for us.  Thanks!

jimklimov, 2024-07-12:
Thanks for the check-up! I have some fixes in progress to wrap up, but will try to not delay too much about a release then.

Scroker, 2024-09-17:
Is it helpful if I start working on a Python library migration from telnetlib to asyncio? I am developing my own custom connection library for this application https://gitlab.com/mkswap/UPSMonitor, but if possible I would prefer to work on the official one.

jimklimov, 2024-09-17:
If that ultimately works, why not.

So far the libs I looked at were quite aimed at proper (perhaps even interactive-ish) telnet/ssh communication sessions, with all the handling for terminal emulation, sending keypresses, etc.

What NUT bindings need is much smaller in scale, and can be seen in C and now C++ client lib implementations  - open a socket, send/receive bytes in one true encoding, parse the replies into language types (dict/list/str/int/... here).

So overheads of switching to alternative libraries, balanced vs. the little time I have to spare and benefit of going from one thing that works to another that also works as much, proved to be a blocker for me - path of least resistance, ain't broken don't fix, bang for buck, and all that.

One other wish is for new NUT releases to keep working on older systems - some may have no Python3 builds packaged for them. Currently UI client code is split by tech (QT vs GTK) and language syntax nuances, but the library source works for both python 2 and 3. Ideally a rewrite would keep that capability (at least at `import PyNUTClient` level, even if that pulls one or the other implementation), as well as API to minimize surprises for other consumers - if it is suitable.

Can always start another version too though, but would cause headaches to adapt any downstreams...

jimklimov, 2025-01-29:
No idea OTOH and no time to research in short-term future. Thanks for giving it a look :)

jimklimov, 2025-02-08:
Fixed by PR #2792 thanks to @cgarz

closed 2025-02-08 by jimklimov

jimklimov, 2025-02-08:
FWIW, current state of the module should be uploaded to https://test.pypi.org/project/pynutclient/2.8.2.2132/ now, so can be tested as a standalone module.