networkupstools/nut #2186: PyNUTClient is missing tests

mtelka, 2023-11-15:
Please add some tests to PyNUTClient sdist package.  Thank you.

jimklimov, 2023-11-15:
There is the self-test program, should be attached, used in NIT (NUT Integration Test suite). I have a hard time imagining tests for this client that would go without a running server anyway, but PRs are welcome. Python is not my native tongue :)

mtelka, 2023-11-16:
> There is the self-test program, should be attached, used in NIT (NUT Integration Test suite).

Where?

> I have a hard time imagining tests for this client that would go without a running server anyway, but PRs are welcome.

Special requirements to run tests are okay.

> Python is not my native tongue :)

My neither. :-)

jimklimov, 2023-11-16:
The `test_nutclient.py` is in `src` (other bug) of https://test-files.pythonhosted.org/packages/c3/6d/f058cf24da6e3f45d8043219b4a17d3e31ba2d637a2c75bb5d04b04b4ceb/PyNUTClient-2.8.1.139.1.tar.gz per https://test.pypi.org/project/PyNUTClient/2.8.1.139.1/#files

The PyPI repo does not really allow to fix old releases, so whatever is uploaded is there forever. Files may be deleted, but names can not be reused...

mtelka, 2023-11-17:
Could you please suggest some complete (dummy) configuration to see tests passed?  I started with this in `ups.conf`:
```
[dummy] 
        driver = dummy-ups
        desc = "Test device"
        port = /src/nut/data/evolution500.seq
```
but I see this failure:
```
[DEBUG] Class initialization...
[DEBUG]  -> Host  = 127.0.0.1 (port 3493)
[DEBUG]  -> Login = 'None' / 'None'
[DEBUG] Connecting to host
--------------------------------------------------------------------------------
Testing 'GetUPSList' :
[DEBUG] GetUPSList from server
{b'dummy': b'Test device'}

--------------------------------------------------------------------------------
Testing 'GetUPSVars' for 'dummy' (should be registered in upsd.conf) :
[DEBUG] GetUPSVars called...
Traceback (most recent call last):
  File "$(BUILD_DIR)/src/test_nutclient.py", line 34, in <module>
    result = nut.GetUPSVars( "dummy" )
  File "$(BUILD_DIR)/src/PyNUT.py", line 199, in GetUPSVars
    raise PyNUTError( result.replace( b"\n", b"" ).decode('ascii') )
PyNUT.PyNUTError: ERR DRIVER-NOT-CONNECTED
```
Thank you.

jimklimov, 2023-11-17:
Did you also start the NUT driver (`dummy-ups`) and `upsd` using that location via `NUT_CONFPATH`, and a writeable (by your user if not root) `NUT_STATEPATH`?

Check `tests/NIT/nit.sh` as it sets up testbeds for several clients including this one.

jimklimov, 2023-11-17:
Did it resolve BUILD_DIR by the way?

jimklimov, 2023-11-18:
Bookmarking some reading along the way :)

* https://stackoverflow.com/questions/54873637/how-to-run-tests-with-tox-ini

jimklimov, 2023-11-18:
Posted something here: https://github.com/jimklimov/nut/pull/new/python-pkg
Not sure I want to waste more time on it, delivers less files than before :\

Feel free to fiddle via `make redist` until the archives in `dist` dir are sane and post a PR :)

jimklimov, 2023-11-18:
Curiosity kills the cat, hard puzzles kill the time :)

Probably found a very sideways manner of doing it, but seems solved locally... streamlining to update the PR.

jimklimov, 2023-11-18:
@mtelka : does https://test.pypi.org/project/PyNUTClient/2.8.1.154/#files look like what would work?

closed 2023-11-18 by jimklimov

mtelka, 2023-11-19:
> Did it resolve BUILD_DIR by the way?

Sure.  The output was edited a bit.

mtelka, 2023-11-19:
> @mtelka : does https://test.pypi.org/project/PyNUTClient/2.8.1.154/#files look like what would work?

Mostly yes.  The `tox.ini` is a bit overengineered (for example both `envlist` and `commands_pre` are not needed, AFAIK), but I can live with that.

Thank you.

jimklimov, 2023-11-20:
FYI: An "official post-release" application of these recipe changes is at https://pypi.org/project/PyNUTClient/2.8.1.post1/ with the original "2.8.1" package "yanked" from the repository.

jimklimov, 2023-11-20:
Otherwise, sort of washing my hands of it (leaving it to CI to publish new increments, welcoming PRs from Python practitioners if the recipes are still wrong somehow).

PR #2191 (2023-11-18): Fix contents and recipes for Python packaging
Closes: #2186
Closes: #2187

* Deliver package payload into `PyNUTClient` not `src` dir
* Deliver test script and `tox.ini` to run it (still needs a NUT setup running to test python client against)
* Re-use the Makefile logic also for the github action implementation (DRY coding)