networkupstools/nut #2189: Support for Eaton 5P1550GR-L?

dhess, 2023-11-16:
Hi, thanks for this project!

I'm looking into rack mount UPSes, and the new-ish Eaton lithium ion models look interesting. The [5P1550GR-L](https://www.eaton.com/gb/en-gb/skuPage.5P1550GR-L.html) isn't on the compatibility list, but the 5P line in general is:

https://networkupstools.org/stable-hcl.html?manufacturer=Eaton&model=5P

Has anyone gotten this specific model to work with `usbhid-ups`? If not, is there any reason to believe that some 5P models, especially newer models and/or rack mount models, wouldn't work as the other 5P models do?

jimklimov, 2023-11-17:
I'll refer to colleagues possibly in the know for comments on the new tech. Cc @aquette @dzomaya

dhess, 2024-02-08:
Bumping &mdash; it’s time to make a purchase and I’d like to get something Li-ion and supported by NUT.

dhess, 2024-02-14:
FYI, this model does appear to be supported. `upsc` output with no load:

```
battery.capacity: 5.00
battery.charge: 100
battery.charge.low: 20
battery.charge.restart: 0
battery.energysave: no
battery.energysave.delay: 300
battery.energysave.load: 5
battery.protection: no
battery.runtime: 17880
battery.type: Li-ion
device.mfr: EATON
device.model: Eaton 5P 1550 GR Li-ion
device.serial: <redacted>
device.type: ups
driver.name: usbhid-ups
driver.parameter.bus: 001
driver.parameter.offdelay: 60
driver.parameter.ondelay: 0
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.product: Eaton 5P
driver.parameter.productid: FFFF
driver.parameter.serial: <redacted>
driver.parameter.synchronous: auto
driver.parameter.vendor: EATON
driver.parameter.vendorid: 0463
driver.version: 2.8.0
driver.version.data: MGE HID 1.46
driver.version.internal: 0.47
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.current: 0.20
input.frequency: 49.8
input.frequency.extended: no
input.frequency.nominal: 50
input.sensitivity: normal
input.transfer.boost.low: 192
input.transfer.high: 294
input.transfer.low: 167
input.transfer.trim.high: 276
input.voltage: 248.6
input.voltage.extended: no
input.voltage.nominal: 230
outlet.1.autoswitch.charge.low: 0
outlet.1.delay.shutdown: 65535
outlet.1.delay.start: 3
outlet.1.desc: PowerShare Outlet 1
outlet.1.id: 1
outlet.1.status: on
outlet.1.switchable: yes
outlet.2.autoswitch.charge.low: 0
outlet.2.delay.shutdown: 65535
outlet.2.delay.start: 6
outlet.2.desc: PowerShare Outlet 2
outlet.2.id: 2
outlet.2.status: on
outlet.2.switchable: yes
outlet.desc: Main Outlet
outlet.id: 0
outlet.switchable: no
output.current: 0.00
output.frequency: 49.8
output.frequency.nominal: 50
output.powerfactor: 0.00
output.voltage: 248.6
output.voltage.nominal: 240
ups.beeper.status: enabled
ups.delay.shutdown: 60
ups.delay.start: 0
ups.efficiency: 0
ups.firmware: 01.08.0020
ups.load: 0
ups.load.high: 105
ups.mfr: EATON
ups.model: Eaton 5P 1550 GR Li-ion
ups.power: 0
ups.power.nominal: 1550
ups.productid: ffff
ups.realpower: 0
ups.realpower.nominal: 1100
ups.serial: <redacted>
ups.shutdown: enabled
ups.start.auto: yes
ups.start.battery: yes
ups.start.reboot: yes
ups.status: OL CHRG
ups.test.interval: 0
ups.test.result: Done and passed
ups.timer.shutdown: -1
ups.timer.start: -1
ups.type: offline / line interactive
ups.vendorid: 0463
```

The product ID (`FFFF`) does look suspicious.

aquette, 2024-02-14:
Indeed supported, but no specifics about the battery.
Also product ids are either 0001 or ffff, so normal

dhess, 2024-02-14:
> Indeed supported, but no specifics about the battery. If you share a "dump tree" (debug level 1, then Ctrl+c), we can try to see if any news Usages for li-ion. Also product ids are either 0001 or ffff, so normal

Like this?

```
# upsdrvctl -D -d start ups-b
Network UPS Tools - UPS driver controller 2.8.0
   0.000000	[D1] Starting UPS: ups-b
Network UPS Tools - Generic HID driver 0.47 (2.8.0)
USB communication driver (libusb 1.0) 0.43
   0.000000	[D1] debug level is '1'
   0.001572	[D1] upsdrv_initups (non-SHUT)...
   0.005396	[D1] Failed to open device (1D6B/0003), skipping: Access denied (insufficient permissions)
   0.006387	interrupt pipe disabled (add 'pollonly' flag to 'ups.conf' to get rid of this message)
   0.305918	[D1] Eaton device v2.02. Using full report descriptor
   4.152354	Using subdriver: MGE HID 1.46
   4.152391	[D1] 218 HID objects found
   4.181300	[D1] Path: UPS.BatterySystem.Battery.AudibleAlarmControl, Type: Feature, ReportID: 0x28, Offset: 0, Size: 8, Value: 2
   4.221698	[D1] Path: UPS.BatterySystem.Battery.BatteryID, Type: Feature, ReportID: 0x20, Offset: 0, Size: 8, Value: 1
   4.253186	[D1] Path: UPS.BatterySystem.Battery.DeepDischargeProtection, Type: Feature, ReportID: 0x22, Offset: 0, Size: 8, Value: 0
   4.292467	[D1] Path: UPS.BatterySystem.Battery.DesignCapacity, Type: Feature, ReportID: 0x23, Offset: 0, Size: 32, Value: 18000
   4.330590	[D1] Path: UPS.BatterySystem.Battery.PresentStatus.Present, Type: Feature, ReportID: 0x27, Offset: 0, Size: 8, Value: 1
   4.330622	[D1] Path: UPS.BatterySystem.Battery.PresentStatus.Present, Type: Input, ReportID: 0x27, Offset: 0, Size: 8, Value: 1
   4.360836	[D1] Path: UPS.BatterySystem.Battery.Test, Type: Feature, ReportID: 0x24, Offset: 0, Size: 8, Value: 1
   4.400562	[D1] Path: UPS.BatterySystem.Battery.TestPeriod, Type: Feature, ReportID: 0x25, Offset: 0, Size: 32, Value: 0
   4.400593	[D1] Path: UPS.BatterySystem.BatterySystemID, Type: Feature, ReportID: 0x20, Offset: 8, Size: 8, Value: 1
   4.432866	[D1] Path: UPS.BatterySystem.Charger.ABMEnable, Type: Feature, ReportID: 0x29, Offset: 0, Size: 8, Value: 0
   4.432895	[D1] Path: UPS.BatterySystem.Charger.ChargerID, Type: Feature, ReportID: 0x20, Offset: 16, Size: 8, Value: 1
   4.471346	[D1] Path: UPS.BatterySystem.Charger.ChargerType, Type: Feature, ReportID: 0x26, Offset: 0, Size: 8, Value: 5
   4.471377	[D1] Path: UPS.BatterySystem.Charger.Mode, Type: Feature, ReportID: 0x26, Offset: 8, Size: 8, Value: 5
   4.471388	[D1] Path: UPS.BatterySystem.Charger.PresentStatus.Floating, Type: Feature, ReportID: 0x27, Offset: 8, Size: 8, Value: 0
   4.471395	[D1] Path: UPS.BatterySystem.Charger.PresentStatus.Floating, Type: Input, ReportID: 0x27, Offset: 8, Size: 8, Value: 0
   4.471406	[D1] Path: UPS.BatterySystem.Charger.PresentStatus.InternalFailure, Type: Feature, ReportID: 0x27, Offset: 16, Size: 8, Value: 0
   4.471423	[D1] Path: UPS.BatterySystem.Charger.PresentStatus.InternalFailure, Type: Input, ReportID: 0x27, Offset: 16, Size: 8, Value: 0
   4.471433	[D1] Path: UPS.BatterySystem.Charger.PresentStatus.Used, Type: Feature, ReportID: 0x27, Offset: 24, Size: 8, Value: 1
   4.471441	[D1] Path: UPS.BatterySystem.Charger.PresentStatus.Used, Type: Input, ReportID: 0x27, Offset: 24, Size: 8, Value: 1
   4.471453	[D1] Path: UPS.BatterySystem.Charger.PresentStatus.VoltageTooHigh, Type: Feature, ReportID: 0x27, Offset: 32, Size: 8, Value: 0
   4.471464	[D1] Path: UPS.BatterySystem.Charger.PresentStatus.VoltageTooHigh, Type: Input, ReportID: 0x27, Offset: 32, Size: 8, Value: 0
   4.471476	[D1] Path: UPS.BatterySystem.Charger.PresentStatus.VoltageTooLow, Type: Feature, ReportID: 0x27, Offset: 40, Size: 8, Value: 0
   4.471484	[D1] Path: UPS.BatterySystem.Charger.PresentStatus.VoltageTooLow, Type: Input, ReportID: 0x27, Offset: 40, Size: 8, Value: 0
   4.471496	[D1] Path: UPS.BatterySystem.Charger.ffff00e9, Type: Feature, ReportID: 0x26, Offset: 16, Size: 8, Value: 1
   4.504944	[D1] Path: UPS.Flow.[1].ConfigFrequency, Type: Feature, ReportID: 0x14, Offset: 0, Size: 8, Value: 50
   4.542156	[D1] Path: UPS.Flow.[1].ConfigVoltage, Type: Feature, ReportID: 0x15, Offset: 0, Size: 16, Value: 230
   4.581468	[D1] Path: UPS.Flow.[1].FlowID, Type: Feature, ReportID: 0x0b, Offset: 0, Size: 8, Value: 1
   4.614146	[D1] Path: UPS.Flow.[4].ConfigActivePower, Type: Feature, ReportID: 0x74, Offset: 0, Size: 16, Value: 1100
   4.650141	[D1] Path: UPS.Flow.[4].ConfigApparentPower, Type: Feature, ReportID: 0x75, Offset: 0, Size: 16, Value: 1550
   4.687145	[D1] Path: UPS.Flow.[4].ConfigFrequency, Type: Feature, ReportID: 0x0d, Offset: 0, Size: 8, Value: 50
   4.720940	[D1] Path: UPS.Flow.[4].ConfigPercentLoad, Type: Feature, ReportID: 0x16, Offset: 0, Size: 8, Value: 105
   4.758106	[D1] Path: UPS.Flow.[4].ConfigVoltage, Type: Feature, ReportID: 0x72, Offset: 0, Size: 16, Value: 240
   4.758157	[D1] Path: UPS.Flow.[4].FlowID, Type: Feature, ReportID: 0x0b, Offset: 8, Size: 8, Value: 4
   4.839724	[D1] Path: UPS.HistorySystem.Event.[1].Code, Type: Feature, ReportID: 0x71, Offset: 0, Size: 16, Value: 0
   4.839758	[D1] Path: UPS.HistorySystem.Event.[2].Code, Type: Feature, ReportID: 0x71, Offset: 16, Size: 16, Value: 0
   4.839767	[D1] Path: UPS.HistorySystem.Event.[3].Code, Type: Feature, ReportID: 0x71, Offset: 32, Size: 16, Value: 0
   4.839774	[D1] Path: UPS.HistorySystem.Event.[4].Code, Type: Feature, ReportID: 0x71, Offset: 48, Size: 16, Value: 0
[... cut ...]
   9.889825	[D1] upsdrv_initinfo...
   9.889860	[D1] upsdrv_updateinfo...
   9.889867	[D1] Not using interrupt pipe...
   9.889873	[D1] Quick update...
   9.890010	[D1] upsdrv_updateinfo...
   9.890023	[D1] Not using interrupt pipe...
   9.890029	[D1] Quick update...
  11.891317	[D1] upsdrv_updateinfo...
  11.891356	[D1] Not using interrupt pipe...
  11.891365	[D1] Quick update...
  13.892255	[D1] upsdrv_updateinfo...
  13.892291	[D1] Not using interrupt pipe...
  13.892304	[D1] Quick update...
^C

```

arnaudquette-eaton, 2024-02-15:
perfect, thanks @dhess ;)
I'll try to put some more cycles into this, and already contacted my 23 years fellow that implemented USB/HID for MGE UPSs(which is now Eaton for this part), to clarify the missing supported usage IDs (all these "ffff<something>) and possible missing HID paths, including the specifics for Li-Ion