networkupstools/nut #2190: Battery low due to battery.run.time=0 when using wall power

YumingChang02, 2023-11-18:
I got an ATEN OL2000LV, which seems to indicate - runtime when connected to wall power, is there a way to only check battery low when running on battery?

```
arch@ram-runner ~> upsc nutdev1
battery.charge: 100
battery.charge.low: 30
battery.charge.warning: 50
battery.energysave: no
battery.packs: 1
battery.protection: yes
battery.runtime: 0
battery.runtime.low: 0
battery.voltage: 54.70
battery.voltage.nominal: 48.0
device.model: OLLV2K0
device.type: ups
driver.flag.ignorelb: enabled
driver.name: nutdrv_qx
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: 5161
driver.parameter.synchronous: auto
driver.parameter.vendorid: 0665
driver.version: 2.8.0
driver.version.data: Voltronic 0.06
driver.version.internal: 0.32
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.current.nominal: 18.0
input.frequency: 60.0
input.frequency.high: 63.0
input.frequency.low: 57.0
input.frequency.nominal: 60.0
input.phases: 1
input.transfer.high: 115
input.transfer.high.max: 122
input.transfer.high.min: 113
input.transfer.low: 105
input.transfer.low.max: 107
input.transfer.low.min: 98
input.voltage: 106.6
input.voltage.nominal: 110.0
outlet.0.switchable: yes
outlet.1.delay.shutdown: 600
outlet.1.status: on
outlet.1.switchable: yes
output.current: 5.6
output.current.nominal: 18
output.frequency: 60.0
output.frequency.nominal: 60.0
output.phases: 1
output.power.maximum.percent: 28
output.power.minimum.percent: 24
output.powerfactor: 1.0
output.voltage: 109.4
output.voltage.nominal: 110.0
ups.beeper.status: enabled
ups.delay.shutdown: 80
ups.delay.start: 180
ups.firmware: 01574.04
ups.firmware.aux: P01
ups.load: 29
ups.power.nominal: 2000
ups.productid: 5161
ups.start.auto: yes
ups.start.battery: yes
ups.status: OL
ups.temperature: 22.1
ups.type: online
ups.vendorid: 0665
```

jimklimov, 2023-11-18:
Interesting case, thanks for reporting :)

At least in the `ups.status` it reports only the `OL` (on line) flag, so immediately in the driver it does not assume that it has a low battery, it seems.  Where do you have the practical problem with this situation? Does `upsmon` raise an alarm and trigger a host shutdown?

jimklimov, 2023-11-18:
Also, just in case, ask the vendor support if their protocol implementation actually mis-behaves this way (does report run-time and reports as 0) which then would be a bug for them to fix (e.g. you as a paying customer can't predict how well protected your server is for a safe shutdown, without causing an outage for the test)? Or if they do report the runtime, but in some manner that earlier contributors to this NUT driver did not encounter and code for (sort of our bug, PRs welcome)?

Also, can it be just a case of old/dead battery?

YumingChang02, 2023-11-20:
@jimklimov If i set battery.runtime.low to anything but 0, it shows OL LB flags. Not really a huge issue, just that it will span over log files. Machines complains about low power but still runs as is. Maybe i'll try to check what the problem is, whether is a vendor or code.

YumingChang02, 2023-11-20:
Fast look over log looks like a driver issue?

```
   3.815847     [D3] send: QBV
   4.159666     [D5] read [  0]: (8 bytes) => 28 30 35 34 2e 37 20 30
   4.239674     [D5] read [  8]: (8 bytes) => 34 20 30 31 20 31 30 30
   4.271677     [D5] read [ 16]: (8 bytes) => 20 30 30 30 31 39 0d 00
   4.271698     [D3] read: (054.7 04 01 100 00019
   4.271726     battery_number: 4
   4.271735     battery_number, settable range: 1..9
   4.271754     [D5] send_to_all: SETINFO battery.packs "1"
   4.271765     [D5] send_to_all: SETFLAGS battery.packs RW
   4.271778     [D5] send_to_all: ADDRANGE battery.packs 1 99
   4.271792     [D5] send_to_all: SETFLAGS battery.packs RW NUMBER
   4.271807     [D5] send_to_all: SETINFO battery.charge "100"
   4.271823     [D5] send_to_all: SETINFO battery.runtime "0"
   4.272062     [D3] send: QLDL
```

according to [ref](https://networkupstools.org/protocols/voltronic.html)
should be (MM.M NN PP QQQ RRR, where RRR is Remaining backup time (minutes)

jimklimov, 2023-11-20:
Thanks for the trace! So it does seem like a NUT bug, with 19 minutes not interpreted well.

Would it be possible for you to custom-build a NUT master (or use a recent 2.8.1 release if packaged for your system) to check if it works better? There were some changes to QX parser, not sure OTOH if they were limited to the `armac` subdriver or benefitted all of them.

Although, looking at source, I think it just picks 3 chars (positions `17..19`) and so only sees the zeroes:
* https://github.com/networkupstools/nut/blob/v2.8.1/drivers/nutdrv_qx_voltronic.c#L1126-L1137
* https://github.com/networkupstools/nut/blob/v2.8.1/drivers/nutdrv_qx.h#L61-L108 - structure elements definition

I am not sure if it is a "protocol violation" to have not exactly 21 chars in that response, or if it is just the expected minimal length.

Per structure comments, probably using `0` for the last element (remainder of line) would be OK; less sure about use of a specific numeric size (e.g. `5` here) with regard to possible buffer overflows (what if a reply does have 3 digits and we walk the array for 5?) - gotta read/trace into the parser to be sure about a safe choice here. Or experiment with larger numbers than the response you have in practice (say, `10`).

I would welcome a PR following a custom build and test on your side, if possible - please check https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests for suggestions about it. Don't forget to bump `VOLTRONIC_VERSION` in the driver and perhaps describe the improvement in `NEWS.adoc` :)

jimklimov, 2024-01-20:
@YumingChang02 : cheers, just in case - did you have any progress with this issue to share?

derritter88, 2024-03-26:
Hello there, same here for me. I am using a Bluewalker/Powerwalker with driver `nutdrv_qx` and it also shows `battery.runtime: 0` but `ups.status: OL`

I just run a deep test to let the ups determine the runtime (battery went down to 10 %) but still no runtime output.