networkupstools/nut #2193: PyNUTClient 2.8.1.post1 wheel and tox.ini

mtelka, 2023-11-20:
The wheel for PyNUTClient 2.8.1.post1 contains `tox.ini`.  It should not.  Or if it is there, then definitely at different location.

```
$ find .local/
.local/
$ pip install -U pynutclient
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: pynutclient in /usr/lib/python3.9/vendor-packages (2.8.1)
Collecting pynutclient
  Using cached PyNUTClient-2.8.1.post1-py3-none-any.whl.metadata (12 kB)
Using cached PyNUTClient-2.8.1.post1-py3-none-any.whl (25 kB)
Installing collected packages: pynutclient
Successfully installed pynutclient-2.8.1.post1
$ find .local/
.local/
.local/lib
.local/lib/python3.9
.local/lib/python3.9/site-packages
.local/lib/python3.9/site-packages/PyNUTClient
.local/lib/python3.9/site-packages/PyNUTClient/__init__.py
.local/lib/python3.9/site-packages/PyNUTClient/__pycache__
.local/lib/python3.9/site-packages/PyNUTClient/__pycache__/PyNUT.cpython-39.pyc
.local/lib/python3.9/site-packages/PyNUTClient/__pycache__/__init__.cpython-39.pyc
.local/lib/python3.9/site-packages/PyNUTClient/__pycache__/test_nutclient.cpython-39.pyc
.local/lib/python3.9/site-packages/PyNUTClient/PyNUT.py
.local/lib/python3.9/site-packages/PyNUTClient/test_nutclient.py
.local/lib/python3.9/site-packages/PyNUTClient-2.8.1.post1.dist-info
.local/lib/python3.9/site-packages/PyNUTClient-2.8.1.post1.dist-info/LICENSE-GPL3
.local/lib/python3.9/site-packages/PyNUTClient-2.8.1.post1.dist-info/WHEEL
.local/lib/python3.9/site-packages/PyNUTClient-2.8.1.post1.dist-info/RECORD
.local/lib/python3.9/site-packages/PyNUTClient-2.8.1.post1.dist-info/REQUESTED
.local/lib/python3.9/site-packages/PyNUTClient-2.8.1.post1.dist-info/top_level.txt
.local/lib/python3.9/site-packages/PyNUTClient-2.8.1.post1.dist-info/INSTALLER
.local/lib/python3.9/site-packages/PyNUTClient-2.8.1.post1.dist-info/METADATA
.local/tox.ini
$
```
Strictly speaking, `test_nutclient.py` is also not needed in wheel, but it is harmless.

mtelka, 2023-11-20:
The proposed fix:
```
--- PyNUTClient-2.8.1.post1/setup.py.orig
+++ PyNUTClient-2.8.1.post1/setup.py
@@ -27,7 +27,6 @@
     packages   =       find_packages(),
     #py_modules =      ['PyNUT'],
     package_dir =      {'PyNUT': 'PyNUTClient'},
-    data_files =       [('', ['tox.ini'])],
     #scripts    =      ['PyNUTClient/test_nutclient.py', 'PyNUTClient/__init__.py'],
     python_requires =  '>=2.6',
     # install_requires =       ['telnetlib'],  # NOTE: telnetlib.py is part of Python core for tested 2.x and 3.x versions, not something 'pip' can download
```
If `tox.ini` is not added automatically to sdist with the above patch, then it should be [added to MANIFEST.in](https://setuptools.pypa.io/en/latest/userguide/miscellaneous.html).

jimklimov, 2023-11-20:
Same as with testpypi upload that you said was ok :)

jimklimov, 2023-11-20:
Strange that source archives won't have test setup by default, I jumped through hoops to get them there as you seem to have asked if I got that right :D

mtelka, 2023-11-20:
> Same as with testpypi upload that you said was ok :)

Sorry, I checked sdist there only, not wheel.

jimklimov, 2023-11-20:
That's okay, seems we both are learning here :) At least, I found that PyPI does not treat versions "X.Y.Z.N" as different from plain "X.Y.Z" when "N" is a number, but has some special patterns to support for the "N" including "rcN" and "postN". So at least there is a chance to follow up on mistakes without superficial new releases for typos.

jimklimov, 2023-11-21:
Took a quick look at it and came back none the wiser: it seems that `tox` utility looks for a `setup.py` in the `toxinidir` (by default at least), so these two files gotta sit together in sources.

Indeed, it probably should not be in the "wheel" archive, but without that "data files" line the `tox.ini` does not get into the sdist. Is there some option for "developer-only data files"?

mtelka, 2023-11-21:
> Indeed, it probably should not be in the "wheel" archive, but without that "data files" line the `tox.ini` does not get into the sdist. Is there some option for "developer-only data files"?

https://github.com/networkupstools/nut/issues/2193#issuecomment-1819013410

mtelka, 2023-11-21:
> it seems that `tox` utility looks for a `setup.py` in the `toxinidir` (by default at least), so these two files gotta sit together in sources.

`tox` does not need `setup.py`.  It probably does not care about `setup.py` at all.  `tox` optionally reads [setup.cfg as a possible config file](https://tox.wiki/en/latest/config.html).

jimklimov, 2023-11-21:
Not convinced this is true (it seems to create and use some sdist-like tarballs), but as you'd say:
````
:; tox
...
.pkg: _optional_hooks> python /usr/local/lib/python3.10/dist-packages/pyproject_api/_backend.py True setuptools.build_meta __legacy__
.pkg: get_requires_for_build_sdist> python /usr/local/lib/python3.10/dist-packages/pyproject_api/_backend.py True setuptools.build_meta __legacy__
.pkg: get_requires_for_build_wheel> python /usr/local/lib/python3.10/dist-packages/pyproject_api/_backend.py True setuptools.build_meta __legacy__
.pkg: prepare_metadata_for_build_wheel> python /usr/local/lib/python3.10/dist-packages/pyproject_api/_backend.py True setuptools.build_meta __legacy__
.pkg: build_sdist> python /usr/local/lib/python3.10/dist-packages/pyproject_api/_backend.py True setuptools.build_meta __legacy__

py310: install_package> python -I -m pip install --force-reinstall --no-deps /home/jim/nut/scripts/python/module/.tox/.tmp/package/6/PyNUTClient-2.8.1.183.tar.gz
py310: commands_pre[0]> python -m pip install -U pip
Requirement already satisfied: pip in ./.tox/py310/lib/python3.10/site-packages (23.3.1)
py310: commands[0]> python PyNUTClient/test_nutclient.py
PyNUTClient test...
[DEBUG] Class initialization...
[DEBUG]  -> Host  = 127.0.0.1 (port 3493)
[DEBUG]  -> Login = 'None' / 'None'
[DEBUG] Connecting to host
...
````

jimklimov, 2023-11-21:
> > Indeed, it probably should not be in the "wheel" archive, but without that "data files" line the `tox.ini` does not get into the sdist. Is there some option for "developer-only data files"?
> 
> [#2193 (comment)](https://github.com/networkupstools/nut/issues/2193#issuecomment-1819013410)

This just removes it from both archives, including sdist.

mtelka, 2023-11-21:
> This just removes it from both archives, including sdist.

Are you sure?  What exactly you tried to add into `MANIFEST.in`?

jimklimov, 2023-11-21:
There isn't any here :\

mtelka, 2023-11-21:
> Not convinced this is true (it seems to create and use some sdist-like tarballs), but as you'd say:

`tox` just tries to install something (usually, but not always).  That does not mean `tox` itself needs `setup.py`.  Some other tools (or hooks) invoked by `tox` (for example `pip`) might use `setup.py` for something.  But again, `setup.py` is not mandatory for a Python project.  It is legacy way how to support packaging of Python projects.  Modern Python projects very often already removed `setup.py` and switched completely to clean PEP 517 based packaging.

So in short, `setup.py` is not needed for a Python project, and definitely it is not needed for `tox` itself.

PR #2196 (2023-11-22): scripts/python/module/MANIFEST.in: distribute "tox.ini" in a different…
…way so it is only in source dist

Closes: #2193
closed 2023-11-22 by jimklimov