networkupstools/nut #2200: Allow to customize PYTHON2 and PYTHON3 macros

bigon, 2023-11-25:
Hello,

In NUT-Monitor-py* scripts, the configure will replace the PYTHON2 and PYTHON3 macros with something like `/usr/bin/python3.X`

In debian, it needs to be `/usr/bin/python3` without the minor version.

Would be nice if it was possible to customize that at configuration time (instead of patching the file)

jimklimov, 2023-11-25:
Check the `configure ... --with-python{2,3}` options - do they achieve the goal? Should be there just for that :)

bigon, 2023-11-27:
Ah yes indeed that works

🤦‍♂️

closed 2023-11-27 by bigon

jimklimov, 2023-11-27:
FWIW, note the possibility of `--without-python2` as well ;)

jimklimov, 2023-11-27:
Regarding the specific vs. generic python versions - it may depend on how you deliver the `PyNUT(Client)` module; there are some configure options for that (install into site/vendor dir, install near app for private consumption), as well as a gestating PyPI module (`pip`-installable results not yet tested vs. `NUT-Monitor` UI app though).

If one way or another you deliver it as a module into the general python installation, and `NUT-Monitor*.py` just happens to be one of the consumers, that may depend on particular python version to be involved (more so if several python3.x variants may coexist) just to look into the correct library paths to find the module.

If it is installed as an internal detail, it should not care about the particular version.