networkupstools/nut #2203: Is there a good guide for installing NUT on a Windows machine?

cableghost, 2023-11-25:
I receive the no driver error when attempting to install the latest install ver. I have found only 1 thread discussing it, but no resolution or guide. I attempted to follow the process of installing libusb via WinUSB, but don't understand it.

jimklimov, 2023-11-25:
Just to be clear: as the "latest ver" do you mean NUT v2.8.1 (from tarballs with binaries prepared by Appveyor CI), or a 2.6.5-based MSI from almost 10 years ago? There was a long hiatus and we're trying to revive the NUT for Windows effort as part of the core project, help is welcome (experiments, packaging, instructions as one big gray area) - see https://github.com/networkupstools/nut/wiki/NUT-for-Windows for more details.

cableghost, 2023-11-26:
Thanks @jimklimov, I had not read this article. I was referring to the 2.6.5 ver... I could/can not locate the 2.81 build.

Alternative would be to follow a couple of the guides cited in that article; 1) is for making 2.6.5 work on my Win machine, the other was porting a Win ver.

jimklimov, 2023-11-26:
A build of latest source should be appearing at https://ci.appveyor.com/project/nut-travis/nut/build/artifacts as noted on that Wiki page as an archive with built and installed binaries for Windows.

These can be unpacked and run just as well as on POSIX systems, but ways need to be found to auto-start things at boot and be allowed to drive a shutdown at power loss. I believe nobody tried and wrote about the experience. Per codebase, there should be a wrapping program to start others (driver(s) and data server, and/or the monitoring/shutdown client) which can register itself as a service. There are at least third-party programs to set up low-level access to USB devices, if your UPS is connected as such.

But there currently is no modern package for that platform, to make it work out of the box, so one may have to roll up sleeves sort of like a programmer or sysadmin to proceed with that right now :\

jimklimov, 2023-11-27:
According to some posted bugs, latest msi packages missed some files or something. I remember some workarounds were posted - check PRs and issues tagged by Windows label (maybe closed, but not sure).

I used NUT upsmon client on Windows in early 2000's (ver 2.2.x I think) which just worked, similarly to Unix builds.

cableghost, 2023-11-27:
Thank you @jimklimov

masterwishx, 2024-06-19:
@jimklimov @cableghost if interested in NUT Client im using https://github.com/nutdotnet/WinNUT-Client

cableghost, 2024-06-20:
> @jimklimov @cableghost if interested in NUT Client im using https://github.com/nutdotnet/WinNUT-Client

I think the issue for me is I'm trying to monitor a UPS that is connected to another Windows PC, on same local network.