networkupstools/nut #2207: design bug: upmon spams the logs when cannot talk to driver or when driver cannot talk to ups

mailinglists35, 2023-11-29:
When `nut-monitor` cannot talk to `/lib/nut/$whatever-driver`, or when the communication between the driver and the ups is unavailable, it repeats this message in logs:

```
nov 29 16:06:01 router upsmon[486203]: Poll UPS [nutdev1@localhost] failed - Driver not connected
````

I wish it allowed a configuration directive or a command line argument to only print this message once. the polite behaviour is already exhibited by `upsd`, that only prints an equivalent message once:

````
nov 29 16:10:44 router upsd[476370]: Can't connect to UPS [nutdev1] (nutdrv_qx-serioux): No such file or directory
````

the repeating of message every 5 seconds in logs provides no use to me, and only annoyance instead.

this is not the equivalent of ups emiting audible beeps - that makes sense in physical world as having the role of catching a human's attention that might not be near it all the time, so it makes perfect sense to repeat the beep.

however the logs are monitored, and a single line is sufficient to trigger an alert, thus it make no logical sense to repeat the line.

the same when comm is lost: upsd politely tells me **once**

````
Broadcast message from nut@router (somewhere) (Wed Nov 29 16:19:58 2023):

Communications with UPS nutdev1@localhost lost

nov 29 16:19:58 router upsmon[494593]: Poll UPS [nutdev1@localhost] failed - Data stale
nov 29 16:19:58 router upsmon[494593]: Communications with UPS nutdev1@localhost lost
````

while upsmon begins spamming the logs every 5 seconds:
````
nov 29 16:20:13 router upsmon[494593]: Poll UPS [nutdev1@localhost] failed - Data stale
````

jimklimov, 2023-11-29:
I agree, and a PR is welcome :)

For context, not always not everywhere logs are monitored (so as to make alerts) from them in real time. As a side note, NUT probably predates most of the fancier log-monitoring systems you see today, so the unseen/unborn techs were not a factor in the original design :)

In fact, upsmon has "notifications" to WALL/SYSLOG/EXEC for a number of events and state changes, and those are throttled (IIRC e.g. `wall`ing once in 5 mins by default for Data Stale).

> no logical sense to repeat...

I believe it just runs the polling loop and reports encountered errors. To stderr typically, which is then sysadmin's or packager's "fault" if it ends up logged (though maybe upsdebugx()/upslogx() does ditto or redirect to syslog at some point like forking and/or based on internal config).  It is extra work to remember that the same error was reported on last cycle (such work is the welcome PR) and e.g. report it only for non-zero debug verbosity for some time. Probably the throttling approavh for notifications can be used here.

mailinglists35, 2023-11-29:
>PR

I am not a programmer, but I am wondering if the code logic that emits the similar log in `upsd` can not be applied to the routine logic blocks in `upsmon`? would it be feasible?

jimklimov, 2023-12-01:
A bit more sleuthing, and PR #1784 comes up, so it seems I've already fixed that earlier in the busy year and it got fully flushed from memory, oops `-\_(o,o)_/=`

The feature should be part of NUT codebase since January, so would be in v2.8.1 release (issued a month ago). Should not have the off-by-one behavior in v2.8.2 and later, thanks ;)

jimklimov, 2023-12-01:
Thanks for the report!

While it was ultimately about an issue already solved and published in a NUT release (seems you've never mentioned which one you are running BTW), it also helped discover some issues in the earlier implementation so the next release would be even better :)

PR #2209 (2023-12-01): upsmon: `POLLFAIL_LOG_THROTTLE_MAX` had an off-by-one error
Closes: #2207
Follows up from #1784

Now `POLLFAIL_LOG_THROTTLE_MAX` is actually the "max" and not "skip N loops and report during the next one after N", so the default behavior of reporting the errors on every loop (also as the negative `POLLFAIL_LOG_THROTTLE_MAX` value) is same as setting `1` here. A `0` still goes for reporting only the start and end of a condition; still one tracked `upserror` condition per UPS.

With this PR in place, it also logs the expected report frequency more clearly (in seconds, not just loop counts):
```` [... cut]
closed 2023-12-01 by jimklimov