networkupstools/nut #2208: on Fry's Electronics MEC0003 device, nut fails to read values unless upsmart is launched first

mailinglists35, 2023-11-29:
scenario 1:

plug usb
start nut
(no matter how many times start stop nut)

result: values are all zero

____

scenario 2:

plug usb
start upsmart once
exit upsmart
start nut
stop/start nut forever

result: values are like in upsmart

remove cable, replug cable, goto scenario 1

____

attached nut.pcap on scenario 1
upsmart.pcap and nut pcap in scenario 2
device commercial name: UPS Serioux Line Interactive 800LI, SKU codename SRXU-LD800LI  or LD800LI 
hope this helps you find what upsmart is doing to "initialize" the ups and improve the hunnox subdriver
linux version of upsmart: https://zircon.co.th/download_software_for_ups scroll down click on download upsmart will take you to a google drive [link](https://drive.google.com/drive/folders/1nq9kgPj4REW1c-U7GZmd4wx8pU-KBid8?usp=sharing) 
PS: 

````
$ dpkg -l nut nut-client nut-server | grep ^ii
ii  nut            2.8.0-1ppa10~ubuntu22.04 all          network UPS tools - metapackage
ii  nut-client     2.8.0-1ppa10~ubuntu22.04 amd64        network UPS tools - clients
ii  nut-server     2.8.0-1ppa10~ubuntu22.04 amd64        network UPS tools - core system

$ sudo grep -Ev '^(#|\s*$)' /etc/nut/ups.conf
maxretry = 3
[serioux]
        driver = "nutdrv_qx"
        port = "auto"
        subdriver = "hunnox"
        vendorid = "0001"
        productid = "0000"
        novendor
        protocol = "megatec"
        desc = "serioux"

$ sudo lsusb -d 0001:0000 -v

Bus 001 Device 008: ID 0001:0000 Fry's Electronics MEC0003
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0         8
  idVendor           0x0001 Fry's Electronics
  idProduct          0x0000
  bcdDevice            1.00
  iManufacturer           0
  iProduct                1 MEC0003
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
[... 671 lines cut ...]
   0.242177     [D3] read: #   .   .  12.00   .
   0.242371     No values for battery high/low voltages
   0.242406     Using 'guesstimation' (low: 10.400000, high: 13.000000)!
   0.242440     Battery runtime will not be calculated (runtimecal not set)
   0.242475     [D1] upsdrv_updateinfo...
   0.242492     [D1] Quick update...
   0.242512     [D3] send: Q1
   0.242534     [D4] command index: 0x03
   0.242572     [D3] asking for: 0C
   0.247754     [D3] read: (235.0 000.0 236.0 000 50.1 13.7 29.0 00001000
   0.248022     [D2] dstate_init: sock /run/nut/nutdrv_qx-serioux open on fd 5
   0.248102     [D1] upsdrv_updateinfo...
   0.248122     [D1] Quick update...
   0.248147     [D3] send: Q1
   0.248174     [D4] command index: 0x03
   0.248193     [D3] asking for: 0D
   0.252589     [D3] read: (235.0 000.0 236.0 000 50.1 13.7 29.0 00001000
   2.250333     [D1] upsdrv_updateinfo...
   2.250393     [D1] Quick update...
   2.250424     [D3] send: Q1
   2.250460     [D4] command index: 0x03
   2.253433     [D3] read: (235.0 000.0 236.0 000 50.1 13.7 29.0 00001000
^C

````

mailinglists35, 2023-11-29:
[mec0003-pcaps.zip](https://github.com/networkupstools/nut/files/13504985/mec0003-pcaps.zip)

mailinglists35, 2023-11-29:
[beeper-pcaps.zip](https://github.com/networkupstools/nut/files/13505010/beeper-pcaps.zip)

mailinglists35, 2023-11-29:
upsmart also produces a debug log:

````
$ cat DebugLog20231129_USB.txt 
Date/Time,State,Command
2023.11.29 22:03:55,Send,80 06 03 03 09 04 00 66 
2023.11.29 22:03:55,Receive,(232.0 000.0 232.0 000 49.9 13.6 29.0 00001000
2023.11.29 22:03:54,Send,80 06 0d 03 09 04 00 66 
2023.11.29 22:03:54,Receive,#   .   .  12.00   . 
2023.11.29 22:03:53,Send,80 06 03 03 09 04 00 66 
2023.11.29 22:03:53,Receive,(232.0 000.0 232.0 000 49.7 13.6 29.0 00001000
2023.11.29 22:03:52,Send,80 06 03 03 09 04 00 66 
2023.11.29 22:03:52,Receive,(232.0 000.0 232.0 000 49.9 13.6 29.0 00001000
2023.11.29 22:03:51,Send,80 06 03 03 09 04 00 66 
2023.11.29 22:03:51,Receive,(232.0 000.0 234.0 000 49.8 13.6 29.0 00001000
2023.11.29 22:03:50,Send,80 06 0c 03 09 04 00 66 
2023.11.29 22:03:50,Receive,#                           V3.8      
2023.11.29 22:03:49,Send,80 06 03 03 09 04 00 66 
2023.11.29 22:03:49,Receive,(232.0 000.0 232.0 000 50.0 13.6 29.0 00001000
2023.11.29 22:03:48,Send,80 06 0d 03 09 04 00 66 
2023.11.29 22:03:48,Receive,#   .   .  12.00   . 
2023.11.29 22:03:47,Send,80 06 07 03 09 04 00 66 
2023.11.29 22:03:47,Receive,UPS No Ack
2023.11.29 22:03:46,Send,80 06 03 03 09 04 00 66 
2023.11.29 22:03:46,Receive,(231.0 000.0 232.0 000 49.7 13.6 29.0 00001000
2023.11.29 22:03:45,Send,80 06 03 03 09 04 00 66 
2023.11.29 22:03:45,Receive,(232.0 000.0 234.0 000 49.9 13.6 29.0 00001000
2023.11.29 22:03:44,Send,80 06 03 03 09 04 00 66 
2023.11.29 22:03:44,Receive,(231.0 000.0 233.0 000 49.8 13.6 29.0 00001000
2023.11.29 22:03:43,Send,80 06 0c 03 09 04 00 66 
2023.11.29 22:03:43,Receive,#                           V3.8      
2023.11.29 22:03:42,Send,80 06 03 03 09 04 00 66 
2023.11.29 22:03:42,Receive,(231.0 000.0 232.0 000 49.6 13.6 29.0 00001000
2023.11.29 22:03:41,Send,80 06 0d 03 09 04 00 66 
2023.11.29 22:03:41,Receive,#   .   .  12.00   . 
2023.11.29 22:03:40,Send,80 06 03 03 09 04 00 66 
2023.11.29 22:03:40,Receive,(235.0 000.0 236.0 000 50.1 13.7 29.0 00001000

````

mailinglists35, 2023-11-29:
maybe you manage to find which one is the beeper (they call it buzzer)

````
$ cat DebugLog20231129_USB.txt 
Date/Time,State,Command
2023.11.29 22:06:49,Send,80 06 03 03 09 04 00 66 
2023.11.29 22:06:49,Receive,(232.0 000.0 234.0 000 50.0 13.6 29.0 00001000
2023.11.29 22:06:48,Send,80 06 0d 03 09 04 00 66 
2023.11.29 22:06:48,Receive,#   .   .  12.00   . 
2023.11.29 22:06:47,Send,80 06 03 03 09 04 00 66 
2023.11.29 22:06:47,Receive,(234.0 000.0 234.0 000 49.9 13.6 29.0 00001000
2023.11.29 22:06:46,Send,80 06 03 03 09 04 00 66 
2023.11.29 22:06:46,Receive,(233.0 000.0 235.0 000 49.8 13.6 29.0 00001000
2023.11.29 22:06:45,Send,80 06 03 03 09 04 00 66 
2023.11.29 22:06:45,Receive,(232.0 000.0 234.0 000 49.9 13.6 29.0 00001000
2023.11.29 22:06:44,Send,80 06 0c 03 09 04 00 66 
2023.11.29 22:06:44,Receive,#                           V3.8      
2023.11.29 22:06:43,Send,80 06 03 03 09 04 00 66 
2023.11.29 22:06:43,Receive,(231.0 000.0 233.0 000 50.0 13.6 29.0 00001000
2023.11.29 22:06:42,Send,80 06 0d 03 09 04 00 66 
2023.11.29 22:06:42,Receive,#   .   .  12.00   . 
2023.11.29 22:06:41,Send,80 06 03 03 09 04 00 66 
2023.11.29 22:06:41,Receive,(232.0 000.0 233.0 000 49.8 13.6 29.0 00001000
2023.11.29 22:06:40,Send,80 06 07 03 09 04 00 66 
2023.11.29 22:06:40,Receive,UPS No Ack
2023.11.29 22:06:39,Send,80 06 03 03 09 04 00 66 
2023.11.29 22:06:39,Receive,(232.0 000.0 233.0 000 49.8 13.6 29.0 00001000
2023.11.29 22:06:38,Send,80 06 03 03 09 04 00 66 
2023.11.29 22:06:38,Receive,(232.0 000.0 234.0 000 49.9 13.6 29.0 00001000
2023.11.29 22:06:37,Send,80 06 0c 03 09 04 00 66 
2023.11.29 22:06:37,Receive,#                           V3.8      
2023.11.29 22:06:36,Send,80 06 03 03 09 04 00 66 
2023.11.29 22:06:36,Receive,(232.0 000.0 234.0 000 49.9 13.6 29.0 00001000
2023.11.29 22:06:35,Send,80 06 0d 03 09 04 00 66 
2023.11.29 22:06:35,Receive,#   .   .  12.00   . 
2023.11.29 22:06:34,Send,80 06 03 03 09 04 00 66 
2023.11.29 22:06:34,Receive,(232.0 000.0 233.0 000 50.0 13.6 29.0 00001000
2023.11.29 22:06:33,Send,80 06 03 03 09 04 00 66 
2023.11.29 22:06:33,Receive,(232.0 000.0 235.0 000 49.9 13.6 29.0 00001000
2023.11.29 22:06:32,Send,80 06 03 03 09 04 00 66 
2023.11.29 22:06:32,Receive,(231.0 000.0 232.0 000 49.9 13.6 29.0 00001000
2023.11.29 22:06:31,Send,80 06 0c 03 09 04 00 66 
2023.11.29 22:06:31,Receive,#                           V3.8      
2023.11.29 22:06:30,Send,80 06 03 03 09 04 00 66 
2023.11.29 22:06:30,Receive,(231.0 000.0 232.0 000 50.0 13.6 29.0 00001000
2023.11.29 22:06:29,Send,80 06 0d 03 09 04 00 66 
2023.11.29 22:06:29,Receive,#   .   .  12.00   . 
2023.11.29 22:06:28,Send,80 06 03 03 09 04 00 66 
2023.11.29 22:06:28,Receive,(231.0 000.0 231.0 000 49.7 13.6 29.0 00001000
2023.11.29 22:06:27,Send,80 06 03 03 09 04 00 66 
2023.11.29 22:06:27,Receive,(231.0 000.0 232.0 000 50.0 13.6 29.0 00001000
2023.11.29 22:06:26,Send,80 06 03 03 09 04 00 66 
2023.11.29 22:06:26,Receive,(231.0 000.0 231.0 000 49.9 13.6 29.0 00001000
2023.11.29 22:06:25,Send,80 06 0c 03 09 04 00 66 
2023.11.29 22:06:25,Receive,#                           V3.8      
2023.11.29 22:06:24,Send,80 06 03 03 09 04 00 66 
2023.11.29 22:06:24,Receive,(230.0 000.0 231.0 000 49.8 13.6 29.0 00001000
2023.11.29 22:06:23,Send,80 06 0d 03 09 04 00 66 
2023.11.29 22:06:23,Receive,#   .   .  12.00   . 
2023.11.29 22:06:22,Send,80 06 03 03 09 04 00 66 
2023.11.29 22:06:22,Receive,(230.0 000.0 231.0 000 49.9 13.6 29.0 00001000
2023.11.29 22:06:21,Send,80 06 03 03 09 04 00 66 
2023.11.29 22:06:21,Receive,(231.0 000.0 232.0 000 49.7 13.6 29.0 00001000
2023.11.29 22:06:20,Send,80 06 03 03 09 04 00 66 
2023.11.29 22:06:20,Receive,*231.0 000.0 232.0 000 50.0 13.6 29.0 00001000
2023.11.29 22:06:19,Send,80 06 0c 03 09 04 00 66 
2023.11.29 22:06:19,Receive,#                           V3.8      
2023.11.29 22:06:18,Send,80 06 03 03 09 04 00 66 
2023.11.29 22:06:18,Receive,(231.0 000.0 232.0 000 49.7 13.6 29.0 00001000
2023.11.29 22:06:17,Send,80 06 0d 03 09 04 00 66 
2023.11.29 22:06:17,Receive,#   .   .  12.00   . 
2023.11.29 22:06:16,Send,80 06 03 03 09 04 00 66 
2023.11.29 22:06:16,Receive,(231.0 000.0 232.0 000 49.7 13.6 29.0 00001000

````

jimklimov, 2023-11-29:
Thank you very much for the detailed report. There were a couple of mentions over the years that some devices needed an initialization by vendor SW to become visible to NUT (though I had the impression that once per lifetime was enough in those cases), but I don't remember any report on it as detailed as yours!

I've posted on the nut-upsdev mailing list to bring some more attention to this ticket, hopefully someone with Qx drivers' programming experience would chime in and be more relevant here than myself.

mailinglists35, 2023-11-30:
additional info: it seems all readings after vendor sw are the same values, as if I'm reading from a cache. so the command that the UPSmart issues it must be done at each reading, or else we will read not fresh data.

mailinglists35, 2024-06-16:
Hi,
Did anyone respond to your mailing list message?

mailinglists35, 2026-01-16:
I see a commented langid_fix in my ups.conf which means I must have tried it already without success.

mailinglists35, 2026-01-16:
lesson for me is don't buy again something not having been validated already by someone else as working with nut.