networkupstools/nut #2210: [BUG?] "systemd-tmpfiles --clean" return an error

ManuLinares, 2023-12-03:
```
~ $ sudo systemd-tmpfiles --clean
/usr/lib/tmpfiles.d/nut.conf:1: Duplicate line for path "/run/nut", ignoring.
/usr/lib/tmpfiles.d/nut.conf:2: Duplicate line for path "/var/lib/nut", ignoring.

```

```
~ $ cat /usr/lib/tmpfiles.d/nut.conf
d /run/nut 0700 nut nut -
d /var/lib/nut 0700 nut nut -
z /etc/nut/upsd.conf 0640 root nut -
z /etc/nut/upsd.users 0640 root nut -
z /etc/nut/upsmon.conf 0640 root nut -
```

My system: archlinux, latest everything as of 03-dec-2023

jimklimov, 2023-12-03:
I suppose this is a packaging bug in that distribution, with their `nut.conf` conflicting with a `/usr/lib/tmpfiles.d/nut-common-tmpfiles.conf` which may be generated and delivered by the NUT vanilla build (if they package and deliver it as well as theirs).

ManuLinares, 2023-12-04:
Thanks. I don't seem to be able to login to post a bug report in https://gitlab.archlinux.org/archlinux/packaging/packages/nut/-/issues

jimklimov, 2024-01-20:
Ok, seems there's some traction on Arch Linux side. I guess the issue can be closed here as a not-NUT bug :)

closed 2024-01-20 by jimklimov