networkupstools/nut #2211: Official docker-compose image for raspberry arm64v8

kitopopo, 2023-12-03:
Dear friends, Is there currently any official docker compose image for network ups tools for arm64v8?

Thanks in advanced

Best regards

Ariessanea1, 2023-12-03:
No not at this time

kitopopo, 2023-12-03:
> And for the original question - no, but also there are currently no other "official container images", either - it's not like we do not like a particular platform :D
> 
> There are several "third-party community" projects which make containers, notably Home Assistant plugins are containers, I think, and a wiki article details how to roll your own (e.g. from master codebase of NUT to try some fixes, or to develop and locally test PRs, etc.)
> 
> We are not averse to receiving reasonably made and documented PRs with Dockerfile etc. recipes from people who are more confident than e.g. myself it is a sane way to go, and know how to do it well :)

Thanks for your answer. Best regards

kitopopo, 2023-12-03:
It was simply a doubt since I couldn't find any image of a nut container, but you are right. Thanks again.