networkupstools/nut #2212: 我的syslog里有好多OL+DISCHRG的日志

xsj684135, 2023-12-05:
I cannot set onlinedischarge to yes because this will cause my UPS to perform ONBATT when it is fully charged to 100%.
How can I block this log and prevent it from being written to syslog?
I tried NOTIFYFLAG OL+DISCHRG IGNORE, but it still keeps writing.
The frequency of writing is once every 2 seconds, which greatly affects my ability to read syslog.

Thanks.

`Dec 05 12:00:37 HPE usbhid-ups[78528]: ups_status_set: seems that UPS [BK650M2-CH] is in OL+DISCHRG state now. Is it calibrating or do you perhaps want to set 'onlinedischarge' option? Some UPS models (e.g. CyberPower UT series) emit OL+DISCHRG when offline.`

xsj684135, 2023-12-05:
Can anyone help me?
Because my native language is not English, I can only rely on translation to read the document, which is a bit difficult to understand.

xsj684135, 2023-12-05:
> [您好，这个问题似乎与#1970](https://github.com/networkupstools/nut/issues/1970)重复，所以我交叉引用了这两个问题。
> 
> 您运行的是哪个 NUT 版本？您是否可以从源代码构建它，例如遵循[https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80% 90disruptive-tests](https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests) - 测试较新的 GitHub 主代码库或某些较新的实验代码是否可以解决此问题？
> 
> 另外，我是否理解正确，您的 UPS 仅在充电至 100%（和/或超过某个百分比，如 90-95%）时报告 OL+DISCHRG，但在部分放电时则不报告？我想知道他们是否有一些化学保护模式来避免电池保持充满电，就像现在许多手机/笔记本电脑试图延长电池寿命一样。
> 
> 这些`NOTIFYFLAG`设置仅采用某些值来命名被认为重要的设备状态/事件`upsmon`，如示例配置文件和您正在使用的 NUT 版本的手册页中`upsmon`所述`upsmon.conf`。不`OL+DISCHRG`属于任何此类字符串。

I installed it directly from apt, and the version is Network UPS Tools upsc 2.8.0. Sorry, I can't compile it from the source code.

It will only prompt OL+DISCHRG when it is fully charged to 100%, which is why I cannot use apcupsd.

This version is the M2 China exclusive version. I searched on domestic forums and it seems that they all have this problem.

But nut can be used, and it can trigger ONBATT to run a custom script when the power is off. However, I cannot accept that it keeps writing logs to syslog, so is there a way to prevent it from writing logs in syslog?

xsj684135, 2023-12-05:
It seems that I have to find another way to prevent syslog writing and wait for version 2.8.2 to be released. thank you very much!

jimklimov, 2023-12-05:
FWIW, if you use e.g. rsyslog-ng, it can filter messages (among other neat features)...

jimklimov, 2023-12-07:
Cheers, I've posted a pull request with changes to logging of these messages. Would be great if someone with the "troublesome" (mostly some APC or CPS) devices could build that code branch and confirm it behaves as theoretically expected :)

https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests

PR source at https://github.com/jimklimov/nut/tree/issue-1970-throttle or for code building purposes:
````
:; git clone https://github.com/jimklimov/nut.git -b issue-1970-throttle nut-issue-1970
:; cd nut-issue-1970
# ... follow the Wiki about prerequisites and build procedure
````

xsj684135, 2023-12-07:
> 干杯，我已经发布了一个拉取请求，其中包含对这些消息日志记录的更改。如果拥有“麻烦”（主要是一些 APC 或 CPS）设备的人可以构建该代码分支并确认其行为符合理论上的预期，那就太好了:)
> 
> https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests
> 
> PR 来源位于https://github.com/jimklimov/nut/tree/issue-1970-throttle或用于代码构建目的：
> 
> ```
> :; git clone https://github.com/jimklimov/nut.git -b issue-1970-throttle nut-issue-1970
> :; cd nut-issue-1970
> # ... follow the Wiki about prerequisites and build procedure
> ```

So quickly? Thanks I'll try it now！！

PR #2216 (2023-12-10): `usbhid-ups`: allow to throttle or suppress logged messages about `OL+DISCHRG` situation
Closes: #1970
Closes: #2212
Closes: #2213
Closes: #2214
Closes: #2215

Introduces a number of new configurable toggles with hopefully reasonable defaults, which should help to reduce log noise on systems where an UPS reports "online" and "discharging" simultaneously (so far some CyberPower and APC models were reported to behave so, with different physical situations).
closed 2023-12-10 by jimklimov