networkupstools/nut #2217: ".1.3.6.1.4.1.5491.6" sysOID

masterx1981, 2023-12-05:
Hi!
I have a Riello UPS, with netman 102 plus network card, firmware 1.21 (latest one that i've found). I'm trying to connect it via snmp to the nut server hosted on an home assistant installation.
The ups isn't recognized, as from the log seem that is using a not already coded sysOID.
I have the mib files (that i attach to the post).
[MIBs.zip](https://github.com/networkupstools/nut/files/13572426/MIBs.zip)
How i can implement the mib files on my installation?
Thanks!

masterx1981, 2023-12-06:
> Hi, thanks for the question and archives!
> 
> I am not sure all the MIBs in the ZIP are even all valid - it seems they took a generated (paginated) text file for e.g. RFC3433 and spliced its text with MIB markup. Most of the paginations landed into text-convention blocks, but some are outside (e.g. Pages 10, 11, 12) - at element-definition level :-\
> 
> Back to your question: https://github.com/networkupstools/nut/blob/master/docs/snmp-subdrivers.txt / rendered version at https://networkupstools.org/docs/developer-guide.chunked/new-drivers.html#snmp-subdrivers has a chapter on "Creating a subdriver". It is at least a good starting point...

Thanks for the fast answer!
I'll try to follow the guides, in the hope than doing it in an haasio installation isn't too difficilt...
As soon as i have some free time i'll play with it, thanks!

masterx1981, 2023-12-08:
Sorry, i've seen that the snmp walk kog identifies 2 different oids, one is the unknown one l, the other one is OID=.1.3.6.1.2.1.33, that seem standard ietf, and the voltage/frequency seem that are under this id.
But why aren't read by the nut service?
I'm using this device configuration:
`- name: riello
  driver: snmp-ups
  port: 192.168.1.183
  desc: Riello Sentinel 5000VA
  snmp_version: v1
  mibs: ietf
  pollinterval: 15
  upsmon_deadtime: 25
  config: []
`

masterx1981, 2023-12-08:
> Hm, that seems to be missing a `community: public` line (or whatever your credential is - note that for instant commands and variable setting, you'd need to enable an RW community like "private" on the device and pass it to NUT).

I've not put the community line as on the configuration manuals is specified that the community option have a default value of "public", that match the read only community set on the interface card. In sny case, i'll try to add it.

And, how i can set the write community? The neteork card is set to "private"

jimklimov, 2023-12-08:
Yeah, "public" should be default, even per codebase:
````
drivers/snmp-ups.c:               community = testvar(SU_VAR_COMMUNITY) ? getval(SU_VAR_COMMUNITY) : "public";
````

Maybe you can bump `debug_min` for that driver (or as a `NUT_DEBUG_LEVEL` envvar) to show more logs - in particular, trace if it gets the `mibs` setting and what else it tries? I have no idea how HA is set up and what info it passes into the actual NUT configs, so have a hard time figuring out from the piecemeal info if it is even a NUT bug or a HA bug...

That said, normally `snmp-ups` without a (seen) `mibs` parameter should try all known mappings and `ietf` in the end. So it probably stumbles upon seeing a reply from the device. Maybe the HA firewall is playing tricks or something - `apc_ats` is likely the first MIB tried:

````
$ NUT_STATEPATH=/tmp ./drivers/snmp-ups -x mibs=--list -s test -x port=x
Network UPS Tools - Generic SNMP UPS driver 1.30 (2.8.1-221-gd8b254dbe)
The 'mibs' argument is '--list', so just listing the mappings this driver knows,
and for 'mibs=auto' these mappings will be tried in the following order until
the first one matches your device

 NUMBER MAPPING NAME            VERSION ENTRY POINT OID                 AUTO CHECK OID
    1   apc_ats                     0.6 .1.3.6.1.4.1.318.1.3.11         .1.3.6.1.4.1.318.1.1.8.1.5.0
    2   apc_pdu                     0.4 .1.3.6.1.4.1.318.1.3.4.4        .1.3.6.1.4.1.318.1.1.4.1.4.0
    3   apc_pdu                     0.4 .1.3.6.1.4.1.318.1.3.4.5        .1.3.6.1.4.1.318.1.1.4.1.4.0
    4   apc_pdu                     0.4 .1.3.6.1.4.1.318.1.3.4.6        .1.3.6.1.4.1.318.1.1.4.1.4.0
    5   apc                         0.1 .1.3.6.1.4.1.318.1.3.4.9        <NULL>
    6   apcc                        1.6 .1.3.6.1.4.1.318.1.1.1.1.1.1.0  .1.3.6.1.4.1.318.1.1.1.1.1.1.0
    7   baytech                  0.4035 .1.3.6.1.4.1.4779               .1.3.6.1.4.1.4779.1.3.5.2.1.24.1
    8   bestpower                   0.4 .1.3.6.1.4.1.2947.1.1.2.0       .1.3.6.1.4.1.2947.1.1.2.0
    9   cpqpower                   1.67 .1.3.6.1.4.1.232.165.3          .1.3.6.1.4.1.232.165.3.1.1.0
   10   cyberpower                 0.55 .1.3.6.1.4.1.3808.1.1.1         .1.3.6.1.4.1.3808.1.1.1.1.1.1.0
   11   cyberpower                 0.55 .1.3.6.1.4.1.3808               .1.3.6.1.4.1.3808.1.1.1.1.1.1.0
   12   delta_ups                   0.5 .1.3.6.1.4.1.2254.2.4           <NULL>
   13   eaton_ats16_nmc            0.21 .1.3.6.1.4.1.705.1              .1.3.6.1.4.1.534.10.2.1.2.0
   14   eaton_ats16_nm2            0.22 .1.3.6.1.4.1.534.10.2           .1.3.6.1.4.1.534.10.2.1.2.0
   15   eaton_ats30                0.03 .1.3.6.1.4.1.534.10.1           .1.3.6.1.4.1.534.10.1.2.1.0
   16   eaton_epdu                 0.69 .1.3.6.1.4.1.534.6.6.7          .1.3.6.1.4.1.534.6.6.7.1.2.1.2.0
   17   eaton_pdu_nlogic            0.1 .1.3.6.1.4.1.534.7.1            <NULL>
   18   eaton_pxg_ups             0.105 .1.3.6.1.4.1.534.2.12           1.3.6.1.4.1.534.1.1.2.0
   19   eaton_pw_nm2              0.105 .1.3.6.1.4.1.534.1              1.3.6.1.4.1.534.1.1.2.0
   20   emerson_avocent_pdu         1.3 .1.3.6.1.4.1.10418.17.1.7       .1.3.6.1.4.1.10418.17.2.1.2.0
   21   aphel_revelation           0.52 .1.3.6.1.4.1.534.6.6.6          .1.3.6.1.4.1.534.6.6.6.1.1.12.0
   22   aphel_genesisII            0.52 .1.3.6.1.4.1.17373              .1.3.6.1.4.1.17373.3.1.1.0
   23   pulizzi_switched1           0.5 .1.3.6.1.4.1.20677.1            .1.3.6.1.4.1.20677.1
   24   pulizzi_switched2           0.5 .1.3.6.1.4.1.20677.2            .1.3.6.1.4.1.20677.1
   25   hpe_epdu                   0.33 .1.3.6.1.4.1.232.165.7          .1.3.6.1.4.1.232.165.7.1.2.1.3.0
   26   hpe_pdu3_cis                0.1 .1.3.6.1.4.1.232.165.11         .1.3.6.1.4.1.232.165.11.1.2.1.3.1
   27   huawei                      0.4 .1.3.6.1.4.1.8072.3.2.10        .1.3.6.1.4.1.2011.6.174.1.2.100.1.2.1
   28   mge                        0.55 .1.3.6.1.4.1.705.1              .1.3.6.1.4.1.705.1.1.1.0
   29   netvision                  0.44 .1.3.6.1.4.1.4555.1.1.1         .1.3.6.1.4.1.4555.1.1.1.1.1.1.0
   30   raritan                     0.8 .1.3.6.1.4.1.13742              .1.3.6.1.4.1.13742.1.1.12.0
   31   raritan-px2                 0.4 .1.3.6.1.4.1.13742.6            .1.3.6.1.4.1.13742.6.3.2.1.1.3.1
   32   xppc                        0.4 .1.3.6.1.4.1.935                <NULL>
   33   tripplite                  1.55 .1.3.6.1.4.1.850.1              <NULL>
   34   ietf                       1.55 .1.3.6.1.2.1.33                 1.3.6.1.2.1.33.1.1.1.0

Overall this driver has loaded 34 MIB-to-NUT mapping tables
upsnotify: notify about state 4 with libsystemd: was requested, but not running as a service unit now, will not spam more about it
upsnotify: failed to notify about state 4: no notification tech defined, will not spam more about it
````

jimklimov, 2023-12-08:
Well, here it seems NUT `snmp-ups` driver does get a reply to initial query for a `sysOID` (roughly "where do I get best detailed info?") which is usually a vendor-extension branch of the data tree.

That sysOID is not known to our mapping tables (see generation of subdriver discussed above), so `snmp-ups` falls back to picking one. Apparently the `mibs` setting does not make it through HA. Can you log into the container (exec a shell in it) or otherwise look at the `ups.conf` file that HA creates?

Not sure what their `log_level: debug` sets up, but I've seen just a few lines with `[D1]` in your recent post (and a statement of "debug level: 1") so it is a minimally verbose mode...

masterx1981, 2023-12-09:
> Well, here it seems NUT `snmp-ups` driver does get a reply to initial query for a `sysOID` (roughly "where do I get best detailed info?") which is usually a vendor-extension branch of the data [...]

I'm trying a bit to understand the snmp message structure, comparing mib with snmpwalk, i understad what you was saying, and that that is right. On oid .1.3.6.1.2.1.1.2.0 my card send an oid of 1.3.6.1.4.1.5491.6, that is the one not recognized. When nut recognize that it not know this oid, simply fail, ignoring as you said the mib definition specified in the config.

jimklimov, 2023-12-09:
To help dissect which part of the data conveyor belt has the actual issue, can you run NUT on the native system, where you have control over configs made, debugs collected, etc.?

That wiki article on "in-place" builds has links to a document with prerequisite packages for various platforms, and stipulates that you don't have to `make install` NUT into your system and how you can run the freshly built drivers right from that workspace to test them.

masterx1981, 2023-12-09:
As soon as i have some spare time i try to get access to the bare hassos, and move from there to spot the problem. Seem a not easy procedure that involve create a certificate and put it on the hassos for allow connections on port 22222.
Really thanks for the help!

jimklimov, 2023-12-11:
CC @frenck : a HA NUT plugin user here struggled with `snmp-ups` apparently not passing the `mibs` setting to the driver... can you please confirm if this keyword is properly supported by the plugin?

Ripmp4, 2024-09-09:
Hi, I am also interested in this topic. I have the exact same problem with my Riello UPS. Any follow-up on this ? Thanks !

masterx1981, 2024-09-09:
> Hi, I am also interested in this topic. I have the exact same problem with my Riello UPS. Any follow-up on this ? Thanks !

I've not resolved... i want try to downgrade the firmware of the network card, i have only the 1.18. From what i've read, the previous versions of the firmware send different  snmp messages, that i hope are compatible with nut and home assistant.
Else, home assistant have an snmp engine that is able to read custom messages, and can be configured to read anything, without using nut.

masterx1981, 2025-03-16:
Reverted to fw ver 1.18, nothing changed...
The network card sends the same id...
Need to fix the problem in the nut version of home assistant...

masterx1981, 2025-03-16:
Damn! Got it working!
The intergation config in HA must be:
```
- name: riello
  driver: snmp-ups
  port: 192.168.1.XXX
  config:
    - mibs = ietf
```
 Instead of
```
- name: riello
  driver: snmp-ups
  port: 192.168.1.XXX
  config: []
- mibs = ietf
```
In this way the mibs parameter is managed correctly by nut!