networkupstools/nut #2218: Highlighting specific cells in the web display.

mjksohonet, 2023-12-07:
NUT 2.8.0 (from the Ubuntu 23.10 packages) running on Ubuntu 22.04.
I would like to be able to highlight cells in the web page where the value has breached a threshold.
For example, I would like to draw attention to any UPS devices where the load has risen to more than 50%.
Is this possible?
I tried doing
```
@IFBETWEEN 50 100 ups.load@
<td BGCOLOR="#FFD133">
@VAR ups.load@
%
@ELSE@
<td BGCOLOR="#00FF00">
@VAR ups.load@
%
```
but that changes the background colour for all the cells in the column, regardless of the value.

jimklimov, 2023-12-10:
CC @aquette : cheers, any pointers on CGI HTML markup?

jimklimov, 2023-12-12:
Can't quickly say... but you can probably run it in a debugger (or otherwise pepper a custom build with `upsdebugx()` trace printouts) and check this theory. If proven, a PR with a fix would be most welcome :)

mjksohonet, 2023-12-13:
The thing is, I'm not a developer and I have zero experience in debugging cgi code.
This should be a fairly easy issue to reproduce, or I'm doing something wrong.