networkupstools/nut #2219: nutdrv_qx: RICHCOMM UPS USB Mon V2.0 - Vultech UPS1400VA-LFP

calillo, 2023-12-09:
I'm trying to run this Vultech UPS1400VA-LFP, I compiled the latest version but it doesn't work, it seems the RICHCOMM UPS USB Mon V2.0 is not supported. My issue is probably identical to #2153.
Is it possible to add support?

Thanks

./nutdrv_qx -s vultech -u root -DDDDDD -x productid=1234 -x vendorid=0925 -x subdriver=armac -x port=auto
```
Network UPS Tools - Generic Q* USB/Serial driver 0.36 (2.8.1-251-g232c853d6)
USB communication driver (libusb 1.0) 0.46
   0.000000     [D3] main_arg: var='productid' val='1234'
   0.000031     [D5] send_to_all: SETINFO driver.parameter.productid "1234"
   0.000051     [D3] main_arg: var='vendorid' val='0925'
   0.000062     [D5] send_to_all: SETINFO driver.parameter.vendorid "0925"
   0.000070     [D3] main_arg: var='subdriver' val='armac'
   0.000078     [D5] send_to_all: SETINFO driver.parameter.subdriver "armac"
   0.000084     [D3] main_arg: var='port' val='auto'
   0.000093     [D6] testinfo_reloadable: var=port, infoname=driver.parameter.port, newval=auto, reloadable=0, reload_flag=0
   0.000100     [D6] testinfo_reloadable: verdict for (re)loading var=port value: 1
   0.000108     [D5] send_to_all: SETINFO driver.parameter.port "auto"
   0.000115     [D1] Network UPS Tools version 2.8.1-251-g232c853d6 (release/snapshot of 2.8.1.1) built with gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0 and configured with flags: --enable-Wcolor --enable-warnings --enable-Werror --enable-keep_nut_report_feature --with-all=auto --with-cgi=auto --with-serial=auto --with-dev=auto --with-doc=skip --with-nut_monitor=auto --with-pynut=auto --disable-force-nut-version-header --enable-check-NIT --enable-maintainer-mode
   0.000122     [D1] debug level is '6'
   0.000133     [D5] send_to_all: SETINFO driver.debug "6"
   0.000146     [D5] send_to_all: SETFLAGS driver.debug RW NUMBER
   0.000317     [D1] Succeeded to become_user(root): now UID=0 GID=0
   0.000334     [D5] send_to_all: SETINFO device.type "ups"
   0.000341     [D5] send_to_all: SETINFO driver.state "init.device"
   0.000348     [D1] upsdrv_initups...
   0.002406     [D2] Checking device 1 of 4 (1D6B/0003)
   0.024138     [D1] nut_libusb_open: invalid libusb bus number 0
   0.024390     [D2] - VendorID: 1d6b
   0.024403     [D2] - ProductID: 0003
   0.024409     [D2] - Manufacturer: Linux 4.4.302+ xhci-hcd
   0.024415     [D2] - Product: xHCI Host Controller
   0.024420     [D2] - Serial Number: 0000:00:15.0
   0.024426     [D2] - Bus: 002
   0.024431     [D2] - Bus Port: unknown
   0.024436     [D2] - Device: 001
   0.024442     [D2] - Device release number: 0404
   0.024447     [D2] Trying to match device
   0.024453     [D3] match_function_regex: matching a device...
   0.024463     [D2] match_function_regex: failed match of VendorID: 1d6b
   0.024468     [D2] Device does not match - skipping
   0.025188     [D2] Checking device 2 of 4 (F400/F400)
   0.028619     [D2] - VendorID: f400
   0.028698     [D2] - ProductID: f400
   0.028704     [D2] - Manufacturer: Synology
   0.028710     [D2] - Product: DiskStation
   0.028715     [D2] - Serial Number: 6500265D8B13F221
   0.028721     [D2] - Bus: 001
   0.028726     [D2] - Bus Port: 004
   0.028732     [D2] - Device: 003
   0.028737     [D2] - Device release number: 0100
   0.028743     [D2] Trying to match device
   0.028749     [D3] match_function_regex: matching a device...
   0.028781     [D2] match_function_regex: failed match of VendorID: f400
   0.028788     [D2] Device does not match - skipping
   0.028854     [D2] Checking device 3 of 4 (0925/1234)
   0.030102     [D2] - VendorID: 0925
   0.030211     [D2] - ProductID: 1234
   0.030230     [D2] - Manufacturer: RICHCOMM
   0.030245     [D2] - Product: UPS USB Mon V2.0
   0.030282     [D2] - Serial Number: unknown
   0.030308     [D2] - Bus: 001
   0.030314     [D2] - Bus Port: 001
   0.030319     [D2] - Device: 002
   0.030325     [D2] - Device release number: 0000
   0.030330     [D2] Trying to match device
   0.030336     [D3] match_function_regex: matching a device...
   0.030372     [D2] Device matches
   0.030379     [D2] Reading configuration descriptor 1 of 1
[... 118 lines cut ...]
                            Preferred_State No_Null_Position Non_Volatile Bitfield
            Item(Main  ): End Collection, data=none
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)
```

jimklimov, 2023-12-09:
To check if it is the same issue, can you try building that PR - if it fixes your situation?

calillo, 2023-12-10:
yes! I confirm with `git clone -b armac-2023 https://github.com/vytautassurvila/nut` it works and I can read some data

```
Network UPS Tools - Generic Q* USB/Serial driver 0.36 (2.8.1.1)
USB communication driver (libusb 1.0) 0.46
   0.000001	[D3] main_arg: var='productid' val='1234'
   0.000021	[D5] send_to_all: SETINFO driver.parameter.productid "1234"
   0.000031	[D3] main_arg: var='vendorid' val='0925'
   0.000038	[D5] send_to_all: SETINFO driver.parameter.vendorid "0925"
   0.000042	[D3] main_arg: var='subdriver' val='armac'
   0.000047	[D5] send_to_all: SETINFO driver.parameter.subdriver "armac"
   0.000051	[D3] main_arg: var='port' val='auto'
   0.000057	[D6] testinfo_reloadable: var=port, infoname=driver.parameter.port, newval=auto, reloadable=0, reload_flag=0
   0.000062	[D6] testinfo_reloadable: verdict for (re)loading var=port value: 1
   0.000067	[D5] send_to_all: SETINFO driver.parameter.port "auto"
   0.000070	[D1] Network UPS Tools version 2.8.1.1 (release/snapshot of 2.8.1.1) built with gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0 and configured with flags: --enable-Wcolor --enable-warnings --enable-Werror --enable-keep_nut_report_feature --with-all=auto --with-cgi=auto --with-serial=auto --with-dev=auto --with-doc=skip --with-nut_monitor=auto --with-pynut=auto --disable-force-nut-version-header --enable-check-NIT --enable-maintainer-mode
   0.000079	[D1] debug level is '6'
   0.000083	[D5] send_to_all: SETINFO driver.debug "6"
   0.000093	[D5] send_to_all: SETFLAGS driver.debug RW NUMBER
   0.000245	[D1] Succeeded to become_user(root): now UID=0 GID=0
   0.000261	[D5] send_to_all: SETINFO device.type "ups"
   0.000268	[D5] send_to_all: SETINFO driver.state "init.device"
   0.000275	[D1] upsdrv_initups...
   0.002951	[D2] Checking device 1 of 4 (1D6B/0003)
   0.024247	[D1] nut_libusb_open: invalid libusb bus number 0
   0.024457	[D2] - VendorID: 1d6b
   0.024466	[D2] - ProductID: 0003
   0.024470	[D2] - Manufacturer: Linux 4.4.302+ xhci-hcd
   0.024474	[D2] - Product: xHCI Host Controller
   0.024478	[D2] - Serial Number: 0000:00:15.0
   0.024481	[D2] - Bus: 002
   0.024485	[D2] - Bus Port: unknown
   0.024488	[D2] - Device: 001
   0.024491	[D2] - Device release number: 0404
   0.024494	[D2] Trying to match device
   0.024516	[D3] match_function_regex: matching a device...
   0.024523	[D2] match_function_regex: failed match of VendorID: 1d6b
   0.024527	[D2] Device does not match - skipping
   0.024820	[D2] Checking device 2 of 4 (F400/F400)
   0.027416	[D2] - VendorID: f400
   0.027425	[D2] - ProductID: f400
   0.027430	[D2] - Manufacturer: Synology
   0.027435	[D2] - Product: DiskStation
   0.027441	[D2] - Serial Number: 6500265D8B13F221
   0.027446	[D2] - Bus: 001
   0.027451	[D2] - Bus Port: 004
   0.027458	[D2] - Device: 003
   0.027464	[D2] - Device release number: 0100
   0.027470	[D2] Trying to match device
   0.027476	[D3] match_function_regex: matching a device...
   0.027501	[D2] match_function_regex: failed match of VendorID: f400
   0.027508	[D2] Device does not match - skipping
   0.027525	[D2] Checking device 3 of 4 (0925/1234)
   0.028171	[D2] - VendorID: 0925
   0.028179	[D2] - ProductID: 1234
   0.028183	[D2] - Manufacturer: RICHCOMM
   0.028186	[D2] - Product: UPS USB Mon V2.0
   0.028189	[D2] - Serial Number: unknown
   0.028192	[D2] - Bus: 001
   0.028195	[D2] - Bus Port: 001
   0.028202	[D2] - Device: 002
   0.028209	[D2] - Device release number: 0000
   0.028215	[D2] Trying to match device
   0.028218	[D3] match_function_regex: matching a device...
   0.028242	[D2] Device matches
   0.028249	[D2] Reading configuration descriptor 1 of 1
   0.028259	[D3] libusb_kernel_driver_active() returned 0: Success
   0.028277	[D2] Claimed interface 0 successfully
   0.028284	[D3] nut_usb_set_altinterface: skipped libusb_set_interface_alt_setting(udev, 0, 0)
   0.028297	[D5] send_to_all: SETINFO ups.vendorid "0925"
[... 191 lines cut ...]
   3.633742	[D5] update_status: !FSD
   3.633761	[D5] send_to_all: SETINFO driver.state "quiet"
   5.383271	[D5] send_to_all: SETINFO driver.state "updateinfo"
   5.383315	[D1] upsdrv_updateinfo...
   5.383328	[D1] Quick update...
   5.383342	[D4] armac command Q1
   5.638321	[D4] read: ret 64 buf ef: 28 32 33 30 2e  >(230.<
   5.638346	[D3] armac command Q1 response read: '(230.0 000.0 230.0 003 50.0 13.7 30.8 00001000'
   5.638355	[D5] update_status: OL
   5.638361	[D5] update_status: !LB
   5.638367	[D5] update_status: !CAL
   5.638376	[D5] update_status: !FSD
   5.638392	[D5] send_to_all: SETINFO driver.state "quiet"
   7.385297	[D5] send_to_all: SETINFO driver.state "updateinfo"
   7.385339	[D1] upsdrv_updateinfo...
   7.385370	[D1] Quick update...
   7.385384	[D4] armac command Q1
   7.643323	[D4] read: ret 64 buf af: 28 32 33 30 2e  >(230.<
   7.643345	[D3] armac command Q1 response read: '(230.0 000.0 230.0 003 49.0 13.7 30.8 00001000'
   7.643358	[D5] update_status: OL
   7.643366	[D5] update_status: !LB
   7.643375	[D5] update_status: !CAL
   7.643380	[D5] update_status: !FSD
   7.643395	[D5] send_to_all: SETINFO driver.state "quiet"
```

jimklimov, 2023-12-10:
Sounds great, thanks! :) That brings more confidence in that PR :)

Tux92, 2024-01-09:
> yes @jimklimov! I confirm with `git clone -b armac-2023 https://github.com/vytautassurvila/nut` it works and I can read some data
> 
> ./nutdrv_qx -s vultech -u root -DDDDDD -x productid=1234 -x vendorid=0925 -x subdriver=armac -x port=auto
> 
> ```
> Network UPS Tools - Generic Q* USB/Serial driver 0.36 (2.8.1.1)
> USB communication driver (libusb 1.0) 0.46
>    0.000001	[D3] main_arg: var='productid' val='1234'
>    0.000021	[D5] send_to_all: SETINFO driver.parameter.productid "1234"
>    0.000031	[D3] main_arg: var='vendorid' val='0925'
>    0.000038	[D5] send_to_all: SETINFO driver.parameter.vendorid "0925"
>    0.000042	[D3] main_arg: var='subdriver' val='armac'
>    0.000047	[D5] send_to_all: SETINFO driver.parameter.subdriver "armac"
>    0.000051	[D3] main_arg: var='port' val='auto'
>    0.000057	[D6] testinfo_reloadable: var=port, infoname=driver.parameter.port, newval=auto, reloadable=0, reload_flag=0
>    0.000062	[D6] testinfo_reloadable: verdict for (re)loading var=port value: 1
>    0.000067	[D5] send_to_all: SETINFO driver.parameter.port "auto"
>    0.000070	[D1] Network UPS Tools version 2.8.1.1 (release/snapshot of 2.8.1.1) built with gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0 and configured with flags: --enable-Wcolor --enable-warnings --enable-Werror --enable-keep_nut_report_feature --with-all=auto --with-cgi=auto --with-serial=auto --with-dev=auto --with-doc=skip --with-nut_monitor=auto --with-pynut=auto --disable-force-nut-version-header --enable-check-NIT --enable-maintainer-mode
>    0.000079	[D1] debug level is '6'
>    0.000083	[D5] send_to_all: SETINFO driver.debug "6"
>    0.000093	[D5] send_to_all: SETFLAGS driver.debug RW NUMBER
>    0.000245	[D1] Succeeded to become_user(root): now UID=0 GID=0
>    0.000261	[D5] send_to_all: SETINFO device.type "ups"
>    0.000268	[D5] send_to_all: SETINFO driver.state "init.device"
>    0.000275	[D1] upsdrv_initups...
>    0.002951	[D2] Checking device 1 of 4 (1D6B/0003)
>    0.024247	[D1] nut_libusb_open: invalid libusb bus number 0
>    0.024457	[D2] - VendorID: 1d6b
>    0.024466	[D2] - ProductID: 0003
>    0.024470	[D2] - Manufacturer: Linux 4.4.302+ xhci-hcd
>    0.024474	[D2] - Product: xHCI Host Controller
>    0.024478	[D2] - Serial Number: 0000:00:15.0
>    0.024481	[D2] - Bus: 002
>    0.024485	[D2] - Bus Port: unknown
>    0.024488	[D2] - Device: 001
>    0.024491	[D2] - Device release number: 0404
>    0.024494	[D2] Trying to match device
>    0.024516	[D3] match_function_regex: matching a device...
>    0.024523	[D2] match_function_regex: failed match of VendorID: 1d6b
>    0.024527	[D2] Device does not match - skipping
>    0.024820	[D2] Checking device 2 of 4 (F400/F400)
>    0.027416	[D2] - VendorID: f400
>    0.027425	[D2] - ProductID: f400
>    0.027430	[D2] - Manufacturer: Synology
>    0.027435	[D2] - Product: DiskStation
>    0.027441	[D2] - Serial Number: 6500265D8B13F221
>    0.027446	[D2] - Bus: 001
>    0.027451	[D2] - Bus Port: 004
>    0.027458	[D2] - Device: 003
>    0.027464	[D2] - Device release number: 0100
>    0.027470	[D2] Trying to match device
>    0.027476	[D3] match_function_regex: matching a device...
>    0.027501	[D2] match_function_regex: failed match of VendorID: f400
>    0.027508	[D2] Device does not match - skipping
>    0.027525	[D2] Checking device 3 of 4 (0925/1234)
>    0.028171	[D2] - VendorID: 0925
>    0.028179	[D2] - ProductID: 1234
>    0.028183	[D2] - Manufacturer: RICHCOMM
>    0.028186	[D2] - Product: UPS USB Mon V2.0
>    0.028189	[D2] - Serial Number: unknown
>    0.028192	[D2] - Bus: 001
>    0.028195	[D2] - Bus Port: 001
>    0.028202	[D2] - Device: 002
>    0.028209	[D2] - Device release number: 0000
>    0.028215	[D2] Trying to match device
>    0.028218	[D3] match_function_regex: matching a device...
>    0.028242	[D2] Device matches
>    0.028249	[D2] Reading configuration descriptor 1 of 1
>    0.028259	[D3] libusb_kernel_driver_active() returned 0: Success
>    0.028277	[D2] Claimed interface 0 successfully
[... 195 lines cut ...]
>    5.383271	[D5] send_to_all: SETINFO driver.state "updateinfo"
>    5.383315	[D1] upsdrv_updateinfo...
>    5.383328	[D1] Quick update...
>    5.383342	[D4] armac command Q1
>    5.638321	[D4] read: ret 64 buf ef: 28 32 33 30 2e  >(230.<
>    5.638346	[D3] armac command Q1 response read: '(230.0 000.0 230.0 003 50.0 13.7 30.8 00001000'
>    5.638355	[D5] update_status: OL
>    5.638361	[D5] update_status: !LB
>    5.638367	[D5] update_status: !CAL
>    5.638376	[D5] update_status: !FSD
>    5.638392	[D5] send_to_all: SETINFO driver.state "quiet"
>    7.385297	[D5] send_to_all: SETINFO driver.state "updateinfo"
>    7.385339	[D1] upsdrv_updateinfo...
>    7.385370	[D1] Quick update...
>    7.385384	[D4] armac command Q1
>    7.643323	[D4] read: ret 64 buf af: 28 32 33 30 2e  >(230.<
>    7.643345	[D3] armac command Q1 response read: '(230.0 000.0 230.0 003 49.0 13.7 30.8 00001000'
>    7.643358	[D5] update_status: OL
>    7.643366	[D5] update_status: !LB
>    7.643375	[D5] update_status: !CAL
>    7.643380	[D5] update_status: !FSD
>    7.643395	[D5] send_to_all: SETINFO driver.state "quiet"
> ```

Hy @calillo, I have exact same UPS, cloud you write here a complete guide cause I'm a newbiee on ups world and I don't managed to work this UPS with Proxmox.

jimklimov, 2024-01-09:
@Tux92 : the guide would be likely https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests (modulo the `git clone` command posted above to point to that PR branch... hope to review and merge such technical debts into NUT master soon-ish). You would need to pre-install some tools and other dependencies as referenced from that article; for those purposes Proxmox is a Debian(-like) system.

Tux92, 2024-01-09:
> @Tux92 : the guide would be likely https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests (modulo the `git clone` command posted above [...]

Ok, i remove previuos versione, then i install prerequisites anche clone `git clone -b armac-2023 https://github.com/vytautassurvila/nut`

Next I `cd nut`

Execute `./autogen.sh`

Finally `./configure && make && make install`

No error on any of previous commands.

So I try: `drivers/nutdrv_qx -s vultech -u root -DDDDDD -x productid=1234 -x vendorid=0925 -x subdriver=armac -x port=auto`

But I have this output:

```
Network UPS Tools - Generic Q* Serial driver 0.36 (2.8.1.1)
   0.000000     [D3] main_arg: var='productid' val='1234'

Fatal error: 'productid' is not a valid variable name for this driver.

Look in the man page or call this driver with -h for a list of
valid variable names and flags.
   0.000040     [D5] send_to_all: SETINFO driver.state "cleanup.exit"
   0.000048     upsnotify: failed to notify about state 4: no notification tech defined, will not spam more about it
```

Any ideas?

jimklimov, 2024-01-09:
Check that you've installed libusb-dev or equivalent for your distro. Without it, nutdrv_qx is serial-only.

Tux92, 2024-01-10:
I'm on Proxmox so Debian distro, I run: `apt install libusb-dev`, then I reboot and my output is the same:

```
Network UPS Tools - Generic Q* Serial driver 0.36 (2.8.1.1)
   0.000000     [D3] main_arg: var='productid' val='1234'

Fatal error: 'productid' is not a valid variable name for this driver.

Look in the man page or call this driver with -h for a list of
valid variable names and flags.
   0.000040     [D5] send_to_all: SETINFO driver.state "cleanup.exit"
   0.000048     upsnotify: failed to notify about state 4: no notification tech defined, will not spam more about it
```

Tux92, 2024-01-10:
Update: I recompile and now with this command `/usr/local/ups/bin/nutdrv_qx -s vultech -u root -DDDDDD -x productid=1234 -x vendorid=0925 -x subdriver=armac -x port=auto` I see this output:
```
   0.000000     [D3] main_arg: var='productid' val='1234'
   0.000025     [D5] send_to_all: SETINFO driver.parameter.productid "1234"
   0.000028     [D3] main_arg: var='vendorid' val='0925'
   0.000031     [D5] send_to_all: SETINFO driver.parameter.vendorid "0925"
   0.000034     [D3] main_arg: var='subdriver' val='armac'
   0.000037     [D5] send_to_all: SETINFO driver.parameter.subdriver "armac"
   0.000038     [D3] main_arg: var='port' val='auto'
   0.000040     [D6] testinfo_reloadable: var=port, infoname=driver.parameter.port, newval=auto, reloadable=0, reload_flag=0
   0.000041     [D6] testinfo_reloadable: verdict for (re)loading var=port value: 1
   0.000043     [D5] send_to_all: SETINFO driver.parameter.port "auto"
   0.000044     [D1] Network UPS Tools version 2.8.1.1 (release/snapshot of 2.8.1.1) built with gcc (Debian 12.2.0-14) 12.2.0 and configured all by default guesswork
   0.000046     [D1] debug level is '6'
   0.000048     [D5] send_to_all: SETINFO driver.debug "6"
   0.000050     [D5] send_to_all: SETFLAGS driver.debug RW NUMBER
   0.000531     [D1] Succeeded to become_user(root): now UID=0 GID=0
   0.000559     [D5] send_to_all: SETINFO device.type "ups"
   0.000563     [D5] send_to_all: SETINFO driver.state "init.device"
   0.000565     [D1] upsdrv_initups...
   0.113551     [D3] usb_busses=0x556bd1e82a60
   0.113583     [D2] Checking device (174C/1153) (002/003)
   0.113981     [D2] - VendorID: 174c
   0.113994     [D2] - ProductID: 1153
   0.113995     [D2] - Manufacturer: ASMedia
   0.113997     [D2] - Product: AS2115
   0.113998     [D2] - Serial Number: 00000000000000000000
   0.113999     [D2] - Bus: 002
   0.114000     [D2] - Device: 003
   0.114001     [D2] - Device release number: 0001
   0.114002     [D2] Trying to match device
   0.114003     [D3] match_function_regex: matching a device...
   0.114011     [D2] match_function_regex: failed match of VendorID: 174c
   0.114013     [D2] Device does not match - skipping
   0.114024     [D2] Checking device (18D1/9302) (002/002)
   0.114053     [D2] - VendorID: 18d1
   0.114055     [D2] - ProductID: 9302
   0.114056     [D2] - Manufacturer: unknown
   0.114057     [D2] - Product: unknown
   0.114058     [D2] - Serial Number: unknown
   0.114059     [D2] - Bus: 002
   0.114060     [D2] - Device: 002
   0.114061     [D2] - Device release number: 0100
   0.114062     [D2] Trying to match device
   0.114063     [D3] match_function_regex: matching a device...
   0.114065     [D2] match_function_regex: failed match of VendorID: 18d1
   0.114066     [D2] Device does not match - skipping
   0.114070     [D2] Checking device (1D6B/0003) (002/001)
   0.114403     [D2] - VendorID: 1d6b
   0.114412     [D2] - ProductID: 0003
   0.114414     [D2] - Manufacturer: Linux 6.5.11-7-pve xhci-hcd
   0.114415     [D2] - Product: xHCI Host Controller
   0.114416     [D2] - Serial Number: 0000:00:14.0
   0.114417     [D2] - Bus: 002
   0.114418     [D2] - Device: 001
0.114419     [D2] - Device release number: 0605
   0.114420     [D2] Trying to match device
   0.114421     [D3] match_function_regex: matching a device...
   0.114425     [D2] match_function_regex: failed match of VendorID: 1d6b
   0.114426     [D2] Device does not match - skipping
   0.114436     [D2] Checking device (8087/0A2A) (001/004)
   0.114454     [D2] - VendorID: 8087
   0.114455     [D2] - ProductID: 0a2a
   0.114456     [D2] - Manufacturer: unknown
   0.114457     [D2] - Product: unknown
   0.114458     [D2] - Serial Number: unknown
   0.114459     [D2] - Bus: 001
   0.114460     [D2] - Device: 004
   0.114461     [D2] - Device release number: 0003
   0.114462     [D2] Trying to match device
[... 54 lines cut ...]
   0.117884     [D2] voltronic_qs_hex_protocol: invalid protocol [M]
   0.117885     [D4] ups_infoval_set: failed to preprocess value [ups.firmware.aux: M]
   0.117888     [D2] armac: unsupported cmd: QS
   0.117890     [D2] qx_process_answer: short reply (input.voltage)
   0.117891     [D2] armac: unsupported cmd: QS
   0.117892     [D2] qx_process_answer: short reply (input.voltage)
   0.117893     [D2] armac: unsupported cmd: QS
   0.117894     [D2] qx_process_answer: short reply (input.voltage)
   0.117896     [D2] armac: unsupported cmd: D
   0.117898     [D2] qx_process_answer: short reply (input.voltage)
   0.117899     [D2] armac: unsupported cmd: D
   0.117900     [D2] qx_process_answer: short reply (input.voltage)
   0.117901     [D2] armac: unsupported cmd: D
   0.117902     [D2] qx_process_answer: short reply (input.voltage)
   0.117905     [D4] armac command Q1
   0.120213     [D1] interrupt read error: error submitting URB: No such file or directory (-2)
   0.120253     [D5] send_to_all: SETINFO driver.state "reconnect.trying"
   0.120309     [D4] qx_process: failed to preprocess answer [input.voltage]
   0.120648     [D3] usb_busses=0x556bd1e82a60
```
and then it retries many times.

Comparing my output whit @calillo I see that afer `armac command Q1` I have` interrupt read error: error submitting URB: No such file or directory (-2)`

I need something else?

Tux92, 2024-01-10:
Last update, I finally managed to run it.

On prerequisite there is a paragraph that eronely I jumped. I execute this: `apt-cache search 'libusb.*1\.0.*'` that return: 

```
libdevice-usb-perl - Perl module to access USB devices using libusb
libftdi1-2 - C Library to control and program the FTDI USB controllers
libftdipp1-3 - C++ Library to control and program the FTDI USB controllers
python3-ftdi1 - Python module to control and program the FTDI USB controllers
libusb-1.0-0 - userspace USB programming library
libusb-1.0-0-dev - userspace USB programming library development files
libusb-1.0-doc - documentation for userspace USB programming
libusb-libusb-perl - Perl interface to the libusb-1.0 API
lm4flash - Command-line firmware flashing tool to communicate with the Stellaris Launchpad
libusb-ocaml - OCaml bindings to libusb-1.0 (runtime)
libusb-ocaml-dev - OCaml bindings to libusb-1.0
```

So I install **libusb-1.0-0-dev** with `apt install libusb-1.0-0-dev` recompile all and now it magically works!