networkupstools/nut #2227: nutdrvctl

aquette, 2023-12-12:
simple rename of upsdrctl or new thing to handle usb scan + user overrides (for cheap units) and manually added units?

Note: simply launching the driver with udev RUN may cause repetitive USB disconnection / reconnection. Using nutdrvctl will safeguard the configuration (inc. addition, update, deletion) and restart according to real updates, not transient events.

jimklimov, 2023-12-12:
Just in case, note there is now also an `upsdrvsvcctl` which emulates `upsdrvctl` CLI API as much as possible, to implement handling of systemd/SMF wrapped units for NUT drivers made with nut-driver-enumerator.

Also there was a branch with some nutconf tooling, waiting to get looked at and to integrate into NUT upstream...

jimklimov, 2024-01-20:
Also note NDE as it already reacts to `ups.conf` changes.