networkupstools/nut #2230: Add support for ups.conf.d

aquette, 2023-12-12:
* implement the standard ".d" config management mechanism for ups.conf entries
* any file named <something>-ups.conf, under the sysconfdir, will be evaluated and considered by upsd and upsdrvctl/nutdrvctl
* either use some "#include" in ups.conf or simply make ups.conf optional (non-existing by default) and always check for .d

Note: this should allow a simplified management of devices config entries, according to their type (autodetected, manual, autodetected-with-override) and greatly ease SmartNUT integration.
For example:
* usb-ups.conf can always be refreshed,
* in case of duplicated entries in manual-ups.conf (user override) and usb-ups.conf, these autogenerated ones will be discarded
(support mandatory but undetectable params for cheap units, and additional user params for any drivers)

jimklimov, 2023-12-12:
Something I wanted for so long... proposed but it was played down back in the day... :)

aquette, 2023-12-25:
Another of these 20+ yo things surfacing again 😉

jimklimov, 2024-01-20:
As mentioned elsewhere: keep in mind the `nut-driver-enumerator` feature, https://github.com/networkupstools/nut/wiki/nut%E2%80%90driver%E2%80%90enumerator-(NDE) as it has potential overlap (or at least would need to tap into the changed config layout), e.g. it takes care of "bus" related nuances to set up unit dependencies.