networkupstools/nut #2232: Split dummy-ups between nutdrv_nutclient and nutdrv_simulation

aquette, 2023-12-12:
Part of #94, with some refresh

dummy-ups code should be split, to better separate its 2 current purposes and serve in SmartUPS:
* nutdrv_nutclient could serve as a general NUT client, which connects to a remote upsd. This can replace upsmon, to repeat devices data from a remote upsd.
* nutdrv_simulation: provide static or dynamic device simulation from a plain text file.
Useful for development, testing and troubleshooting purpose

jimklimov, 2023-12-12:
While looking at it, keep in mind the earlier discussion about the two `clone*` drivers which seem to share a lot with `dummy-ups` and among themselves. Maybe here's an opportunity to refactor a lot into one?..

aquette, 2023-12-21:
@jimklimov iirc, the clone* are local (dstate, same figth as dstate_nut2mqtt...)
still, I'll take a look. At least for future consolidation...

Good for the names otherwise? (nutdrv_nutclient & nutdrv_simulation)

jimklimov, 2023-12-22:
I'm in favor of names that fit in narrow-ish terminals and tools, so under ~16 chars of unambiguous name is a benefit. The prefix makes sense but eats 7 chars, so gotta be careful with the rest :)